mavlink/mavros #734: Distance Sensor Plugin Msg not received by FCU

JD-ETH, 2017-06-19:
I am converting a spinning lidar information to distance sensor information, and sequentially sent it over to my FCU. But after setup the plugin and make sure my node is publishing data, I cant oberserve the Mavlink messages on the MavLink inspector of QGroundControl

Mavros Node info:

> Node [/mavros]
> Publications: 
>  * /mavros/manual_control/control [mavros_msgs/ManualControl]
>  * /mavros/imu/data_raw [sensor_msgs/Imu]
>  * /mavros/imu/mag [sensor_msgs/MagneticField]
>  * /mavros/radio_status [mavros_msgs/RadioStatus]
>  * /mavros/setpoint_raw/target_local [mavros_msgs/PositionTarget]
>  * /mavros/imu/atm_pressure [sensor_msgs/FluidPressure]
>  * /mavros/altitude [mavros_msgs/Altitude]
>  * /mavros/mission/waypoints [mavros_msgs/WaypointList]
>  * /mavros/global_position/compass_hdg [std_msgs/Float64]
>  * /mavros/distance_sensor/rangefinder_pub [sensor_msgs/Range]
>  * /mavlink/from [mavros_msgs/Mavlink]
>  * /mavros/local_position/pose [geometry_msgs/PoseStamped]
>  * /mavros/imu/data [sensor_msgs/Imu]
>  * /mavros/wind_estimation [geometry_msgs/TwistStamped]
>  * /mavros/setpoint_raw/target_global [mavros_msgs/GlobalPositionTarget]
>  * /mavros/global_position/global [sensor_msgs/NavSatFix]
>  * /mavros/imu/temperature [sensor_msgs/Temperature]
>  * /mavros/px4flow/raw/optical_flow_rad [mavros_msgs/OpticalFlowRad]
>  * /diagnostics [diagnostic_msgs/DiagnosticArray]
>  * /mavros/global_position/raw/fix [sensor_msgs/NavSatFix]
>  * /mavros/battery [sensor_msgs/BatteryState]
>  * /mavros/setpoint_raw/target_attitude [mavros_msgs/AttitudeTarget]
>  * /mavros/cam_imu_sync/cam_imu_stamp [mavros_msgs/CamIMUStamp]
>  * /mavros/state [mavros_msgs/State]
>  * /mavros/rc/in [mavros_msgs/RCIn]
>  * /rosout [rosgraph_msgs/Log]
>  * /mavros/global_position/local [nav_msgs/Odometry]
>  * /mavros/hil_actuator_controls [mavros_msgs/HilActuatorControls]
>  * /tf [tf2_msgs/TFMessage]
>  * /mavros/vfr_hud [mavros_msgs/VFR_HUD]
>  * /mavros/px4flow/ground_distance [sensor_msgs/Range]
>  * /mavros/px4flow/temperature [sensor_msgs/Temperature]
>  * /mavros/hil_controls/hil_controls [mavros_msgs/HilControls]
>  * /mavros/global_position/rel_alt [std_msgs/Float64]
>  * /mavros/local_position/odom [nav_msgs/Odometry]
>  * /mavros/extended_state [mavros_msgs/ExtendedState]
>  * /mavros/local_position/velocity [geometry_msgs/TwistStamped]
>  * /mavros/global_position/raw/gps_vel [geometry_msgs/TwistStamped]
>  * /mavros/rc/out [mavros_msgs/RCOut]
>  * /mavros/time_reference [sensor_msgs/TimeReference]
> 
> Subscriptions: 
>  * /mavros/setpoint_attitude/attitude [unknown type]
>  * /mavros/actuator_control [unknown type]
>  * /mavros/setpoint_raw/global [unknown type]
>  * /mavros/setpoint_attitude/att_throttle [unknown type]
>  * /mavros/setpoint_attitude/cmd_vel [unknown type]
>  * /tf [tf2_msgs/TFMessage]
>  * /mavros/setpoint_raw/attitude [unknown type]
>  * /mavros/setpoint_velocity/cmd_vel [unknown type]
>  * /mavros/setpoint_position/local [unknown type]
>  * /mavlink/to [mavros_msgs/Mavlink]
>  * /mavros/mocap/pose [unknown type]
>  * /tf_static [unknown type]
>  * /mavros/vision_pose/pose_cov [unknown type]
>  * /mavros/fake_gps/fix [unknown type]
>  * /mavros/setpoint_accel/accel [unknown type]
>  * /mavros/rc/override [unknown type]
>  * /mavros/setpoint_raw/local [unknown type]
>  * /mavros/vision_pose/pose [unknown type]
>  * /mavros/vision_speed/speed_vector [unknown type]
>  * /mavros/distance_sensor/lidar_split6 [sensor_msgs/Range]
>  * /mavros/distance_sensor/lidar_split7 [sensor_msgs/Range]
>  * /mavros/distance_sensor/lidar_split4 [sensor_msgs/Range]
>  * /mavros/distance_sensor/lidar_split5 [sensor_msgs/Range]
>  * /mavros/distance_sensor/lidar_split2 [sensor_msgs/Range]
>  * /mavros/distance_sensor/lidar_split3 [sensor_msgs/Range]
>  * /mavros/distance_sensor/lidar_split0 [sensor_msgs/Range]
>  * /mavros/distance_sensor/lidar_split1 [sensor_msgs/Range]
> 
> Services: 
>  * /mavros/cmd/set_home
>  * /mavros/param/get
>  * /mavros/ftp/checksum
>  * /mavros/cmd/land
>  * /mavros/ftp/mkdir
>  * /mavros/param/pull
>  * /mavros/mission/push
>  * /mavros/ftp/close
>  * /mavros/ftp/rmdir
>  * /mavros/ftp/list
>  * /mavros/cmd/command_int
>  * /mavros/param/push
>  * /mavros/param/set
>  * /mavros/cmd/takeoff
>  * /mavros/ftp/truncate
>  * /mavros/get_loggers
>  * /mavros/mission/pull
>  * /mavros/cmd/command
>  * /mavros/ftp/rename
>  * /mavros/ftp/open
>  * /mavros/ftp/read
>  * /mavros/ftp/remove
>  * /mavros/ftp/reset
>  * /mavros/mission/clear
>  * /mavros/set_stream_rate
>  * /mavros/set_mode
>  * /mavros/cmd/arming
>  * /mavros/cmd/trigger_control
>  * /mavros/set_logger_level
>  * /mavros/mission/set_current
>  * /mavros/ftp/write
> 
> 
> contacting node http://erle-brain-3:40925/ ...
> Pid: 1114
> Connections:
>  * topic: /rosout
>     * to: /rosout
>     * direction: outbound
>     * transport: TCPROS
>  * topic: /tf
>     * to: /mavros
>     * direction: outbound
>     * transport: INTRAPROCESS
>  * topic: /tf
>     * to: /mavros (http://erle-brain-3:40925/)
>     * direction: inbound
>     * transport: INTRAPROCESS
>  * topic: /mavros/distance_sensor/lidar_split0
>     * to: /laser_to_mavlink (http://erle-brain-3:36352/)
>     * direction: inbound
>     * transport: TCPROS
>  * topic: /mavros/distance_sensor/lidar_split1
>     * to: /laser_to_mavlink (http://erle-brain-3:36352/)
>     * direction: inbound
>     * transport: TCPROS
>  * topic: /mavros/distance_sensor/lidar_split2
>     * to: /laser_to_mavlink (http://erle-brain-3:36352/)
>     * direction: inbound
>     * transport: TCPROS
>  * topic: /mavros/distance_sensor/lidar_split3
>     * to: /laser_to_mavlink (http://erle-brain-3:36352/)
>     * direction: inbound
>     * transport: TCPROS
>  * topic: /mavros/distance_sensor/lidar_split4
>     * to: /laser_to_mavlink (http://erle-brain-3:36352/)
>     * direction: inbound
>     * transport: TCPROS
>  * topic: /mavros/distance_sensor/lidar_split5
>     * to: /laser_to_mavlink (http://erle-brain-3:36352/)
>     * direction: inbound
>     * transport: TCPROS
>  * topic: /mavros/distance_sensor/lidar_split6
>     * to: /laser_to_mavlink (http://erle-brain-3:36352/)
>     * direction: inbound
>     * transport: TCPROS
>  * topic: /mavros/distance_sensor/lidar_split7
>     * to: /laser_to_mavlink (http://erle-brain-3:36352/)
>     * direction: inbound
>     * transport: TCPROS
> 
> ----

### Issue details

_Please describe the problem, or desired feature_

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic on Erle Brain 3 board (rapsberry pi 3)
Rapsberry pi linux

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 3.4 rc2

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

TSC21, 2017-06-19:
Please follow the issue topics as requested above and edit the comment.

vooon, 2017-06-19:
How you check messages sent from mavros to FCU?
You need some sort of sniffer tool to see what packets are sent upstream. How you connect to FCU?
Because regular QGC only shows what FCU send (downstream).

JD-ETH, 2017-06-20:
@vooon 
My fcu is a erle ros hat running ardupilot. The connection is fine now, I am able to send rc/override message from ground control station, and receiving messages on both ends for example. (imu data both on GCS and erle linux board). Messages are being published in the corresponding topics as sensor/Range.

Can you recommand any sniffer tool?

TSC21, 2017-06-20:
@JD-ETH doesn't Ardupilot publish `DISTANCE_SENSOR` msgs to the ground station? If yes, then you can basically write a wrapper on Ardupilot side that publishes back the msgs that it is receiving.

vooon, 2017-06-21:
You may use WireShark, with some setup it should be able to deduce mavlink packets (need wlua).

TSC21, 2017-06-21:
> My fcu is a erle ros

This is a Linux-based autopilot. I suppose you can sniff the msgs easily on it.

TSC21, 2017-06-22:
@JD-ETH are you already able to check if the msgs are being sent?

TSC21, 2017-06-28:
@JD-ETH updates?

JD-ETH, 2017-06-28:
@TSC21  Sorry for the late reply.  

I was not confident with WireShark nor any sniffing tools and I was in a hurry to find a solution(I was in a hackathon). So I worked around and used RC/Override instead on the linux board. the result using just RC/Override 
 https://www.youtube.com/watch?v=aWOI7xQgk4Q&t=4s. The code was quite easy too running on ros.

I couldn't see the distance sensor message being published and after a lot of trouble with Erle brain, I am currently switching to Pixhawk2 (cube). I plan to test on that system instead and see whether the plugin works better there.

TSC21, 2017-06-28:
Oh cool stuff @JD-ETH! Well what were you using for position estimation? Why didn't you use velocity setpoints instead of overriding the RC controller?

TSC21, 2017-06-28:
> I couldn't see the distance sensor message being published

Why don't you write a mavlink sender on the Firmware side that publishes back what it receives? Like a `ping`?

TSC21, 2017-06-28:
Oh now I see. You were in `ALTHOLD` and just letting it drift and avoid the walls in the process.

TSC21, 2017-06-28:
In the mean time, if you are willing to find a solution for `DISTANCE_SENSOR` publishing for Laser Scans, I'll leave this PR open. Otherwise, we can close it. But it would be cool if you could get to a solution useful on this case (not the practical case of the Hackaton).

JD-ETH, 2017-07-02:
@TSC21  I would love to help, but now that I am switching to Pixhawk 2 that doesn't natively run ROS, I wonder if this could still help? My setup is going to be Odroid XU4 + pixhawk 2.1 (cube). I am quite unsure still whether I need ROS running on the odroid yet.

I didn't have position estimation because I had big trouble using px4flow on the erle board. So as cool as the object avoidance is, this won't work in real life :P

TSC21, 2017-07-03:
> @TSC21 I would love to help, but now that I am switching to Pixhawk 2 that doesn't natively run ROS, I wonder if this could still help? My setup is going to be Odroid XU4 + pixhawk 2.1 (cube). I am quite unsure still whether I need ROS running on the odroid yet.

Pixhawk 2 is a Mavlink-based autopilot, so it can run both PX4 and APM. And since it is Mavlink-based, it can pair with MAVROS.

I'm closing this then. If you have intentions on doing a PR to add support for LaserScan in the `distance_sensor` plugin, you are welcomed.

closed 2017-07-03 by TSC21