mavlink/mavros #735: MAVROS not publishing unter Router WIFI connection

JD-ETH, 2017-06-19:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Mavros doesn't publish after connected to GCS with WIFI

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic/Indigo
Ubuntu: Rapsberry pi / 14.04

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 3.5rc8

### Node logs

```
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/lidar_split0/id: 1
 * /mavros/distance_sensor/lidar_split0/orientation: NONE
 * /mavros/distance_sensor/lidar_split0/subscriber: True
 * /mavros/distance_sensor/lidar_split1/id: 2
 * /mavros/distance_sensor/lidar_split1/orientation: YAW_315
 * /mavros/distance_sensor/lidar_split1/subscriber: True
 * /mavros/distance_sensor/lidar_split2/id: 3
 * /mavros/distance_sensor/lidar_split2/orientation: YAW_270
 * /mavros/distance_sensor/lidar_split2/subscriber: True
 * /mavros/distance_sensor/lidar_split3/id: 4
 * /mavros/distance_sensor/lidar_split3/orientation: YAW_225
 * /mavros/distance_sensor/lidar_split3/subscriber: True
 * /mavros/distance_sensor/lidar_split4/id: 5
 * /mavros/distance_sensor/lidar_split4/orientation: YAW_180
 * /mavros/distance_sensor/lidar_split4/subscriber: True
 * /mavros/distance_sensor/lidar_split5/id: 6
 * /mavros/distance_sensor/lidar_split5/orientation: YAW_135
 * /mavros/distance_sensor/lidar_split5/subscriber: True
 * /mavros/distance_sensor/lidar_split6/id: 7
 * /mavros/distance_sensor/lidar_split6/orientation: YAW_90
 * /mavros/distance_sensor/lidar_split6/subscriber: True
 * /mavros/distance_sensor/lidar_split7/id: 8
 * /mavros/distance_sensor/lidar_split7/orientation: YAW_45
 * /mavros/distance_sensor/lidar_split7/subscriber: True
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: True
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/fcu_url: udp://:6001@
 * /mavros/gcs_url: udp://192.168.1.2...
[... 311 lines cut ...]
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---

```

### Check ID

```
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

```

It definitely worked before once, but now it stopped working again...

vooon, 2017-06-19:
You do not have a connection. Check your configuration.
Your remote FCU should send packets to ROS machines IP, port 6001.
You may use wireshark to verify data flow.

vooon, 2017-06-19:
Also if you use firewall, check that you allow input on 6001.

JD-ETH, 2017-06-19:
Do you mean at the GCS side?

vooon, 2017-06-20:
At ROS side, since you should find why there is no messages at `udp://:6001@`.
Also check FCU configs, also verify that FCU uses UDP and not TCP. Because APM uses TCP on SITL serial ports, if so use `fcu_url:=tcp://:6001`.

JD-ETH, 2017-06-20:
It was misconfiguration by me. I was indeed not connected to 6001 port. My mistake.

TSC21, 2017-06-20:
Is it solved already @JD-ETH?

closed 2017-06-20 by JD-ETH