mavlink/mavros #736: LPE version in 1.6.1 does not work

SandW, 2017-06-20:
We are working on visual hovering indoor using 6 optitrack cameras. However, it does not work with the following configuration.

----

### Issue details

I can see the data stream in the mocap node and the data was received by the mavros correctly.

> [ INFO] [1495755189.973277609]: FCU: [lpe] mocap position init: 0.11, 0.18, 0.20 m 
> [ INFO] [1495755190.162792328]: FCU: [lpe] mocap timeout

However, the terminal kept sending timeout during the flight and posctrl was rejected. The drone doesnot fly correctly.

### MAVROS version and platform,  Autopilot type and version

Odroid side: 14.04+mavros+mocap_optitrack+wifi; 

Flight controller side: px4 stack 1.6.1 lpe+921600@telem2+Lpe_fusion is set to use vision position and vision yaw and othersare unchecked + ATT_EXT_HDG_M is set to motion capture;

PC side: motive tracker is broadcasting the rigid body frames through wired ethernet connection.

### References

1. https://dev.px4.io/en/ros/external_position_estimation.html
1. https://gitter.im/mavlink/mavros search @SandW, @mzahana and @cjc159_twitter

vooon, 2017-06-20:
Is Header.stamp updates? And perhaps better to ask at http://discuss.px4.io/ as you have problem with firmware.

SandW, 2017-06-20:
Thanks @vooon !
I cannot post topic to the discussion forum as a newbie.
Where can I find Header.stamp?

vooon, 2017-06-20:
It is ROS standard message, normally called `header`.

TSC21, 2017-06-21:
@SandW can you please paste the copy of `rostopic echo -n1 /diagnostics` output please?

SandW, 2017-06-22:
px4.launch:

> [ INFO] [1498106814.436805653]: FCU URL: /dev/ttyUSB0:921600
> [ INFO] [1498106814.437457066]: serial0: device: /dev/ttyUSB0 @ 921600 bps
> [ INFO] [1498106814.439729511]: GCS bridge disabled
> [ INFO] [1498106814.615226658]: Plugin 3dr_radio loaded and initialized
> [ INFO] [1498106814.628204749]: Plugin actuator_control loaded and initialized
> [ INFO] [1498106814.633419885]: Plugin altitude loaded and initialized
> [ INFO] [1498106815.229380130]: Plugin cam_imu_sync loaded and initialized
> [ INFO] [1498106815.252056623]: Plugin command loaded and initialized
> [ INFO] [1498106815.252297247]: Plugin distance_sensor blacklisted
> [ INFO] [1498106815.284015854]: Plugin ftp loaded and initialized
> [ INFO] [1498106815.319096025]: Plugin global_position loaded and initialized
> [ INFO] [1498106815.322166341]: Plugin hil_controls loaded and initialized
> [ INFO] [1498106815.322376631]: Plugin image_pub blacklisted
> [ INFO] [1498106815.355205024]: Plugin imu_pub loaded and initialized
> [ INFO] [1498106815.373844832]: IMU: High resolution IMU detected!
> [ INFO] [1498106815.374523745]: IMU: Attitude quaternion IMU detected!
> [ INFO] [1498106815.383502693]: Plugin local_position loaded and initialized
> [ INFO] [1498106815.386566591]: Plugin manual_control loaded and initialized
> [ INFO] [1498106815.407237971]: Plugin mocap_pose_estimate loaded and initialized
> [ INFO] [1498106815.418638572]: Plugin param loaded and initialized
> [ INFO] [1498106815.442804848]: Plugin px4flow loaded and initialized
> [ INFO] [1498106815.460668744]: Plugin rc_io loaded and initialized
> [ INFO] [1498106815.461000492]: Plugin safety_area blacklisted
> [ INFO] [1498106815.477814603]: Plugin setpoint_accel loaded and initialized
> [ INFO] [1498106815.534282275]: Plugin setpoint_attitude loaded and initialized
> [ INFO] [1498106815.550377139]: Listen to position setpoint transform local_origin -> setpoint
> [ INFO] [1498106815.550980177]: Plugin setpoint_position loaded and initialized
> [ INFO] [1498106815.596244706]: Plugin setpoint_raw loaded and initialized
> [ INFO] [1498106815.609151381]: Plugin setpoint_velocity loaded and initialized
> [ INFO] [1498106815.636874887]: Plugin sys_status loaded and initialized
> [ INFO] [1498106815.651956382]: Plugin sys_time loaded and initialized
> [ INFO] [1498106815.657729099]: Plugin vfr_hud loaded and initialized
> [ INFO] [1498106815.658007889]: Plugin vibration blacklisted
> [ INFO] [1498106815.698114364]: Plugin vision_pose_estimate loaded and initialized
> [ INFO] [1498106815.714650435]: Plugin vision_speed_estimate loaded and initialized
> [ INFO] [1498106815.732623913]: Plugin waypoint loaded and initialized
> [ INFO] [1498106815.732926912]: Autostarting mavlink via USB on PX4
> [ INFO] [1498106815.733108411]: Built-in SIMD instructions: None
> [ INFO] [1498106815.733229702]: Built-in MAVLink package version: 2016.5.20
> [ INFO] [1498106815.733455992]: Built-in MAVLink dialect: ardupilotmega
> [ INFO] [1498106815.733635283]: MAVROS started. MY ID 1.240, TARGET ID 1.1
> [ WARN] [1498106815.755849195]: TM: Clock skew detected (-1498106255.784842014 s). Hard syncing clocks.
[ INFO] [1498108772.709479960]: VER: 1.1: Capabilities 0x00000000000024ef
[ INFO] [1498108772.709706584]: VER: 1.1: Flight software:     010601ff (b4b8a03d10429c85)
[ INFO] [1498108772.709859416]: VER: 1.1: Middleware software: 010601ff (b4b8a03d10429c85)
[ INFO] [1498108772.710033082]: VER: 1.1: OS software:         071000ff (0000000000000000)
[ INFO] [1498108772.710150498]: VER: 1.1: Board hardware:      00000011
[ INFO] [1498108772.710271331]: VER: 1.1: VID/PID: 26ac:0011
[ INFO] [1498108772.710388122]: VER: 1.1: UID: 3034511138353334
[ WARN] [1498108772.710617329]: CMD: Unexpected command 520, result 0
[ INFO] [1498109226.406630331]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ INFO] [1498109226.456784873]: FCU: [lpe] reinit P (1, 1) negative
[ INFO] [1498109228.209286646]: FCU: [lpe] mocap timeout 
[ INFO] [1498109230.611550392]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ INFO] [1498109234.357245497]: FCU: [lpe] mocap timeout 
[ INFO] [1498109237.757345694]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ INFO] [1498109238.256972082]: FCU: [lpe] mocap timeout 
[ INFO] [1498109239.705721537]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ INFO] [1498109241.905954874]: FCU: [lpe] mocap timeout 
[ INFO] [1498109243.554815999]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ INFO] [1498109245.107062436]: FCU: [lpe] mocap timeout 
[ INFO] [1498109247.058304637]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ INFO] [1498109248.858044677]: FCU: [lpe] mocap timeout 
[ INFO] [1498109251.455771121]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ INFO] [1498109252.254385938]: FCU: [lpe] mocap timeout 
[ INFO] [1498109255.356879739]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ INFO] [1498109255.556873452]: FCU: [lpe] mocap timeout 
[ INFO] [1498109257.605863335]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[... 11 lines cut ...]
[ INFO] [1498109268.505314363]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ERROR] [1498109268.707692103]: FCU: REJECT AUTO HOLD
[ERROR] [1498109270.055424520]: FCU: REJECT AUTO HOLD
[ERROR] [1498109270.755174703]: FCU: REJECT AUTO HOLD
[ERROR] [1498109271.652786945]: FCU: REJECT AUTO HOLD
[ INFO] [1498109272.503649041]: FCU: [lpe] mocap timeout 
[... cut ...]
[ERROR] [1498109278.305015115]: FCU: REJECT AUTO HOLD
[ INFO] [1498109280.504610081]: FCU: [lpe] mocap timeout 
[ INFO] [1498109282.454493457]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ INFO] [1498109283.907476804]: FCU: [lpe] mocap timeout 
[ERROR] [1498109284.356329279]: FCU: REJECT AUTO HOLD
[ INFO] [1498109284.556079784]: FCU: [lpe] mocap position init: -0.21,  0.11, -0.12 m 
[ERROR] [1498109285.154713222]: FCU: REJECT AUTO HOLD

`rostopic echo -n1 /diagnostics` ：

[image]

[image]

TSC21, 2017-06-22:
You are getting consecutive timeouts on the mocap streaming. At frequency are you publishing the mocap data? The minimum frequency on the LPE side is 5hz.

SandW, 2017-06-22:
How can I observe the mocap frequency?

TSC21, 2017-06-22:
`rostopic hz /mavros/mocap/pose` or `rosrun tf tf_monitor <mocap_parent_frame> <mocap_child_frame>, depending on what you are currently subscribing? Are you using tf or pose msg?

TSC21, 2017-06-27:
@SandW updates?

SandW, 2017-06-28:
Going to update soon.

SandW, 2017-07-04:
> rostopic hz /mavros/mocap/pose

subscribed to [/mavros/mocap/pose]
average rate: 119.996
	min: 0.008s max: 0.009s std dev: 0.00026s window: 112
average rate: 119.993
	min: 0.008s max: 0.009s std dev: 0.00021s window: 232
average rate: 120.005
	min: 0.007s max: 0.009s std dev: 0.00030s window: 352
average rate: 120.004
	min: 0.007s max: 0.009s std dev: 0.00035s window: 473
average rate: 120.007
	min: 0.007s max: 0.009s std dev: 0.00034s window: 593
^Caverage rate: 120.004
	min: 0.007s max: 0.009s std dev: 0.00033s window: 639

TSC21, 2017-07-07:
Seems to me that the problem may be between the fcu and where you are running mavros. Does your wifi link is stable? Have you tried to increase the timeout time?

SandW, 2017-07-08:
Where can I  increase the timeout time?

TSC21, 2017-07-08:
https://github.com/PX4/Firmware/blob/master/src/modules/local_position_estimator/sensors/mocap.cpp#L10

You can try another thing - throttle the msgs to a lower rate (say 10hz), using http://wiki.ros.org/roscpp/Overview/Time#Sleeping_and_Rates.

Also check the wifi link bandwidth.

TSC21, 2017-07-22:
@SandW updates?

TSC21, 2017-07-27:
Closing now. If the problem persists let us know.

closed 2017-07-27 by TSC21