mavlink/mavros #738: mocap_tf_cb, program not entering callback

crvogt, 2017-06-21:
### Issue details

Hi there,
I'm trying to use MAVROS with a vicon system. Vicon is publishing on /tf and /vicon/{object}/pose, but on testing MAVROS, no data is coming through on the /mavros/mocap/pose topics. I set the "use_tf" parameter to true in the px4_config.yaml file, and added a ROS_INFO_STREAM output to the mocap_tf_cb function, which isn't outputting anything, so not being called is my theory. In starting mavros, it is listing the mocap plugin as loaded and initialized.

Any ideas or direction would be helpful.

Thanks

### MAVROS version and platform

Mavros: Most recent installed by https://dev.px4.io/en/ros/mavros_installation.html
ROS: Indigo
Ubuntu: 14.04

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

TSC21, 2017-06-21:
If `mocap_pose_estimate` plugin is listening to a topic named `/mavros/mocap/pose`, the publishing topic should have that same name, not `/vicon/{object}/pose`. Or you change the publishing topic or you can always change the subscriber to listen to `/vicon/{object}/pose` instead.

TSC21, 2017-06-21:
Besides, check `mocap_pose_estimate` code: or you have `use_tf` set or you have `use_pose` set, not both. Also, if it is publishing on `/tf` topic, same as above, you need to listen to that topic instead of `/mavros/mocap/tf` or you change the publishing topic to ``/mavros/mocap/tf`.

crvogt, 2017-06-21:
Thanks @TSC21 for the quick reply!
I setup mocap_pose_estimate to ensure it was only using use_tf or use_pose, and it's definitely set to use_tf. 

By default, mocap_tf_cb is subscribing to the "tf" topic which is being advertised by Vicon (Vicon ALSO advertises /vicon/{object}/pose), but mocap_pose_estimator is not set to subscribe to that, only "tf". The issue I'm running into is that the callback, mocap_tf_cb, is not even being entered/called. So I don't understand if there's something else I need to do to activate the plugin :/

TSC21, 2017-06-21:
> but mocap_pose_estimator is not set to subscribe to that, only "tf"

That is the problem I reference above and which you should change:
```c++
mocap_tf_sub = mp_nh.subscribe("tf", 1, &MocapPoseEstimatePlugin::mocap_tf_cb, this);
```
should be,
```c++
mocap_tf_sub = mp_nh.subscribe("/vicon/tf", 1, &MocapPoseEstimatePlugin::mocap_tf_cb, this);
```
Do that and recompile in your `catkin_ws`. We will probably add a parameter to add the possibility of selecting the topic to subscribe to.

mhkabir, 2017-06-21:
> We will probably add a parameter to add the possibility of selecting the topic to subscribe to.

No need to. Just remap it.

TSC21, 2017-06-21:
> No need to. Just remap it.

Yeah forgot that detail.
Add this to `mavros_node` in the launch file:
```xml
<remap from="/mavros/mocap/tf" to="/vicon/tf" />
```

crvogt, 2017-06-21:
Ahhhhh! Got it working, thank you very much. The issue I was having was that on running vicon_bridge, the published topics were:
/diagnostics
/rosout
/rosout_agg
/tf
/vicon/object/object
/vicon/markers
"/tf" gives the same output as "/vicon/object/object", and the mocap_pose_estimate cb says to subscribe to tf, so I didn't really look too hard at "/vicon/object/object". But I see now. Thank you again, saved me a lot of frustration.

TSC21, 2017-06-21:
@crvogt good to hear. Can we close this now?

closed 2017-06-21 by mhkabir