mavlink/mavros #740: Distance sensors improvements

khancyr, 2017-06-21:
### FEATURE details

- Add support for laser subscriber and publisher
- Limit the error stream at https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/distance_sensor.cpp#L208 
In case we send distance to FCU, at least on ardupilot, the message will be resend by the FCU so mavros spam the console.

TSC21, 2017-06-21:
> Add support for laser subscriber and publisher

You mean: http://docs.ros.org/api/sensor_msgs/html/msg/LaserScan.html ?

> Limit the error stream at https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/distance_sensor.cpp#L208
In case we send distance to FCU, at least on ardupilot, the message will be resend by the FCU so mavros spam the console.

Ok makes sense. @JD-ETH this may interest you, as Ardupilot already does what you are looking for.

> support mavlink/mavlink#740 (if merged)

Yeah that should be straightforward.

Waiting your PR then :)

khancyr, 2017-06-21:
Yes I was thinking about laserscan, but it doesn't seem that ROS use a lot of 1D laser rangefinder. I will dig a little to see how they handle it. 
Otherwise it could be use for Ardupilot proximity feature that act like a simplified 360 laserscan

TSC21, 2017-06-21:
@khancyr check this: https://github.com/mavlink/mavros/pull/292#issuecomment-101750640. For 1D Laser, we use `INFRARED` type in `sensor_msgs/Range`.

> Otherwise it could be use for Ardupilot proximity feature that act like a simplified 360 laserscan

How do you have that implemented? Multiple `DISTANCE_SENSOR` msgs, one for each beam?

TSC21, 2017-06-21:
Related: https://github.com/ros/common_msgs/pull/73

khancyr, 2017-06-21:
interesting ! Thanks for the information !

About proximity, yes , each sector send a distance_sensor msg with a different id : http://ardupilot.org/dev/docs/code-overview-object-avoidance.html

TSC21, 2017-06-21:
> About proximity, yes , each sector send a distance_sensor msg with a different id : http://ardupilot.org/dev/docs/code-overview-object-avoidance.html

Thanks for sharing

TSC21, 2017-06-22:
@khancyr are you considering doing a PR to this in the mean time?

khancyr, 2017-06-22:
yes probably next week, I am waiting some material to do test

TSC21, 2017-06-22:
> yes probably next week, I am waiting some material to do test

Awesome

TSC21, 2017-07-26:
Solved with #747, #748, #749.

closed 2017-07-26 by TSC21