mavlink/mavros #741: Mavros : Mocap system

b-rohit, 2017-06-23:
### question
### MAVROS version and platform

Mavros: ?0.19.0?
ROS: ?Kinetic?
Ubuntu: ?16.04?
### Autopilot type and version

[  ] ArduPilot
[ X] PX4

I have written a small program based on [https://dev.px4.io/en/ros/mavros_offboard.html](MAVROS offboard control example) which works fine in Gazebo simulation environment. I want to build autonomous quadcopter for indoor purpose. In gazebo simulation as I understand there is virtual coordinate system which helps to fly to specific coordinate. I was wondering how could I achieve the same inside a room. I have nexonar mocap system which gives positional coordinates(x,y,z) of drone with delta, mean and stddev. please someone could give pointers to move ahead.

TSC21, 2017-06-23:
@b-rohit use the [`mocap_pose_estimate`](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/mocap_pose_estimate.cpp) plugin. Just be sure that you configure it to subscribe to whatever the type of msgs you are publishing from your mocap wrapper node (set `use_tf` for `geometry_msgs::TransformStamped` topic or set `use_pose` for `geometry_msgs::PoseStamped` topic).
[`ATT_POS_MOCAP`](http://mavlink.org/messages/common#ATT_POS_MOCAP), currently the msg that MAVROS publishes through this plugin to the FCU, does not support delta, mean or stddev, as their are not currently being used on the FCU estimators.

b-rohit, 2017-06-26:
@TSC21 I subscribed to **_mavros/mocap/pose_** instead of **_mavros/setpoint_position/local_** but I am not able to set OFFBOARD mode. 
 `[ INFO] [1498469378.762178155]: FCU: [lpe] mocap position init:  0.00, -0.00, -2.00 m 
[ WARN] [1498469380.107159166]: PR: request param #430 timeout, retries left 2, and 266 params still missing
[ WARN] [1498469381.707485984]: PR: request param #431 timeout, retries left 2, and 265 params still missing
[ INFO] [1498469382.413052188]: HP: requesting home position
[ WARN] [1498469386.322398840]: PR: request param #435 timeout, retries left 2, and 262 params still missing
[ WARN] [1498469387.393196762]: CMD: Unexpected command 176, result 1
[ WARN] [1498469387.696477616]: PR: request param #436 timeout, retries left 2, and 261 params still missing
[ WARN] [1498469387.917095979]: TM: Clock skew detected (0.011793525 s). Hard syncing clocks.
[ WARN] [1498469389.207856948]: PR: request param #437 timeout, retries left 2, and 260 params still missing
[ INFO] [1498469392.414092694]: HP: requesting home position
[ WARN] [1498469393.544351018]: PR: request param #441 timeout, retries left 2, and 257 params still missing
[ WARN] [1498469395.117230260]: PR: request param #442 timeout, retries left 2, and 256 params still missing
[ WARN] [1498469397.528123501]: CMD: Unexpected command 176, result 1
`
What is the difference between these two topics ? Can I not use mavros/setpoint_position/local topic to send my mocap data to fcu?

TSC21, 2017-06-26:
> Can I not use mavros/setpoint_position/local topic to send my mocap data to fcu?

@b-rohit `mavros/setpoint_position/local` is the topic you subscribe to (being sent from another node that you have which is publishing a `PoseStamped` msg or a TF) which has the position you want the drone to move to when it is on `OFFBOARD` mode. Here's an example code of a ROS node that publishes a position setpoint: https://dev.px4.io/en/ros/mavros_offboard.html (it is a very simple example). This setpoint is received on the FCU and processed in the position controller.

`mavros/mocap/pose` is the topic you subscribe that has the position estimated by your mocap system. If your mocap node is sending this same data in another topic name (`/tf` or `/robot/pose/` for example), you should rename the MAVROS topic (`mavros/mocap/pose`) to the name of these topics or vice-versa (remap these topics to `mavros/mocap/pose`). Example: https://github.com/mavlink/mavros/issues/738#issuecomment-310139828
The mocap data is, in other hand, handle in the FCU on the position estimators, so it is used to estimate the vehicle position, not to give a position `goal` to the drone.

b-rohit, 2017-06-26:
@TSC21 Thanks a lot for explanation. 
I have 2 follow up questions. 
1. what should be the unit of x,y,z position while publishing data on setpoint_position/local or mocap/pose topic? 
2.  when I test in gazebo simulation the drone flies with certain speed in x,y and z direction. where can I modify these values in firmware source code ?

TSC21, 2017-06-26:
1. SI units - meter
2. Change the following parameters: `MPC_XY_CRUISE`, `MPC_Z_VEL_MAX_UP`, `MPC_Z_VEL_MAX_DN`, though these two are max velocities and not nominal as the first.

b-rohit, 2017-06-28:
@TSC21 once again thanks for simple explanation. I have successfully mapped mocap data to /mocap/pose topic and compared the values received by /local_position/pose topic , both are same.
I have not understood fully what is `orientation ` in `geometry_msgs/PoseStamped.msg` ? I get angles in x,y,z direction from mocap system, do I need to send these angle values(degree) as orientation. In that case what should be the value of `orientation.w` ?

TSC21, 2017-06-28:
http://mathworld.wolfram.com/Quaternion.html

TSC21, 2017-06-28:
You should convert your Euler angles into Quaternion format before you send them on the `PoseStamped` msg. You can use:
```c++
<your_mocap_msg>.pose.orientation = tf::createQuaternionMsgFromRollPitchYaw(<mocap_roll>, <mocap_pitch>, <mocap_yaw>);
```

TSC21, 2017-06-28:
Another thing to take into consideration: you need to be sure that the coordinates you send from your mocap node to MAVROS are in the ENU frame.

TSC21, 2017-07-08:
@b-rohit updates?

b-rohit, 2017-07-08:
@TSC21 I am sorry  for delay. I am still working to understand mocap coordinate frame system. I will update you on monday.

TSC21, 2017-07-22:
@b-rohit closing this. We can reopen if you still have problems.

closed 2017-07-22 by TSC21

b-rohit, 2017-08-09:
@TSC21  I was caught up in my semester exams. I made the changes for `Quaternion` and tested without propellers . It is working now.
````
header: 
  seq: 3725
  stamp: 
    secs: 1502272800
    nsecs: 147387027
  frame_id: fcu
pose: 
  position: 
    x: -0.620195
    y: -0.168693
    z: 1.8312235
  orientation: 
    x: -0.00354375641753
    y: 0.635045277224
    z: -0.0101683907273
    w: 0.772399858555
header: 
  seq: 3726
  stamp: 
    secs: 1502272800
    nsecs: 288305044
  frame_id: fcu
pose: 
  position: 
    x: -0.620195
    y: -0.168693
    z: 1.8312235
  orientation: 
    x: -0.00354375641753
    y: 0.635045277224
    z: -0.0101683907273
    w: 0.772399858555
```
Thanks a lot for your support.

TSC21, 2017-08-09:
Glad you could make it

lucasantoro, 2020-04-28:
Hi! There is a little tutorial to set-up this project? i want to understand how to use the plugin inside, for example, the offboard example.
Thank you for your time

TSC21, 2020-04-28:
@lucasantoro please use https://gitter.im/mavlink/mavros.