mavlink/mavros #743: MODE: Unsupported FCU when launched...!!

anudeepsekhar, 2017-06-29:
I have recently updated my pixhawk firmware with the latest stable release PX4 v.1.6.3 and upgraded it using QGC. 
This was the file i used for upgrade `px4fmu-v2_lpe.px4`

I am trying to launch the `iforoffb.launch` and it gives me the following error: 

> [ERROR] [1498730824.057626067]: MODE: Unsupported FCU

I am really confused whats causing this issue please can somebody help me. I have been at it since a month with no success .
Please help me out....!! 
@TSC21 

Here is my launch file:
```
<launch>

<arg name="fcu_url" default= "/dev/ttyUSB0:921600"/>
<arg name="gcs_url" default="" />
<arg name="tgt_system" default="1" />
<arg name="tgt_component" default="1" />

<node name="mavros" pkg="mavros" type="mavros_node" output="screen" required="true" clear_params="true">
	<param name="fcu_url" value="$(arg fcu_url)" />
	<param name="gcs_url" value="$(arg gcs_url)" />
	<param name="target_system_id" value="$(arg tgt_system)" />
	<param name="target_component_id" value="$(arg tgt_component)" />
	<param name="conn_heartbeat" value="5.0" />
	<param name="conn_timeout" value="5.0" />
	<param name="startup_px4_usb_quirk" value="true" />
	<param name="mocap/use_tf" value="true"/>
	<param name="mocap/use_pose" value="false"/>

	<rosparam command="load" file="$(find mavros)/launch/px4_pluginlists.yaml" />
	<rosparam command="load" file="$(find mavros)/launch/px4_config.yaml" />
</node>

<node name="iforoffb_node" pkg="iforoffb" type="iforoffb_node" output="screen">
</node> 
 
</launch>
```

cpp file:

```
#include <ros/ros.h>
#include <geometry_msgs/PoseStamped.h>
#include <mavros_msgs/CommandBool.h>
#include <mavros_msgs/SetMode.h>
#include <mavros_msgs/State.h>

mavros_msgs::State current_state;
void state_cb(const mavros_msgs::State::ConstPtr& msg){
    current_state = *msg;
}

int main(int argc, char **argv)
{
    ros::init(argc, argv, "offb_node");
    ros::NodeHandle nh;

    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
            ("mavros/state", 10, state_cb);
    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
            ("mavros/setpoint_position/local", 10);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
            ("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
            ("mavros/set_mode");

    //the setpoint publishing rate MUST be faster than 2Hz
[... 371 lines cut ...]
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: /dev/ttyUSB0:921600
    values:
      -
        key: Timesyncs since startup
        value: 0
      -
        key: Frequency (Hz)
        value: 0.000000
      -
        key: Last dt (ms)
        value: 0.000000
      -
        key: Mean dt (ms)
        value: 0.000000
      -
        key: Last system time (s)
        value: 0.000000000
      -
        key: Time offset (s)
        value: 0.000000000
---
```

vooon, 2017-06-29:
You do not have a connection with flight controller. Check wiring and param config.

anudeepsekhar, 2017-06-29:
Yes the problem seems to be with the connector as when we re-soldered the wires of the connectors it worked. We were able to launch it successfully with Odroid as our companion computer and the quad armed as well, but when i used putty to serial into pixhawk it is still simply spitting out garbage.

TSC21, 2017-06-29:
@anudeepsekhar the nsh shell will only work for 57600 baud rate.

Since this is working now, I'm closing this.

closed 2017-06-29 by TSC21