mavlink/mavros #744: Feature Request: Support FCU shutdown/reboot from OBC

anuppari, 2017-06-30:
Currently there is no way to signal a reboot of the FCU if e.g., some parameters are changed or there is a [bad boot](https://github.com/PX4/Firmware/issues/7327). This feature is available in QGC, so presumably it can be done through mavlink?

### MAVROS version and platform

Mavros: 0.19
ROS: Kinetic
Ubuntu: 16.04

TSC21, 2017-06-30:
Duplicated of #333. You are welcomed to make a PR with what that feature to be reviewed.

anuppari, 2017-06-30:
This is not a duplicate of #333, this is the opposite. #333 asks to shutdown the OBC from the FCU (or GCS), this issue is to shutdown the FCU from the OBC. Please reopen.

TSC21, 2017-06-30:
Sorry misunderstood. Still, you are welcomed to add the support. We will gladly review it.

TSC21, 2017-07-17:
@anuppari any updates on this? Have you tried to implement the feature already?

vooon, 2017-07-17:
I'm pretty sure that all what you need is call command_long, so you may use `~cmd/long` service to do that.

TSC21, 2017-08-10:
@anuppari `MAV_CMD_PREFLIGHT_REBOOT_SHUTDOWN` is the command code you want, having parameter 1 set to 1. You can create a script for that (PR welcomed) or you can use raw `cmd/long` srv as @vooon stated to that same purpose.

anuppari, 2017-08-23:
How can I access the mavlink enumerations (is there a mavros header that defines these)? Or do I just have to hardcode the value (e.g., 246)?

TSC21, 2017-08-23:
http://mavlink.org/messages/common

hardeman, 2020-04-06:
Is this feature already available?

TSC21, 2020-04-06:
> Is this feature already available?

No.

shakeebbb, 2020-05-27:
I am using PX4 simulator in gazebo. 
rosservice call /mavros/cmd/command "{broadcast: false, command: 246, confirmation: 0, param1: 2, param2: 0.0, param3: 0.0,  param4: 0.0, param5: 0.0, param6: 0.0, param7: 0.0}" seems to shut PX4 down with success: True.

However, the service call below returns success: False, result: 2 and does not reboot PX4.
rosservice call /mavros/cmd/command "{broadcast: false, command: 246, confirmation: 0, param1: 1, param2: 0.0, param3: 0.0,  param4: 0.0, param5: 0.0, param6: 0.0, param7: 0.0}". 

Is there anything I am missing? What are broadcast and confirmation variables used for inside the mavros_msgs/CommandLong.srv? Thanks in advance!

amilcarlucas, 2020-11-19:
@shakeebbb thanks for the tip. That flight controller reboot rosservice call works fine on ArduPilot.