mavlink/mavros #745: Question: Confused on how to get data from rangefinder

NickBoeker, 2017-07-01:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

Using the APM flight stack me and my team can't seem to access the rangefinder data.  We can't seem to properly configure the distance_sensor plugin and while we can seem msgs in the mavlink stream that have the id=173 which would be rangefinder we don't know how to access the payload to get the actual data.

### MAVROS version and platform

Mavros: 0.19.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

ArduPilot
Version: 3.4.6

### Node logs

```
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://apsync:46322/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 1
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fcu_url: /dev/ttyS0:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
[... 357 lines cut ...]
OK. I got messages from 1:1.

---
Received 406 messages, from 19 addresses
sys:comp   list of messages
 206:75    1
 83:95    71
  0:0     0
  0:122   1
 58:1     155
 103:1     1
 136:1     1
  0:206   227
  1:155   184, 0
  1:254   9
 65:1     155
 81:2     132
  1:153   14
 32:0     0
 124:1     155
  2:101   1
 83:84    65
  1:1     0, 2, 70, 22, 253
 206:214   1
```

TSC21, 2017-07-01:
Can you please share your `apm_config.yaml` config?

TSC21, 2017-07-01:
What's the result of `rostopic echo /mavros/distance_sensor/rangefinder_pub/range`?

NickBoeker, 2017-07-01:
apm_config.yaml
```
# Common configuration for APM2 autopilot
#
# node:
startup_px4_usb_quirk: false

# --- system plugins ---

# sys_status & sys_time connection options
conn:
  heartbeat_rate: 1.0    # send hertbeat rate in Hertz
  timeout: 10.0          # hertbeat timeout in seconds
  timesync_rate: 0.0     # TIMESYNC rate in Hertz (feature disabled if 0.0)
  system_time_rate: 0.0  # send system time to FCU rate in Hertz (disabled if 0.0)

# sys_status
sys:
  min_voltage: 10.0   # diagnostics min voltage
  disable_diag: false # disable all sys_status diagnostics, except heartbeat

# sys_time
time:
  time_ref_source: "fcu"  # time_reference source
  timesync_mode: NONE
  timesync_avg_alpha: 0.6 # timesync averaging factor

# --- mavros plugins (alphabetical order) ---

# 3dr_radio
tdr_radio:
  low_rssi: 40  # raw rssi lower level for diagnostics

# actuator_control
# None

# command
cmd:
  use_comp_id_system_control: false # quirk for some old FCUs

# dummy
# None

# ftp
# None

# global_position
global_position:
  frame_id: "fcu"           # pose and fix frame_id
  rot_covariance: 99999.0   # covariance for attitude?
  tf:
    send: false               # send TF?
    frame_id: "local_origin"  # TF frame_id
    child_frame_id: "fcu_utm" # TF child_frame_id

# imu_pub
imu:
  frame_id: "fcu"
  # need find actual values
  linear_acceleration_stdev: 0.0003
  angular_velocity_stdev: !degrees 0.02
  orientation_stdev: 1.0
  magnetic_stdev: 0.0

# local_position
local_position:
  frame_id: "fcu"
  tf:
    send: false
    frame_id: "local_origin"
    child_frame_id: "fcu"

# param
# None, used for FCU params

# rc_io
# None

# safety_area
safety_area:
  p1: {x:  1.0, y:  1.0, z:  1.0}
  p2: {x: -1.0, y: -1.0, z: -1.0}

# setpoint_accel
setpoint_accel:
  send_force: false

# setpoint_attitude
setpoint_attitude:
  reverse_throttle: false   # allow reversed throttle
  tf:
    listen: false           # enable tf listener (disable topic subscribers)
    frame_id: "local_origin"
    child_frame_id: "attitude"
    rate_limit: 10.0

# setpoint_position
setpoint_position:
  tf:
    listen: false           # enable tf listener (disable topic subscribers)
    frame_id: "local_origin"
    child_frame_id: "setpoint"
    rate_limit: 50.0

# setpoint_velocity
# None

# vfr_hud
# None

# waypoint
mission:
  pull_after_gcs: true  # update mission if gcs updates

# --- mavros extras plugins (same order) ---

distance_sensor:
  rangefinder_pub:
    id: 0
    frame_id: "lidar"
    #orientation: PITCH_270 # sended by FCU
    field_of_view: 0.0  # XXX TODO
    send_tf: false
    sensor_position: {x:  0.0, y:  0.0, z:  -0.1}
  rangefinder_sub:
    subscriber: true
    id: 0
    orientation: PITCH_270  # only that orientation are supported by APM 3.4+

## Currently available orientations:
#    Check http://wiki.ros.org/mavros/Enumerations
##

# image_pub
image:
  frame_id: "px4flow"

# mocap_pose_estimate
mocap:
  # select mocap source
  use_tf: false   # ~mocap/tf
  use_pose: true  # ~mocap/pose

# px4flow
px4flow:
  frame_id: "px4flow"
  ranger_fov: !degrees 0.0  # XXX TODO
  ranger_min_range: 0.3     # meters
  ranger_max_range: 5.0     # meters

# vision_pose_estimate
vision_pose:
  tf:
    listen: false           # enable tf listener (disable topic subscribers)
    frame_id: "local_origin"
    child_frame_id: "vision"
    rate_limit: 10.0

# vision_speed_estimate
vision_speed:
  listen_twist: false

# vibration
vibration:
  frame_id: "vibration"

# vim:set ts=2 sw=2 et:
```
echoing /mavros/distance_sensor/rangefinder_pub/range doesn't produce any output

vooon, 2017-07-02:
Distance sensor wants message [DISTANCE_SENSOR №132][1], not RANGEFINDER №173 from apm's dialect.
Fresh APM send both messages, default apm_config.yaml configured for that recent release.

To support RANGEFINDER you need to write plugin, it may be placed anywhere (see PluginLib docs).

[1]: http://mavlink.org/messages/common#DISTANCE_SENSOR

khancyr, 2017-07-02:
Distance_sensor message will only be support on copter > 3.5.0-rc8

vooon, 2017-07-02:
@NickBoeker you may try rangefinder plugin from #746, just update to master.
@khancyr thank you.

TSC21, 2017-07-10:
@NickBoeker is this solved on your side already?

NickBoeker, 2017-07-10:
Unfortunately, it seems like it hasn't but I don't quite know what is the problem anymore.  I think that mavlink is having trouble since we're running the lidar in pwm format instead of i2c.  Please let me know if you know how I could get access to the data or if we should switch to the i2c setup

TSC21, 2017-07-10:
The Mavlink stream doesn't have anything to do with the driver that you are using to get the LIDAR data. You should firstly test that the data is being received correctly on your FCU. Then you can test the Mavlink stream outside. Also if you were getting the msg id for `RANGEFINDER` supposedly that data was being sent on the stream.

NickBoeker, 2017-07-10:
I assume the data is correctly being processed by the fcu as Mission Planner is reporting accurate measurements for lidar.  Do you have a recommendation on what to use to test the stream?

TSC21, 2017-07-11:
If Mission Planner receives it so the msg is on the stream. Are you using latest master of MAVROS? What's the output of `rostopic echo /mavros/rangefinder/rangefinder`?

NickBoeker, 2017-07-11:
I just checked and the output doesn't give anything. I know I updated it to include the new rangefinder plugin but I'm not sure if I updated it since then

TSC21, 2017-07-11:
Can you confirm the plugin is loaded when you run the node?

NickBoeker, 2017-07-11:
Yes, the startup script says that rangefinder plugin has been loaded and initialized

TSC21, 2017-07-11:
Are you getting any other data on the other topics? Can you please post the startup node log here please? Also the output of `diagnostics`?

NickBoeker, 2017-07-11:
node startup:
```
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://apsync:48410/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: RNGFND
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 0
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_url: /dev/ttyS0:921600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/global_position/use_relative_alt: True
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['actuator_contro...
 * /mavros/plugin_whitelist: []
 * /mavros/px4flow/frame_id: px4flow
 * /mavros/px4flow/ranger_fov: 0.0
[... 292 lines cut ...]
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: /dev/ttyS0:921600
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---
```

TSC21, 2017-07-11:
> [ERROR] [1499471828.442373747]: UAS: GeographicLib exception: File not readable /usr/share/GeographicLib/geoids/egm96-5.pgm

Just off-topic: this means you didn't install the datasets for GeographicLib - please run the script which is on the script folder of mavros.

But...
```
 level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: /dev/ttyS0:921600
    values: 
```

This means there is no connection to the FCU.

TSC21, 2017-07-11:
You can even compare to the 1st output of `diagnostics` you posted:
```bash
level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyS0:921600
    values: 
```
Are you doing something different this time?

NickBoeker, 2017-07-11:
I don't think so, I ran the same thing and here is the log.  Should we run all the scripts from the folder? or would that not matter.
```
header: 
  seq: 44
  stamp: 
    secs: 1499471860
    nsecs: 981941287
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyS0:921600
    values: 
      - 
        key: Received packets:
        value: 364
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 55
      - 
        key: Tx sequence number:
        value: 96
      - 
        key: Rx total bytes:
        value: 20721
      - 
        key: Tx total bytes:
        value: 2085
      - 
        key: Rx speed:
        value: 188.000000
      - 
        key: Tx speed:
        value: 25.666666
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyS0:921600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 1
    name: mavros: Heartbeat
    message: Frequency too low.
    hardware_id: /dev/ttyS0:921600
    values: 
      - 
        key: Heartbeats since startup
        value: 11
      - 
        key: Frequency (Hz)
        value: 0.055251
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: ArduPilotMega / ArduCopter
      - 
        key: Mode
        value: LOITER
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyS0:921600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 2
    name: mavros: Battery
    message: No data
    hardware_id: /dev/ttyS0:921600
    values: 
      - 
        key: Voltage
        value: -1.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 0.0
---
```

TSC21, 2017-07-11:
> Should we run all the scripts from the folder?

`install_geographiclib_datasets.sh`

NickBoeker, 2017-07-11:
Ok, we just ran that and it succesfully installed

TSC21, 2017-07-11:
Are you able to get data on other topics?

NickBoeker, 2017-07-11:
Yes, /mavlink/from returns a stream of data

TSC21, 2017-07-11:
Ok so MAVROS is working. Now I can't tell if the `RANGEFINDER` msgs are coming through the stream or not. If yes, `rangefinder` plugin supposedly should handle them. But I'm not able to debug this myself since I use PX4 and not APM. So maybe an APM user can help out? @khancyr can you give an hand here?

NickBoeker, 2017-07-11:
I saw that msgid-173 is ignored when the quad starts up so maybe that is part of the problem but I don't know how that could be fixed

TSC21, 2017-07-11:
Ignored on which side?

NickBoeker, 2017-07-11:
I'm not entirely sure, it is in the node startup log which says something about matrixpilot and slugs ignoring it which goes over my head

TSC21, 2017-07-11:
> I'm not entirely sure, it is in the node startup log which says something about matrixpilot and slugs ignoring it which goes over my head

It's no issue at all for this case.

TSC21, 2017-07-11:
https://github.com/ArduPilot/ardupilot/blob/279072cf255eb1a36d1c29c24752fd2a951b7098/libraries/GCS_MAVLink/GCS_Common.cpp#L259-L270 this is the sending method. Note that it need to bee downward facing on config, which I suppose you already have. It's weird that Mission Planner is able to process the msg but the plugin not. Do you get readable data on Mission Planner?

NickBoeker, 2017-07-11:
It is facing downward, there are distance and voltage numbers coming through to mission planner

TSC21, 2017-07-11:
> It is facing downward, there are distance and voltage numbers coming through to mission planner

Ok so I don't know why it doesn't get the data on MAVROS. @khancyr probably you were able to, since you wrote the plugin, so please give a hand here. thanks

TSC21, 2017-07-11:
Just a last thing: can you post the output of `rostopic list`?

NickBoeker, 2017-07-11:
No problem, let me know if there is anything else that might help
```
/diagnostics
/mavlink/from
/mavlink/to
/mavros/adsb/send
/mavros/adsb/vehicle
/mavros/battery
/mavros/cam_imu_sync/cam_imu_stamp
/mavros/distance_sensor/rangefinder_sub
/mavros/extended_state
/mavros/fake_gps/mocap/tf
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/gp_lp_offset
/mavros/global_position/gp_origin
/mavros/global_position/home
/mavros/global_position/local
/mavros/global_position/raw/fix
/mavros/global_position/raw/gps_vel
/mavros/global_position/rel_alt
/mavros/global_position/set_gp_origin
/mavros/hil_actuator_controls
/mavros/hil_controls/hil_controls
/mavros/home_position/home
/mavros/home_position/set
/mavros/imu/atm_pressure
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/mag
/mavros/imu/temperature
/mavros/local_position/odom
/mavros/local_position/pose
/mavros/local_position/velocity
/mavros/manual_control/control
/mavros/mission/waypoints
/mavros/odometry/odom
/mavros/radio_status
/mavros/rangefinder/rangefinder
/mavros/rc/in
/mavros/rc/out
/mavros/rc/override
/mavros/setpoint_accel/accel
/mavros/setpoint_position/local
/mavros/setpoint_raw/attitude
/mavros/setpoint_raw/global
/mavros/setpoint_raw/local
/mavros/setpoint_raw/target_attitude
/mavros/setpoint_raw/target_global
/mavros/setpoint_raw/target_local
/mavros/setpoint_velocity/cmd_vel
/mavros/setpoint_velocity/cmd_vel_unstamped
/mavros/state
/mavros/time_reference
/mavros/vfr_hud
/mavros/wind_estimation
/rosout
/rosout_agg
/tf
/tf_static
```

TSC21, 2017-07-11:
Ok was just to be sure `rangefinder` was there. Well right now I run out of ideas. Maybe an APM guy can help out now.

TSC21, 2017-07-11:
I don't know but this is the first plugin that is implemented using an `ardupilotmega` msg, and not a common one. I have not tested the plugin myself to assert over it's functioning so I can't tell it is working right now. Also another last thing you can do is disable the `distance_sensor` plugin just to be sure there is no incompatibility with the msgs being sent or received on the stream.

khancyr, 2017-07-11:
@TSC21 @NickBoeker I recheck on SITL with the same copter version and it works fine. Normally if you get the message on mission planner you  should get it too on mavros. 
A thing that can be blocking is the Stream rate : try to call rosservice call /mavros/set_stream_rate 0 10 1 after the connection is established... Or on mission planner go on full parameter tabs and change the stream rate for the telem port that you are using SRX_EXTRA3 (change X by the telem used for mavros)

TSC21, 2017-07-11:
Yeah the stream rate may actually do some difference. You can configure it to all stream or to a specific msg.

khancyr, 2017-07-11:
Ardupilot stick to stream for now : see https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L597 . I hope we can have more granularity soon.

TSC21, 2017-07-11:
> Ardupilot stick to stream for now : see https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L597 . I hope we can have more granularity soon. 

Ok thanks for the info.

NickBoeker, 2017-07-12:
Ok either disabling the distance_sensor plugin or setting the stream rate fixed it, its intermittent but data is appearing so thank you so much

khancyr, 2017-07-12:
you should check your wiring ! it is not normal to have in diagnostic : Frequency too low. Or having intermittend data !

TSC21, 2017-07-12:
True. Usually you should get a heartbeat higher than 1hz

TSC21, 2017-07-17:
@NickBoeker what's the current status of this?

vooon, 2017-07-17:
No, not more that 1Hz. Usual is 1 Hz ± some reasonable deviation (0.01-0.3 Hz).

TSC21, 2017-07-17:
> No, not more that 1Hz. Usual is 1 Hz ± some reasonable deviation (0.01-0.3 Hz).

That's what I wanted to say, as I usually get a little higher than 1Hz.

NickBoeker, 2017-07-19:
Yup, the plug-in works great. The data came in and performed flawlessly in the Ros program. We were trying to transmit too much data over the connection and it was causing the intermittent data. Thanks again for helping me out!

closed 2017-07-19 by mhkabir

AlexWUrobot, 2022-12-20:
I find this solution will be helpful:
https://discuss.px4.io/t/how-to-read-distance-sensor-data-from-mavros-topic/21944