mavlink/mavros #750: No data when echoing Mavros topics

LArayane, 2017-07-06:
hey, guys, I'm using Navio+ with Raspberry Pi 2 to fly a drone but I have a problem with mavros topics. I tried to echo the topics but I got no data streamed, I tried to run `rosrun mavros mavsys rate --all 10` before echoing like suggested in some forums but still couldn't get any data stream. 
Any suggestion would be appreciated Thank you!!

### MAVROS version and platform

Mavros: ?0.18.4?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version

[  ] ArduPilot
[  ] PX4

Version: ?3.7.1?

### Node logs

```
copy output of mavros_node. Usually console where you run roslaunch
```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics
```

### Check ID

```
rosrun mavros checkid
```

TSC21, 2017-07-06:
Pleaae follow what is asked on the issue template

LArayane, 2017-07-06:
I'll give the rest the template tomorrow because i left My navio+raspberry at Lab

TSC21, 2017-07-07:
Did you properly setup the port IP in `/etc/default/ardupilot`? 'Cause it seems you are not gettting any connection on that `FCU_URL`.

LArayane, 2017-07-07:
thanks for the reply. well, I tried with 14550 first and I thought that the problem might be fcu_url and gcs_url on the same port so I changed it to 14650 and both of them worked. I mean the plug of the GCS was green so that means it was getting something RIGHT!! or is there anyway to check it!!

TSC21, 2017-07-07:
Run `result of rostopic echo -n1 /diagnostics` and check.

LArayane, 2017-07-07:
ok, get it now. do you have any idea how could that occur??

TSC21, 2017-07-07:
What could occur?

LArayane, 2017-07-07:
not getting connection on FCU_URL because I am pretty much sure that it was the same port in `/etc/default/ardupilot`

TSC21, 2017-07-07:
Can't tell but probably an incompatibility with the port id. So this is solved?

LArayane, 2017-07-07:
So you think I should try different ports to know which is the right one!!
I'll try that on monday.  can you give the range I can use for ports?? Thanks

TSC21, 2017-07-07:
`MAVlink usually uses port 14550 for IP-based packet forwarding, though other ports can be used if needed.` from Ardupilot docs

LArayane, 2017-07-10:
It worked, but it wasn't really a port problem, I should have just wait for long for it to get running. Thanks a lot for your help

TSC21, 2017-07-10:
Great! Closing then.

closed 2017-07-10 by TSC21