mavlink/mavros #751: Yaw offset in /mavros/setpoint_raw/local

ChrisGermany, 2017-07-11:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

The yaw rotation has a 90degree offset when publishing to /mavros/setpoint_raw/local

### MAVROS version and platform

Mavros: latest
ROS: indigo
Ubuntu: 14.04

### Autopilot type and version

[  ] ArduPilot
[X] PX4

Version: 1.6.3v4

TSC21, 2017-07-17:
@ChrisGermany were you able to test the modifications I did on my branch?

TSC21, 2017-07-27:
@ChrisGermany what's the status of this?

Seanmatthews, 2017-08-05:
I've also seen incorrect yaw offsets in the system. When using MAVROS with PX4, I can't get PX4 to realize that the yaw is not pi/2 at the origin. Along with #762, there are some odd transformations occurring through MAVROS.

TSC21, 2017-08-05:
The issue reported is related to angular velocities being published to the FCU and not the other way around.

Seanmatthews, 2017-08-05:
@TSC21 Thanks for the clarification. To that effect, I see a 90 degree offset when publishing to setpoint_position/local. I believe this is related to what I described, and that there's some more inherent transformation issue at large.

TSC21, 2017-08-10:
@ChrisGermany is this solved to you already?

TSC21, 2017-08-10:
> I believe this is related to what I described, and that there's some more inherent transformation issue at large.

@Seanmatthews as I said this issue is completly different, as there was a lack of a propper frame transform on the `transform_frame_yaw` method (only).

closed 2018-04-18 by TSC21

epiception, 2019-02-14:
Is this is still an issue on v1.8.2? We seem to be facing a 90 degree offset in yaw when publishing 0 to /mavros/setpoint_position/local. Any solutions to this?