mavlink/mavros #754: local_position/pose unable to adjust properly when /gazebo/set_model_state is called

schmittlema, 2017-07-13:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details
When I call /gazebo/set_model_state to reset my model to 0,0 the local position does not hop to  0,0. Instead it slowly moves down.
For Example:
My drone is at 25,25 (x,y) and I reset it to 0,0. I keep printing the local position and instead of it reading 0,0 it reads 25,25   24,24   23,23 etc etc. until it reaches 0,0. 

Is this a bug or am I implementing the local_position topic wrong?

### MAVROS version and platform

Mavros: 0.19.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: 1.0.0

### Node logs
```
INFO  [dataman] Unkown restart, data manager file 'rootfs/fs/microsd/dataman' size is 103090 bytes
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [platforms__posix__drivers__ledsim] LED::init
INFO  [simulator] Not using /dev/ttyACM0 for radio control input. Assuming joystick input via MAVLink.
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
process[gazebo-2]: started with pid [2537]
process[vehicle_spawn_59d5e7024476_2400_2780903416454810023-3]: started with pid [2546]
[ INFO] [1499950034.561136368]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1499950034.561821757]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
process[mavros-4]: started with pid [2582]
SpawnModel script started
[INFO] [1499950034.692981, 0.000000]: Loading model XML from file
[INFO] [1499950034.693421, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1499950034.706463448]: FCU URL: udp://:14540@localhost:14557
[ WARN] [1499950034.707644298]: init: message from ASLUAV, MSG-ID 201 ignored! Table has different entry.
[ WARN] [1499950034.708014755]: init: message from autoquad, MSG-ID 150 ignored! Table has different entry.
[ WARN] [1499950034.708077204]: init: message from autoquad, MSG-ID 152 ignored! Table has different entry.
[ WARN] [1499950034.708348147]: init: message from matrixpilot, MSG-ID 150 ignored! Table has different entry.
[ WARN] [1499950034.708384201]: init: message from matrixpilot, MSG-ID 151 ignored! Table has different entry.
[ WARN] [1499950034.708401564]: init: message from matrixpilot, MSG-ID 152 ignored! Table has different entry.
[ WARN] [1499950034.708419553]: init: message from matrixpilot, MSG-ID 153 ignored! Table has different entry.
[ WARN] [1499950034.708448537]: init: message from matrixpilot, MSG-ID 155 ignored! Table has different entry.
[ WARN] [1499950034.708468509]: init: message from matrixpilot, MSG-ID 156 ignored! Table has different entry.
[ WARN] [1499950034.708514241]: init: message from matrixpilot, MSG-ID 157 ignored! Table has different entry.
[ WARN] [1499950034.708534678]: init: message from matrixpilot, MSG-ID 158 ignored! Table has different entry.
[ WARN] [1499950034.708552518]: init: message from matrixpilot, MSG-ID 170 ignored! Table has different entry.
[ WARN] [1499950034.708570465]: init: message from matrixpilot, MSG-ID 171 ignored! Table has different entry.
[ WARN] [1499950034.708594187]: init: message from matrixpilot, MSG-ID 172 ignored! Table has different entry.
[ WARN] [1499950034.708622454]: init: message from matrixpilot, MSG-ID 173 ignored! Table has different entry.
[ WARN] [1499950034.708640543]: init: message from matrixpilot, MSG-ID 174 ignored! Table has different entry.
[ WARN] [1499950034.708659387]: init: message from matrixpilot, MSG-ID 175 ignored! Table has different entry.
[ WARN] [1499950034.708677082]: init: message from matrixpilot, MSG-ID 176 ignored! Table has different entry.
[ WARN] [1499950034.708696116]: init: message from matrixpilot, MSG-ID 177 ignored! Table has different entry.
[ WARN] [1499950034.708714912]: init: message from matrixpilot, MSG-ID 178 ignored! Table has different entry.
[ WARN] [1499950034.708737161]: init: message from matrixpilot, MSG-ID 179 ignored! Table has different entry.
[ WARN] [1499950034.708758287]: init: message from matrixpilot, MSG-ID 180 ignored! Table has different entry.
[ WARN] [1499950034.708777400]: init: message from matrixpilot, MSG-ID 181 ignored! Table has different entry.
[ WARN] [1499950034.708812526]: init: message from matrixpilot, MSG-ID 182 ignored! Table has different entry.
[ WARN] [1499950034.708837819]: init: message from matrixpilot, MSG-ID 183 ignored! Table has different entry.
[ WARN] [1499950034.708857470]: init: message from matrixpilot, MSG-ID 184 ignored! Table has different entry.
[ WARN] [1499950034.708870050]: init: message from matrixpilot, MSG-ID 185 ignored! Table has different entry.
[ WARN] [1499950034.708881738]: init: message from matrixpilot, MSG-ID 186 ignored! Table has different entry.
[ WARN] [1499950034.709082816]: init: message from paparazzi, MSG-ID 180 ignored! Table has different entry.
[... 292 lines cut ...]
        key: Frequency (Hz)
        value: 10.000000
      - 
        key: Last dt (ms)
        value: 0.079462
      - 
        key: Mean dt (ms)
        value: 0.162211
      - 
        key: Last system time (s)
        value: 47.732918000
      - 
        key: Time offset (s)
        value: 1.167113784
```
### Check ID

```
OK. I got messages from 1:1.

---
Received 7026 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 24, 30, 32, 33, 36, 74, 76, 83, 85, 87, 230, 105, 111, 241, 242, 245
```

jgoppert, 2017-07-13:
Use spawn from gazebo_ros to set the initial position. You are effectively moving it after you initialize the filter, so it is probably using gps to update to the new location.

TSC21, 2017-07-13:
I assume that will be the position estimator recomputing the position through the filter.

schmittlema, 2017-07-13:
Thank you for the quick response! The looked up the filter and that sounds right!
I am currently using spawn_model in my roslaunch:
<node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
    args="-sdf -file $(arg sdf) -model $(arg vehicle) -x $(arg x) -y $(arg y    ) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>

Is this what you mean? I have to reset the model position multiple times during the simulation so that is why I am using gazebo/set_model_state.

TSC21, 2017-07-13:
If you restart the position you will have to restart the position estimator also.

schmittlema, 2017-07-13:
Ok this is good to know for future work. For the purpose of my simulation, I wil just used the /gazebo/get_model_state service to get this info. Thank you for the quick help!

closed 2017-07-13 by schmittlema

schmittlema, 2017-07-17:
Going off of this, is it possible to reset the lpe programmatically? I can open another ticket if that is prefered

jgoppert, 2017-07-17:
Currently mavlink doesn't have any kind of state reset message to my knowledge, this would also be useful to me.

TSC21, 2017-07-17:
Agree. Please open an issue on PX4 Firmware. You can also create a state reset command and create a PR to Mavlink repo so to update common msg definitions with that command.

schmittlema, 2017-07-17:
https://github.com/PX4/Firmware/issues/7632

TSC21, 2017-07-17:
This will only be possible with a Mavlink command. And from there a state reset on the position estimator.