mavlink/mavros #755: Multiple-Vehicles Gazebo Simulation Local Position Estimation (LPE) error

gyang1011, 2017-07-17:
### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Issue details
Hello,

I am trying to write my own launch files that create 2 iris drones (iris_1, iris_2) in a single gazebo simulator. Here is my launch file:

```
<launch>

    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>

    <arg name="est" default="lpe"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>

    <arg name="headless" default="false"/>
    <arg name="gui" default="true"/>

    <arg name="pluginlists_yaml" default="$(find mavros)/launch/px4_pluginlists.yaml" />
    <arg name="config_yaml" default="$(find mavros)/launch/px4_config.yaml" />

    <!-- Load world -->
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
      <arg name="headless" value="$(arg headless)"/>
      <arg name="gui" value="$(arg gui)"/>
      <arg name="world_name" value="$(arg world)" />
      <arg name="debug" value="$(arg debug)" />
      <arg name="verbose" value="$(arg verbose)" />
      <arg name="paused" value="$(arg paused)" />
  </include>

  <!-- UAV1 iris_1-->
<group ns="iris_1">
    <arg name="name" default="iris_1"/>
    <arg name="ns" default="/iris_1/"/>
    <arg name="fcu_url" default="udp://:14540@localhost:14557"/>
    <arg name="vehicle" default="iris_1"/>
    <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/iris/iris.sdf"/>
    <arg name="gcs_url" value=""/>

    <arg name="x" default="5"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>

    <node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
        args="-sdf -file $(arg sdf) -model $(arg name) -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>

    <node name="sitl" pkg="px4" type="px4" output="screen"
            args="$(find px4) $(arg rcS)">
    </node>

    <include file="$(find px4)/launch/mavros.launch">
        <arg name="ns" value="$(arg ns)"/>
        <arg name="gcs_url" value=""/> <!-- GCS link is provided by SITL -->
        <arg name="fcu_url" value="$(arg fcu_url)"/>
        <arg name="pluginlists_yaml" value="$(arg pluginlists_yaml)" />
        <arg name="config_yaml" value="$(arg config_yaml)" />
    </include>
</group>

<!-- UAV1 iris_2-->
<group ns="iris_2">
  <arg name="name" default="iris_2"/>
  <arg name="ns" default="/iris_2/"/>
  <arg name="fcu_url" default="udp://:14541@localhost:14559"/>
  <arg name="vehicle" default="iris_2"/>
  <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)"/>
  <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/iris/iris_2.sdf"/>
  <arg name="gcs_url" value=""/>

  <arg name="x" default="8"/>
  <arg name="y" default="0"/>
  <arg name="z" default="0"/>
  <arg name="R" default="0"/>
  <arg name="P" default="0"/>
  <arg name="Y" default="0"/>

  <node name="$(anon vehicle_spawn)" output="screen" pkg="gazebo_ros" type="spawn_model"
      args="-sdf -file $(arg sdf) -model $(arg name) -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>

  <node name="sitl" pkg="px4" type="px4" output="screen"
          args="$(find px4) $(arg rcS)">
  </node>

  <include file="$(find px4)/launch/mavros_2.launch">
      <arg name="ns" value="$(arg ns)"/>
      <arg name="gcs_url" value=""/> <!-- GCS link is provided by SITL -->
      <arg name="fcu_url" value="$(arg fcu_url)"/>
      <arg name="pluginlists_yaml" value="$(arg pluginlists_yaml)" />
      <arg name="config_yaml" value="$(arg config_yaml)" />
  </include>
</group>

</launch>

<!-- vim: set et ft=xml fenc=utf-8 ff=unix sts=0 sw=4 ts=4 : -->

```

Notice I created two copies of mavros.launch and iris.sdf accordingly (i.e. iris.sdf , iris_2.sdf and mavros.launch, mavros_2,launch)  to set the unique UDP link and everything seems to be working. 
I can launch the the launch file and two drones are loaded without any error. However, when I do `rostopic echo` each drone's `/mavros/local_pose/pose` . Both of them shows the location is x: 0, y:0, z:0, which does not make any sense. Because in my launch file, iris_1 is set to be x:5, y:0, z:0 and iris_2 is set to be x:8, y:0 , z:0 and they do spawn at two different locations. 

Then, I try to use `rosrun rqt_graph rqt_graph` to debug the error. And I got the following graph:

[image]

It looks like both iris nodes are listening to the the /tf and /tf_static. I am really confused now because both iris_1 and iris_2 have the wrong `local_pose/pose`. Just for comparison, this is when I run the single agent simulator using `roslaunch px4 mavros_posix_sitl.launch` and everything works perfectly. The rqt_graph is the following:

[image]

In addition, I am also getting warning:
`VER: unicast request timeout, retries left 2`
`VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities`

Any help would be greatly appreciated! 

### Diagnostics

```
header: 
  seq: 27
  stamp: 
    secs: 22
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: iris_1/mavros: FCU connection
    message: connected
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Received packets:
        value: 10640
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 152
      - 
        key: Tx sequence number:
        value: 216
      - 
        key: Rx total bytes:
        value: 425891
      - 
        key: Tx total bytes:
        value: 11054
      - 
        key: Rx speed:
        value: 20542.000000
      - 
        key: Tx speed:
        value: 517.000000
  - 
    level: 0
    name: iris_1/mavros: GPS
    message: 3D fix
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Satellites visible
        value: 10
      - 
        key: Fix type
        value: 3
      - 
        key: EPH (m)
        value: 0.00
      - 
        key: EPV (m)
        value: 0.00
  - 
    level: 0
    name: iris_1/mavros: Heartbeat
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Heartbeats since startup
        value: 20
      - 
        key: Frequency (Hz)
        value: 1.000000
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: PX4 Autopilot
      - 
        key: Mode
        value: AUTO.LOITER
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: iris_1/mavros: System
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: iris_1/mavros: Battery
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Voltage
        value: 12.15
      - 
        key: Current
        value: -1.0
      - 
        key: Remaining
        value: 100.0
  - 
    level: 0
    name: iris_1/mavros: Time Sync
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Timesyncs since startup
        value: 203
      - 
        key: Frequency (Hz)
        value: 10.000000
      - 
        key: Last dt (ms)
        value: -0.036141
      - 
        key: Mean dt (ms)
        value: 0.155618
      - 
        key: Last system time (s)
        value: 20.542488000
      - 
        key: Time offset (s)
        value: 1.357497543

```

vooon, 2017-07-17:
You don't need two copy of mavros.launch, since you passes `fcu_url` via `<arg />`.
But you may need two config.yaml files to configure different `frame_id` params.
Because `tf_prefix` param is not working and deprecated at Indigo or Jade release (don't remember exact date).

Diag looks normal, but i don't see second `iris_2/mavros` node.
What rqt topic shows on `/iris_1/mavros/local_position/pose`? What publish rate?

gyang1011, 2017-07-17:
I cannot see iris_2/mavros node in diag.. but it appears when I do `rostopic list`. 
When I do `rostopic echo /iris_1/mavros/local_position/pose`, the output is the following:
```
header: 
  seq: 9149
  stamp: 
    secs: 308
    nsecs: 368637894
  frame_id: fcu
pose: 
  position: 
    x: 0.0012251031585
    y: 0.000943960796576
    z: -0.0137275494635
  orientation: 
    x: 0.00771037393645
    y: -0.00622368764844
    z: 0.0112980106524
    w: -0.999887155772
```

I don't know how to show the publish rate though, but it's updating very quickly.

vooon, 2017-07-17:
That is why i recommend **rqt**, in gui you could simultaneously see rate, bandwidth and data.
Else look `rostopic hz`.

Iris_2 diag also looks ok.
I'm unsure, but perhaps local position has different home position - that's why you see 0-es on both.
Because local uses home as origin.

gyang1011, 2017-07-17:
Here is what I got for the pose update:
```
average rate: 30.083
	min: 0.032s max: 0.036s std dev: 0.00122s window: 30
average rate: 30.041
	min: 0.032s max: 0.036s std dev: 0.00121s window: 60
average rate: 30.047
	min: 0.032s max: 0.036s std dev: 0.00128s window: 90
average rate: 30.035
	min: 0.032s max: 0.036s std dev: 0.00129s window: 120
average rate: 30.040
	min: 0.032s max: 0.036s std dev: 0.00125s window: 150
average rate: 30.034
	min: 0.032s max: 0.036s std dev: 0.00126s window: 180
average rate: 30.037
	min: 0.032s max: 0.036s std dev: 0.00124s window: 210
```

I will try to dig deeper to figure out the issue.

gyang1011, 2017-07-17:
I think I figure out the issue. 

The local_pose uses whatever the starting point (can be anywhere on the map) as its origin. For example, my global position can be x: 10000, y: 20000, z:0. But the local_pose will always start at x:0, y:0, z:0 (That is why it's called "local_pose"). 

But my questions is that is there anyway I can overwrite the initial position to be the global position? The `\mavros\global_pose` only gives me GPS location, instead of in Cartesian coordinate system.

gyang1011, 2017-07-17:
OK. I'm just going to hardcode an offset based on each agent's spawning location. This is not a beautiful way to fix the issue, but is certainly the easiest.

closed 2017-07-17 by gyang1011

vooon, 2017-07-18:
You may try to change home position via `/mavros/home_position/set` topic (or older `~cmd/set_home` service).
Maybe that allow you to set same origin.

tensarflow, 2019-06-16:
Hey, could you bind both UAVs to the control station and controlled seperately..? If you used any.. And which control station did you used if you did?

KapilRawal19, 2019-09-02:
Hey, 
I did set the home position but there wasn't any change in the /mavros/local_position/pose does anyone know what can be the issue @vooon, Also for @paradox56  was the issue solved for you?
Thank you

TSC21, 2019-09-02:
> Hey,
> I did set the home position but there wasn't any change in the /mavros/local_position/pose does anyone know what can be the issue @vooon, Also for @paradox56 was the issue solved for you?
> Thank you

@KapilRawal19 setting the home position does not change the vehicle local origin - that's dependent of the estimator and not of the home position of the navigator (assuming you are using PX4).

TSC21, 2019-09-02:
Also multi vehicle simulation is already something documented: https://dev.px4.io/v1.9.0/en/simulation/multi-vehicle-simulation.html. If you are having issues please bring them to the respective forums and issues list and don't comment on closed issues. Thanks

rk800506, 2020-10-29:
@TSC21 , can you  please give the solution of the problem?

TSC21, 2020-10-29:
> @TSC21 , can you please give the solution of the problem?

> If you are having issues please bring them to the respective forums and issues list and don't comment on closed issues. Thanks