mavlink/mavros #757: Not able to see any output from custom plugin LOCAL_POSITION_NED_COV

muralivnv, 2017-07-17:
### Issue details
I wrote a custom plugin that subscribes to mavlink message LOCAL_POSITION_NED_COV, so that i can view the pixhawk EKF covariances. I'm able to see the topic in "rostopic list", but after echoing the topic in terminal I'm not able to see any message details. LOCAL_POSITION_NED is giving the output but the above LOCAL_POSITION_NED_COV is not.

This is how i wrote the plugin,
First of all I wrote the new plugin and added it under "src/plugins/"
Then i changed the "mavros_plugins.xml"  to specify the new plugin
Then added the plugin in "CMakeLists.Txt"
The following the program for custom plugin "LOCAL_POSITION_NED_COV"
` 
#include <mavros/mavros_plugin.h>
#include <eigen_conversions/eigen_msg.h>

#include <geometry_msgs/PoseWithCovarianceStamped.h>
#include <geometry_msgs/AccelWithCovarianceStamped.h>
#include <geometry_msgs/TwistWithCovarianceStamped.h>
#include <geometry_msgs/TransformStamped.h>

#include <nav_msgs/Odometry.h>

namespace mavros {
namespace std_plugins {

/**
 * @brief Local position covariance plugin.
 * Publish local position to TF, PositionWithCovarianceStamped, TwistWithCovarianceStamped
 * and AccelerationWithCovarianceStamped
**/

class LocalPositionCovPlugin : public plugin::PluginBase {
public:
	LocalPositionCovPlugin() : PluginBase(),
		lpc_nh("~local_position_ned_cov"),
		tf_send(false)
	{ }

	void initialize(UAS &uas_)
	{
		PluginBase::initialize(uas_);

		// header frame_id.
		// default to map (world-fixed,ENU as per REP-105).
		lpc_nh.param<std::string>("frame_id", frame_id, "map");

		// Important tf subsection
		// Report the transform from world to base_link here.
		lpc_nh.param("tf/send", tf_send, true);
		lpc_nh.param<std::string>("tf/frame_id", tf_frame_id, "map");
		lpc_nh.param<std::string>("tf/child_frame_id", tf_child_frame_id, "base_link");

		// Debug tf info
		// broadcast the following transform: (can be expanded to more if desired)
		// NED -> aircraft
		lpc_nh.param("tf/send_fcu",tf_send_fcu,false);

		local_position_cov = lpc_nh.advertise<geometry_msgs::PoseWithCovarianceStamped>("pose", 10);
		local_velocity_cov = lpc_nh.advertise<geometry_msgs::TwistWithCovarianceStamped>("velocity", 10);
		local_accel_cov = lpc_nh.advertise<geometry_msgs::AccelWithCovarianceStamped>("acceleration",10);
	}

	Subscriptions get_subscriptions() {
		return {
		       make_handler(&LocalPositionCovPlugin::handle_local_position_ned_cov)
		};
	}

private:
	ros::NodeHandle lpc_nh;

	ros::Publisher local_position_cov;
	ros::Publisher local_velocity_cov;
[... 109 lines cut ...]
			auto ned_position = ftf::transform_frame_enu_ned(enu_position);
			tf::vectorEigenToMsg(ned_position, ned_aircraft_tf.transform.translation);

			auto ned_orientation = ftf::transform_orientation_enu_ned(
						ftf::transform_orientation_baselink_aircraft(enu_orientation));
			tf::quaternionEigenToMsg(ned_orientation,ned_aircraft_tf.transform.rotation);
			m_uas->tf2_broadcaster.sendTransform(ned_aircraft_tf);
		}
	}
};
}	// namespace std_plugins
}	// namespace mavros

#include <pluginlib/class_list_macros.h>
PLUGINLIB_EXPORT_CLASS(mavros::std_plugins::LocalPositionCovPlugin, mavros::plugin::PluginBase)
 `
### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version
[  ] ArduPilot
Version: 3.7.1

vooon, 2017-07-18:
I don't see any usage of `_COV` in APM code.

TSC21, 2017-07-18:
Covariance msgs are not sent through Mavlink stream so it would never work unless you add that feature on Ardupilot Firmware.

muralivnv, 2017-07-18:
When I looked inside common.xml under msgs in mavlink msgs package, I can see the message LOCAL_POSITION_NED_COV defined. Just to confirm what you said @TSC21, so you are saying that APM doesn't support this message and this message is defined for px4 stack autopilots. Is this right?

TSC21, 2017-07-18:
One thing is having the msg defined. Another is the autopilot using it. Neither PX4 is sending it (eventhough is able to receive it).

muralivnv, 2017-07-18:
Thank you @TSC21 for clarifying. I had this doubt for long time. I'm closing this issue.

closed 2017-07-18 by muralivnv