mavlink/mavros #758: Disable tf subscription

anuppari, 2017-07-17:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

TF can use a lot of bandwidth (e.g., if high rate mocap data of multiple UAVs is published on tf), and a tf listener may be unnecessary depending on which plugins are used and their respective parameters. Unfortunately, the tf listener is in the plugin base, which is good to avoid redundant tf subscriptions, but makes disabling it problematic.

TSC21, 2017-07-20:
@anuppari where do you want to reach with this? tf listeners and publishers are optional to use in each plugin, and that's why we use a flag on the plugins to signal if we want to listen or publish the transforms. So what's the idea?

anuppari, 2017-07-21:
The problem is that even if all the flags are false, a tf listener is still instantiated, and therefore a subscriber to the tf topic is instantiated, and therefore there is a a significant bandwidth usage even if none of the plugins are actively using tf. Is there a way to check if any of the plugins are using tf at runtime, and if not, destroy the listener object?

For anyone else running into this issue, a temporary fix is to just remap the /tf and /tf_static topics to dummy topic names when I launch the mavros node.

TSC21, 2017-07-21:
Ok now I understand what you are talking about. Yeah when the `UAS` object is created, also the `tf2_listener` is initiated - https://github.com/mavlink/mavros/blob/2e00d86b773c95fff479c1cf0705511745a34461/mavros/src/lib/uas_data.cpp#L26 - maybe a parameter flag can be used at launch so to init or not a listener when creating the UAS Manager. @vooon what do you think?

vooon, 2017-07-21:
@TSC21 and you forgot that in other topic you wanted to use TF for all transforms?

@anuppari did you tried to use static tf? That may reduce number of messages.

TSC21, 2017-07-21:
@vooon no. I did not forget that. This should be presented as a temporary solution, as I consider that the transforms should be used to apply frame transfom, as @jgoppert suggested

TSC21, 2017-08-10:
@anuppari due to a possible future implementation of tf2 lookup-based frame transformations, that will support frame transforms for different frames of reference (described on `MAV_FRAME` enumeration of Mavlink), and then replacing the static transformations we are applying, this feature you are requesting will be on hold and if the above implementation reveals to be stable, maintaining the performance, probably your idea won't make much sense, as the system will relly on `tf2` on the core in order to support frame transforms.

closed 2017-08-14 by TSC21