mavlink/mavros #759: Sync ROS time with FCU time

Baycken, 2017-07-18:
@vooon @mhkabir
For on-board computers that does not have an accurate RTC, or for UAV that have both  an on-board computer and ground station computer, it is useful to have the ability to sync the FCU time back to ROS time, thus allowing ROS logs and PX4 logs to be all in sync to help debugging.

mhkabir, 2017-07-18:
This is already done. You just need to use the `ntpd_driver` ros node to set the companion clock. FCU already sends a GPS/RTC time reference.

closed 2017-07-18 by mhkabir

vooon, 2017-07-18:
Oh, i forgot to release ntpd_driver for lunar! Anyway it requires system setup for ntpd (or chrony) to use SHM source.

https://github.com/vooon/ntpd_driver.git

On raspberry shm_driver also calls `sudo date <d>` to fix system time.
Because usually GPS module has it's own battery backed rtc, so date are known since first SYSTEM_TIME.

Baycken, 2017-07-18:
Thanks for the quick reply!

If possible, could you guys add a bit of explaining on the wiki for `timesync_mode` for `sys_time` plugin?