mavlink/mavros #760: Obtaining sonar data through mavros using other sonar sensors than px4

L4ncelot, 2017-07-20:
Hi,

I'm working on automatic landing and I need data from sonar sensor connected to px4 to be published to ROS topic. I was looking at [mavros_extras](http://wiki.ros.org/mavros_extras) documentation where's px4flow that does that. The problem is that I'm using different sonar sensor than the px4's one.

Is it still possible to obtain the sonar data using mavros and if so how is the setup different from using px4's sonar?

Thank you for answer.

khancyr, 2017-07-20:
you should use distance_sensor plugin, I think px4 already publish distance_sensors message

TSC21, 2017-07-20:
@khancyr that is correct.
@L4ncelot check how to use the `distance_sensor` plugin here: https://github.com/mavlink/mavros/blob/master/mavros/launch/px4_config.yaml#L120-L142. Basically if you want to receive data from a sonar, you need to be sure that it is published by the Firmware using `DISTANCE_SENSOR` msg. Then, on the MAVROS side, you basically configure a topic which publishes the data in `sensor_msgs/range` msg - basically you can see in the link I sent you that it was created two topics, one for a sonar and other for the LIDAR. Check if you are already receiving data in one of these topics.

L4ncelot, 2017-07-20:
@khancyr, @TSC21, thank you for so fast response. I will have a look at it once I get the sonar sensor working with px4 and then I will report back to you how it went.

L4ncelot, 2017-07-26:
So I've finally got it to work. I had to write analog driver to px4 firmware for my **maxbotix mb1240** sonar sensor. Then I published the messages received from pixhawk's **ADC** using **uORB** messaging in px4 firmware. Afther that I just commented the distance sensor plugin in `plugin_blacklist` variable in `px4_pluginlist.yaml` file. And now it's working and I can see the messages even in ROS. 

Again, thank you for support, really appreciate it.

closed 2017-07-26 by L4ncelot

TSC21, 2017-07-26:
Good to hear @L4ncelot. Ar you willing to commit a PR to PX4 Firmware with the ADC variant of the driver you wrote? Many more people will benefit from it.

L4ncelot, 2017-07-26:
Yeah sure, tomorrow I'll have a look how to contribute and then commit it to PR.

r1b4z01d, 2018-09-13:
@L4ncelot I was able to get my sonar data in via mavros but did you get a chance to commit your code?

L4ncelot, 2018-09-14:
I was trying to do the PR [#7688](https://github.com/PX4/Firmware/pull/7688) but there was some issue in QGC which I didn't solve. Otherwise the code worked fine. But then new version of px4 firmware came and I haven't ported it yet.