mavlink/mavros #762: IMU linear acceleration discrepancy

Seanmatthews, 2017-08-03:
### Issue details

When running MAVROS with PX4, despite the FCU reporting the IMU's linear acceleration in NED, the `/mavros/imu/data` topic does not come back in ENU. Perhaps I'm missing some important setting?

Here's a graph comparison of the IMU linear accelerations I see coming back from both PX4 and MAVROS. The three "humps" are with the snapdragon flight board with 1) bottom facing ground, 2) right side facing ground, and 3) front facing ground. PX4 values in black, MAVROS in white.

[image]
[image]

### MAVROS version and platform

Mavros: 0.19
Mavlink: release/kinetic/mavlink/2017.7.7-0
ROS: Indigo
Ubuntu: 14.04

### Autopilot type and version

[  ] ArduPilot
[ X] PX4

Version: 1.6.4

TSC21, 2017-08-10:
@Seanmatthews does this still bothers you? Based on https://github.com/PX4/Firmware/issues/6562#issuecomment-321028862, does the rotations make sense to you already? The rotations and transformations applied on MAVROS have been flight tested by many people already over and over, some of them using MOCAP systems in order to verify it. So I suppose something is not well explained on the Snapdragon side. Maybe you don't need a aircraft<-> baselink rotation as they may be coincident on the same frame.

TSC21, 2017-08-10:
If you have doubts on how the rotations are applied, here's the code that make them possible: https://github.com/mavlink/mavros/blob/master/mavros/src/lib/ftf_frame_conversions.cpp.

Seanmatthews, 2017-08-10:
@TSC21 I've seen the documentation. The accelerations coming from the board occur in the NED frame. Should they be coming in some other form? I'm told that everything coming from (and going to) MAVROS should occur in the ENU frame, but at least several people find that not to be the case. I believe that it's been flight tested over and over. I've successfully flight tested with MAVROS. However, the coordinate frames align with neither the documentation nor what I'm told. What am I missing?

TSC21, 2017-08-10:
> I'm told that everything coming from (and going to) MAVROS should occur in the ENU frame, but at least several people find that not to be the case.

Correction: every input of MAVROS coming from ROS should come on the ENU frame. Every input coming from the FCU to MAVROS should come on the NED frame. Every output coming from MAVROS to ROS is on the ENU frame (orientation: WRT `base_link` frame, position: WRT `map` or `odom` frame) and every output coming from MAVROS to the FCU is on the NED frame (orientation: WRT `aircraft` frame, position: WRT local/world frame). What may change is the velocities (in the case of the odometry msg for example): in can be WRT body frame or WRT local/world frame. The position is always on the local/world frame.

Seanmatthews, 2017-08-10:
@TSC21 Yes, I misspoke. NED is coming from the FCU, but MAVROS is spitting out IMU data in body frame coordinates. Is there somewhere that base link is set? I there some other setting that perhaps I'm missing?

TSC21, 2017-08-10:
http://www.ros.org/reps/rep-0103.html should clarify your doubt

Seanmatthews, 2017-08-10:
@TSC21 To be clear, are you saying that the IMU data coming from `mavros/imu/data` should occur in the body frame (NWU)? Is that MAVROS' intention?

TSC21, 2017-08-10:
@Seanmatthews what IMU does Snapdragon Flight uses?

TSC21, 2017-08-10:
Ok just to make sure and to clear out that the frame transforms are being well applied: @andrea-nisti are you able to do a test on your Optitrack system so you please share with us the plots for the angular acceleration and attitude both on QGC and `rqt_plot` (for MAVROS)? I just want to clarify this matter once and for all so it doesn't mess with my mind anymore (at least while I don't bring the `tf2` implementation for taking care of the frame transforms).
@andrea-nisti please share also your vehicle setup, meaning your flightcontroller.
Thank you!

TSC21, 2017-08-10:
Another thing: @Seanmatthews do you correctly apply the required transforms from your camera frame so the result estimation comes on ENU to MAVROS?

mhkabir, 2017-08-11:
@Seanmatthews @poitato This is actually fine. Please see ROS [REP 103](http://www.ros.org/reps/rep-0103.html). The quantities you're measuring (linear acceleration) are represented in the PX4 **body frame** ("aircraft"). When these are transformed in mavros to the ROS **body** frame ("base_link"). For body frames, there is no notion of an earth-fixed representation like "NWU" or "ENU". This is only valid for quantities represented in a global frame, like position.

REP 103 classifies the body frame as : 
> In relation to a body the standard is:
> x forward
> y left
> z up

which could indeed be thought of as "NWU", but that is not the correct physical idea because no global alignment exists.

closed 2017-08-11 by mhkabir