mavlink/mavros #763: Wrong default frame_id for local position publisher

okalachev, 2017-08-05:
### Issue details

The default ``frame_id`` for the topic ``~local_position/frame_id`` is set to ``fcu``, which is incorrect: the pose coordinates are relative to the local origin (``local_origin`` TF frame).
Or am I wrong with understanding this?

### MAVROS version and platform

Mavros: 0.18.7
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[X] ArduPilot
[X] PX4

Version: any

### Node logs

```
$ rostopic echo /mavros/local_position/pose
header: 
  seq: 11599
  stamp: 
    secs: 1501958360
    nsecs: 164691648
  frame_id: fcu
pose: 
  position: 
    x: 0.00119905907195
    y: 0.0313315317035
    z: -0.00790994334966
  orientation: 
    x: 0.000341216605005
    y: -0.000721860105067
    z: -0.709982656907
    w: -0.70421868288
```

vooon, 2017-08-06:
Default config keep compatible with previous versions. You may change it via config.yaml.

okalachev, 2017-08-07:
I'm sorry, I just would like to make it clear to me.
You admit, that this is an incorrect default setting, and it's kept it for the compatibility? (I really doubt, that anyone can use this ``frame_id`` field with ``fcu`` value).
Thank you!

TSC21, 2017-08-07:
There is no `local_origin` frame in ROS. Either you consider the frame of the orientation (`aircraft` or `fcu` though those are also not conventional for ROS, as they try to represent the IMU frame, which may be different from the `base_link`, the robot frame) or you consider the frame of the position, and in that case it is named `map` or `odom` depending on the case. Check ROS REP 105. So it really depends on the frame you want to represent. You give the option to the user to change it through a parameter. Though, we still need to review some of the name conventions in some cases.

okalachev, 2017-08-08:
Thanks, I'll read closely the REP-105.
But, as I understand, there exists a frame ``local_origin``, and it's published by mavros, if the appropriate setting is enabled.

TSC21, 2017-08-08:
That was an original naming convention we were using, but not coincident with the ROS comvention.

closed 2017-08-09 by TSC21