mavlink/mavros #765: mavros.register_on_namespace_update(_setup_services)

AldrichCabrera, 2017-08-08:
Hi.

I have a problem when run the command for arm the drone. 

odroid@odroid:~/catkin_ws$ rosrun mavros mavsafety arm
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/mavros/mavsafety", line 18, in <module>
    from mavros import command
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/mavros/command.py", line 51, in <module>
    mavros.register_on_namespace_update(_setup_services)
AttributeError: 'module' object has no attribute 'register_on_namespace_update'

Does anyone know how to solve this?

I have ROS kinetic on odroid, the version the mavros, mavros_msgs is 0.19.0. I try all the solutions that I find in the internet but nobody can solved it.

TSC21, 2017-08-08:
Have you tried to update python?

AldrichCabrera, 2017-08-08:
I have Python 2.7.12 and another computer i have 2.7.6.  I even installed the pkg with:
sudo apt-get install --reinstall python-catkin-pkg

but the error persists

TSC21, 2017-08-08:
What's the ouput of `rostopic echo -n1 /diagnostics` and the ouput of your mavros_node?

vooon, 2017-08-08:
Check content of `/opt/ros/kinetic/lib/python2.7/dist-packages/mavros/__init__.py` file. It should have function `register_on_namespace_update()`.

AldrichCabrera, 2017-08-08:
# To rostopic echo -n1 /diagnostics i get:

header: 
  seq: 42
  stamp: 
    secs: 1502224247
    nsecs: 778723721
  frame_id: ''
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Received packets:
        value: 3093
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 118
      - 
        key: Tx sequence number:
        value: 68
      - 
        key: Rx total bytes:
        value: 99920
      - 
        key: Tx total bytes:
        value: 13559
      - 
        key: Rx speed:
        value: 2893.000000
      - 
        key: Tx speed:
        value: 462.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 1
      - 
        key: EPH (m)
        value: 99.99
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Heartbeats since startup
        value: 13
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: ArduPilotMega / ArduCopter
      - 
        key: Mode
        value: LOITER
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Sensor present
        value: 0x0061FC2F
      - 
        key: Sensor enabled
        value: 0x0061FC2F
      - 
        key: Sensor helth
        value: 0x0061FC2F
      - 
        key: 3D gyro
        value: Ok
      - 
        key: 3D accelerometer
        value: Ok
      - 
        key: 3D magnetometer
        value: Ok
      - 
        key: absolute pressure
        value: Ok
      - 
        key: GPS
        value: Ok
      - 
        key: 3D angular rate control
        value: Ok
      - 
        key: attitude stabilization
        value: Ok
      - 
        key: yaw position
        value: Ok
      - 
        key: z/altitude control
        value: Ok
      - 
        key: x/y position control
        value: Ok
      - 
        key: motor outputs / control
        value: Ok
      - 
        key: rc receiver
        value: Ok
      - 
        key: AHRS subsystem health
        value: Ok
      - 
        key: Terrain subsystem health
        value: Ok
      - 
        key: CPU Load (%)
        value: 23.8
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Voltage
        value: 15.26
      - 
        key: Current
        value: 0.6
      - 
        key: Remaining
        value: 46.0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
  - 
    level: 1
    name: mavros: 3DR Radio
    message: Low RSSI
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: RSSI
        value: 0
      - 
        key: RSSI (dBm)
        value: -127.0
      - 
        key: Remote RSSI
        value: 226
      - 
        key: Remote RSSI (dBm)
        value: -8.1
      - 
        key: Tx buffer (%)
        value: 0
      - 
        key: Noice level
        value: 0
      - 
        key: Remote noice level
        value: 0
      - 
        key: Rx errors
        value: 1
      - 
        key: Fixed
        value: 0

AldrichCabrera, 2017-08-08:
To rostopic echo -n1 /diagnostics i get:

header: 
  seq: 42
  stamp: 
    secs: 1502224247
    nsecs: 778723721
  frame_id: ''
status: 
  - 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Received packets:
        value: 3093
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 118
      - 
        key: Tx sequence number:
        value: 68
      - 
        key: Rx total bytes:
        value: 99920
      - 
        key: Tx total bytes:
        value: 13559
      - 
        key: Rx speed:
        value: 2893.000000
      - 
        key: Tx speed:
        value: 462.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 1
      - 
        key: EPH (m)
        value: 99.99
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Heartbeats since startup
        value: 13
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: ArduPilotMega / ArduCopter
      - 
        key: Mode
        value: LOITER
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Sensor present
        value: 0x0061FC2F
      - 
        key: Sensor enabled
        value: 0x0061FC2F
      - 
        key: Sensor helth
        value: 0x0061FC2F
      - 
        key: 3D gyro
        value: Ok
      - 
        key: 3D accelerometer
        value: Ok
      - 
        key: 3D magnetometer
        value: Ok
      - 
        key: absolute pressure
        value: Ok
      - 
        key: GPS
        value: Ok
      - 
        key: 3D angular rate control
        value: Ok
      - 
        key: attitude stabilization
        value: Ok
      - 
        key: yaw position
        value: Ok
      - 
        key: z/altitude control
        value: Ok
      - 
        key: x/y position control
        value: Ok
      - 
        key: motor outputs / control
        value: Ok
      - 
        key: rc receiver
        value: Ok
      - 
        key: AHRS subsystem health
        value: Ok
      - 
        key: Terrain subsystem health
        value: Ok
      - 
        key: CPU Load (%)
        value: 23.8
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Voltage
        value: 15.26
      - 
        key: Current
        value: 0.6
      - 
        key: Remaining
        value: 46.0
  - 
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
  - 
    level: 1
    name: mavros: 3DR Radio
    message: Low RSSI
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: RSSI
        value: 0
      - 
        key: RSSI (dBm)
        value: -127.0
      - 
        key: Remote RSSI
        value: 226
      - 
        key: Remote RSSI (dBm)
        value: -8.1
      - 
        key: Tx buffer (%)
        value: 0
      - 
        key: Noice level
        value: 0
      - 
        key: Remote noice level
        value: 0
      - 
        key: Rx errors
        value: 1
      - 
        key: Fixed
        value: 0

AldrichCabrera, 2017-08-08:
To mavros_node are:
/diagnostics
/mavlink/from
/mavlink/to
/mavros/actuator_control
/mavros/adsb/send
/mavros/adsb/vehicle
/mavros/altitude
/mavros/battery
/mavros/cam_imu_sync/cam_imu_stamp
/mavros/extended_state
/mavros/fake_gps/fix
/mavros/global_position/compass_hdg
/mavros/global_position/global
/mavros/global_position/local
/mavros/global_position/raw/fix
/mavros/global_position/raw/gps_vel
/mavros/global_position/rel_alt
/mavros/hil_actuator_controls
/mavros/hil_controls/hil_controls
/mavros/home_position/home
/mavros/home_position/set
/mavros/imu/atm_pressure
/mavros/imu/data
/mavros/imu/data_raw
/mavros/imu/mag
/mavros/imu/temperature
/mavros/local_position/odom
/mavros/local_position/pose
/mavros/local_position/velocity
/mavros/manual_control/control
/mavros/mission/waypoints
/mavros/mocap/pose
/mavros/px4flow/ground_distance
/mavros/px4flow/raw/optical_flow_rad
/mavros/px4flow/temperature
/mavros/radio_status
/mavros/rc/in
/mavros/rc/out
/mavros/rc/override
/mavros/setpoint_accel/accel
/mavros/setpoint_attitude/att_throttle
/mavros/setpoint_attitude/attitude
/mavros/setpoint_attitude/cmd_vel
/mavros/setpoint_position/local
/mavros/setpoint_raw/attitude
/mavros/setpoint_raw/global
/mavros/setpoint_raw/local
/mavros/setpoint_raw/target_attitude
/mavros/setpoint_raw/target_global
/mavros/setpoint_raw/target_local
/mavros/setpoint_velocity/cmd_vel
/mavros/setpoint_velocity/cmd_vel_unstamped
/mavros/state
/mavros/time_reference
/mavros/vfr_hud
/mavros/vision_pose/pose
/mavros/vision_pose/pose_cov
/mavros/vision_speed/speed_vector
/mavros/wind_estimation
/rosout
/rosout_agg
/tf
/tf_static

TSC21, 2017-08-08:
@AldrichCabrera that's the rostopic list. I meant the output of the terminal console when you launch the node. And also from `rostopic echo -n1 /diagnostics`. Use markdown formatting for that please.

AldrichCabrera, 2017-08-08:
header: 
    seq: 42
    stamp: 
    secs: 1502224247
    nsecs: 778723721
    frame_id: ''
    status: 
    level: 1
    name: mavros: FCU connection
    message: not connected
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Received packets:
        value: 3093
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 118
      - 
        key: Tx sequence number:
        value: 68
      - 
        key: Rx total bytes:
        value: 99920
      - 
        key: Tx total bytes:
        value: 13559
      - 
        key: Rx speed:
        value: 2893.000000
      - 
        key: Tx speed:
        value: 462.000000
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 1
      - 
        key: EPH (m)
        value: 99.99
      - 
        key: EPV (m)
        value: Unknown
    level: 2
    name: mavros: Heartbeat
    message: No events recorded.
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Heartbeats since startup
        value: 13
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: ArduPilotMega / ArduCopter
      - 
        key: Mode
        value: LOITER
      - 
        key: System status
        value: Standby
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Sensor present
        value: 0x0061FC2F
      - 
        key: Sensor enabled
        value: 0x0061FC2F
      - 
        key: Sensor helth
        value: 0x0061FC2F
      - 
        key: 3D gyro
        value: Ok
      - 
        key: 3D accelerometer
        value: Ok
      - 
        key: 3D magnetometer
        value: Ok
      - 
        key: absolute pressure
        value: Ok
      - 
        key: GPS
        value: Ok
      - 
        key: 3D angular rate control
        value: Ok
      - 
        key: attitude stabilization
        value: Ok
      - 
        key: yaw position
        value: Ok
      - 
        key: z/altitude control
        value: Ok
      - 
        key: x/y position control
        value: Ok
      - 
        key: motor outputs / control
        value: Ok
      - 
        key: rc receiver
        value: Ok
      - 
        key: AHRS subsystem health
        value: Ok
      - 
        key: Terrain subsystem health
        value: Ok
      - 
        key: CPU Load (%)
        value: 23.8
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Voltage
        value: 15.26
      - 
        key: Current
        value: 0.6
      - 
        key: Remaining
        value: 46.0
    level: 2
    name: mavros: Time Sync
    message: No events recorded.
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: Timesyncs since startup
        value: 0
      - 
        key: Frequency (Hz)
        value: 0.000000
      - 
        key: Last dt (ms)
        value: 0.000000
      - 
        key: Mean dt (ms)
        value: 0.000000
      - 
        key: Last system time (s)
        value: 0.000000000
      - 
        key: Time offset (s)
        value: 0.000000000
    level: 1
    name: mavros: 3DR Radio
    message: Low RSSI
    hardware_id: udp://:14550@10.1.1.10:14555
    values: 
      - 
        key: RSSI
        value: 0
      - 
        key: RSSI (dBm)
        value: -127.0
      - 
        key: Remote RSSI
        value: 226
      - 
        key: Remote RSSI (dBm)
        value: -8.1
      - 
        key: Tx buffer (%)
        value: 0
      - 
        key: Noice level
        value: 0
      - 
        key: Remote noice level
        value: 0
      - 
        key: Rx errors
        value: 1
      - 
        key: Fixed
        value: 0

AldrichCabrera, 2017-08-08:
[image]
[image]
[image]

AldrichCabrera, 2017-08-08:
@vooon hi i already checked the content and i definitely have what you tell me

AldrichCabrera, 2017-08-08:
Solved only need install these pkg:
sudo apt-get install geographiclib-tools
sudo apt-get install python-geographiclib

Thanks for you help.

TSC21, 2017-08-08:
That's a result of a bad source install then. Closing

closed 2017-08-08 by TSC21