mavlink/mavros #766: Getting visible satellites count

okalachev, 2017-08-08:
### Issue details

Had not found a topic or something, where I can get ``satellites_visible`` parameter from ``GPS_RAW_INT`` message. Analysing the code did not help.
It there a possibility to get it?

### MAVROS version and platform

Mavros: 0.18.7
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[X] ArduPilot
[X] PX4

Version: Any

TSC21, 2017-08-08:
`rostopic echo -n1 /diagnostics`

vooon, 2017-08-08:
No such topic, because standard `sensor_msgs/NavSatFix` does not have a field for sat count.
Only one place where sat count exposed is diagnostics.

okalachev, 2017-08-08:
What is the standard way to take out the information from /diagnostic (for example, in Python)?
To iterate over DiagnosticArray, and find the value by the ``key`` field? Or there is any more straight solutions?

TSC21, 2017-08-09:
> By the way, it seems to me, that satellites count is an important thing, don't you consider to move it to a separate topic, for example ~global_position/raw/satellites_count?

Unless it is indispensable, we are not going to create a new msg just for that. `/diagnostics` topic publishes msgs of type `diagnostic_msgs/DiagnosticArray` which contains an array of `diagnostic_msgs/DiagnosticStatus`. So you just have to find the `KeyValue[]` (`value.key == Satellites visible`) and get the `value` to that key. That would be on DiagnosticStatus Message with name `mavros: GPS`.

okalachev, 2017-08-09:
I get it, generally.

> Unless it is indispensable, we are not going to create a new msg just for that.
Could it be of type ``std_msgs/UInt8``?

It seems a little bit strange, that the user has to find the data by textual description of it (``Satellites visible``), not by an identifier.

TSC21, 2017-08-09:
One thing you can do is propose adding that field to NavSatFix msg on the ROS repo. That way we could support it also.

TSC21, 2017-08-13:
@okalachev if you plan to propose a change for ROS msgs, probably you would want to propose the field to [`sensor_msgs/NavSatStatus`](http://docs.ros.org/lunar/api/sensor_msgs/html/msg/NavSatStatus.html) instead.

okalachev, 2017-08-13:
I agree, that this should be a proper place for this information (``NavSatStatus`` msg). But I'm afraid, that changing a standard ROS message is too hard thing, and can take really long time. In the ROS ``common_msgs`` repo, I noticed, that it's a really rare event they change a structure of some standard sensor message (not sure they changed any, actually).

Instead, in my Python script, I handle this information from the ``/diagnostics`` topic, like so:

```python
def handle_diagnostics(self, data):
    for status in data.status:
        if status.name in ('mavros: GPS', 'mavros: Heartbeat', 'mavros: FCU connection'):  # diagnostics arrays, I'm interested in
            for value in status.values:
                if value.key == 'Satellites visible':
                     satellites_visible = value.value
                # other stuff...
    # ...

rospy.Subscriber('/diagnostics', DiagnosticArray, handle_diagnostics)
```

As an another decision, I offer to put this stuff to separate topics.

TSC21, 2017-08-14:
> But I'm afraid, that changing a standard ROS message is too hard thing, and can take really long time. In the ROS ``common_msgs`` repo, I noticed, that it's a really rare event they change the structure of some standard sensor message (not sure they changed any, actually).

Yeah that surely is true but making the proposal at least leaves it open to discussion on the ROS `common_msgs` issue list.

> I think you may add stamped sat count, but perhaps you may add sat info too (PRN etc).

Yeah you could create and PR's us a msg proposal where you not only add sat count but other fields related to satellite - you can base the structure on [`GPS_STATUS`](http://mavlink.org/messages/common#GPS_STATUS) Mavlink msg. Though not using all those fields, a future support of this msg on the FCU side will already then be supported on MAVROS side.

okalachev, 2017-08-14:
@vooon @TSC21 According to my tests neither PX4 nor APM (SITLs) send the ``GPS_STATUS`` messages. Satellites count is now taken from the ``GPS_RAW_INT`` message.

Most of the ``GPS_RAW_INT `` fields are present in mavros:
* ``lat``, ``lon``, ``alt`` are present in ``~global_position/raw/fix``
* ``eph``, ``epv`` are present somehow in ``position_covariance`` (this is not completely clear for me)
* ``vel`` is in ``~global_position/raw/gps_vel``

These fields are not present:
* ``satellites_visible``
* ``alt_ellipsoid``, ``h_acc``, ``v_acc``, ``vel_acc``, ``hdg_acc``

So, a new topics for these fields like ``~global_position/raw/satellites_visible`` (UInt8) can be added.

Or it can be a separate message type for the whole raw ``GPS_RAW_INT`` Mavlink package (``mavros_msgs/GpsRawInt``), like the ``mavros_msgs/PositionTarget`` message, corresponding ``SET_POSITION_TARGET_LOCAL_NED`` Mavlink package.

TSC21, 2017-08-14:
> @vooon @TSC21 According to my tests neither PX4 nor APM (SITLs) send ``GPS_STATUS`` messages. Satellites count is now taken from the ``GPS_RAW_INT`` message.

That's what I stated on my previous comment when I said,

> Though not using all those fields, a future support of this msg on the FCU side will already then be supported on MAVROS side.

@okalachev on MAVROS we rarely do one-to-one conversions from MAVLink to ROS correspondetly, since we always look for std ROS msgs that can support the msgs being parsed.
I suggested `GPS_STATUS` has it has more fields related to the satellite info related to satellites and there's no msg on ROS that currently supports all that info - so that way it will be supporting more data coming from the FCU when the proper support is added.
So IMO opinion it would make more senses to add, in the first place, support for `GPS_STATUS` support on the autopilots firmware's and then add the corresponding support on MAVROS. Note: this already a discussion opened [here](https://github.com/mavlink/mavros/issues/717) and on [PX4 Firmware](https://github.com/PX4/Firmware/issues/7392) issue lists.
About the fields you talk about: for `_acc` fields, I support the creation of a msg for that called `NavSatUncer`, which deals with the dimensions uncertainties. For `alt_ellipsoid`, we already support conversions to and from height above the WGS-84 ellipsoid, so what can be added is a verification for that field to see if it has data or not, and if so use it on `NavSatFix` msgs being published on `global_position/global`, so no new topic is needed.
`PositionTarget` was created as a need for raw control of the Setpoints, as it allows to setup the bitmask the way you want.

TSC21, 2017-08-22:
@okalachev are you planning to implement the above?

TSC21, 2017-08-23:
I'm closing this. @okalachev feel free to update us on this when you are able and if you have an idea to a PR, better. thanks

closed 2017-08-23 by TSC21

zubair123khan, 2018-06-24:
I am able to arm my drone (IRIS) , but using takeoff command "rosrun mavros mavcmd takeoffcur 0 0 5
" gives the error "NavSatFix topic not exist".

okalachev, 2018-06-24:
A possible reason can be that there is no global position – no GPS lock, incorrect GPS or compass connection, or incorrect fusion settings (GPS can be disabled in EKF2/LPE).

zubair123khan, 2018-06-25:
GPS seems to be enabled but the error says NAVSATfiX not connected.

Not getting GPS data.

Thanks