mavlink/mavros #767: Problem with Optitrack + px4 ?

EdXian, 2017-08-09:
### Issue details

Before ran the offboard program , i had set my px4  parameter as following website 
https://dev.px4.io/en/ros/external_position_estimation.html 

First , I published  data vrpn_rigidbody_pose( drone's position)  to topic "/mavros/mocap/pose" .  

Second, I checked  data form  /mavros/mocap/pose , it's consistent with data from local_position_pose in GCS mavlink inspector or  rostopic "local_position/pose" . 

Now i try to run offboard program to set my drone at the desired point x,y,z = 0,0,0.3(as the following offb program) , but it seems the drone cant hold and out of control.  

### Autopilot type and version
PX4 : px4fmu-v2_lpe
Firmware Version: 1.6.5

### px4 parameter setting
*LPE_FUSION :  12
*ATT_EXT_HDG_M : 2 (for mocap)
*Unchecked fuse barometer
*LPE_VIC_P  : 0.0001 (default : 0.001)
*LPE_PN_V : 0.2 (default:0.1)
*SYS_COMPANION : 921600

### Companion computer
raspberry pi 2
ROS: Kinetic 
Ubuntu: 16.04
ttyACM0 : 921600

### A part of offboard code
    ros::Subscriber state_sub = nh.subscribe<mavros_msgs::State>
                                ("mavros/state", 10, state_cb);
    ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
                                   ("mavros/setpoint_position/local", 10);
    ros::Publisher mocap_pos_pub = nh.advertise<geometry_msgs::PoseStamped>
                                   ("mavros/mocap/pose", 10);
    ros::ServiceClient arming_client = nh.serviceClient<mavros_msgs::CommandBool>
                                       ("mavros/cmd/arming");
    ros::ServiceClient set_mode_client = nh.serviceClient<mavros_msgs::SetMode>
                                         ("mavros/set_mode");
    ros::Subscriber host_sub = nh.subscribe<geometry_msgs::PoseStamped>("/vrpn_client_node/RigidBody1/pose", 1000, host_pos);
    // The setpoint publishing rate MUST be faster than 2Hz.
    ros::Rate rate(20.0);

    // Wait for FCU connection.
    while (ros::ok() && current_state.connected) {
        ros::spinOnce();
        rate.sleep();
    }

    
    geometry_msgs::PoseStamped pose;

    pose.pose.position.x = 0;            //the desired position
    pose.pose.position.y = 0;
    pose.pose.position.z = 0.3;          
    pose.pose.orientation.w = 1;

    mavros_msgs::SetMode offb_set_mode;
    offb_set_mode.request.custom_mode = "OFFBOARD";

    mavros_msgs::CommandBool arm_cmd;
    arm_cmd.request.value = true;

    ros::Time last_request(0);

    while (ros::ok()) {
        if (current_state.mode != "OFFBOARD" &&
                (ros::Time::now() - last_request > ros::Duration(5.0))) {
            if( set_mode_client.call(offb_set_mode) &&
                    offb_set_mode.response.success) {
                ROS_INFO("Offboard enabled");
            }
            last_request = ros::Time::now();
        } else {

            if (!current_state.armed &&
                    (ros::Time::now() - last_request > ros::Duration(5.0))) {
                if( arming_client.call(arm_cmd) &&
                        arm_cmd.response.success) {
                    ROS_INFO("Vehicle armed");
                }
                last_request = ros::Time::now();
            }
        }

 
	ROS_INFO("setpoint: %.f, %.1f, %.1f", host_mocap.pose.position.x, host_mocap.pose.position.y, host_mocap.pose.position.z);
        mocap_pos_pub.publish(host_mocap);
        local_pos_pub.publish(pose);

        ros::spinOnce();
        rate.sleep();
    }
    return 0;
    }

TSC21, 2017-08-09:
@EdXian node console output (after launching mavros) would help a lot on debigging this. Did you alligned the vehicle body frame with the mocap system frame? Also you have to send a stream of offboard setpoints so it is able to change correctly to offboard and keep the FCU on that mode. I don't see that code above (on the offboard control example you have a for loop stating `//send a few setpoints before starting`). Also try to send a setpoint to z=0.6 or higher in order to avoid problems with ground effect.

TSC21, 2017-08-09:
`pose.pose.orientation.w = 1;` also I advise to send the whole quaternion instead.

EdXian, 2017-08-10:
@TSC21 Thank you for your advice. I'll try to use node console to debug . Is it a utility in QGC? 
After many tests, I found the problem is "The drone is very very unstable at the beginning stage of offb program(atleast 2~3 second)" . 
So I try the follwing steps:
1.reduce  take_off_speed(Min).
2.use joystick to arm the motors and make the drone hover.
3.change mode to offb
After doing above steps , the drone can go into desired position and hold.
Next step , ill try what you said , 
1. send a few setpoints before starting.
2. set higher point to avoid ground effect.
Thanks!!

EdXian, 2017-08-10:
HI @TSC21 : 
I have tried the method you said but it's still unstable.
Now , i find another problem by using MAVlink inspector : After many tests , the value of z-axis in /local_position/pose will increase automatically . Eventhough the drone is static on the ground  , it still increase gradually. However , the value of data in x and y direction is consistent with the data from mocap system.

**Figure 1** :  some data of first test after calibrating px4 . Left side is world frame and right side is px4 frame.
[image]

**Figure 2** : After many tests, the value of z-axis is increasing to 0.35 m.
[image]

I suspect that the px4 still fuse barometer data or others , but I have allready disable` fuse baro` in `LPE_FUSION`. Now i want to check px4 parameter setting  again. Perhaps , there are some parameters
that i haven't take into consideration yet. 

Version : **px4fmu-v2_lpe**

original paramter setting : 
1. `LPE_FUSION` : 12 (only enable fuse vision position and yaw and disable fuse baro)
2. `ATT_EXT_HDG_M` : 2 (for mocap)
3. `take_off_speed` : 1.0M/s
4. `SYS_COMPANION` : 921600
5. `LPE_PN_V` and `LPE_VIC_P` (option)

TSC21, 2017-08-10:
@EdXian possibly related with https://github.com/PX4/Firmware/issues/7723. This is not related with MAVROS anymore so you should move this discussion to the PX4 Firmware issue list.

closed 2017-08-10 by TSC21

m1577, 2017-08-18:
Hello ! @EdXian 

Maybe I have the same problem.

- I am using a optitrack system of six cameras.

-  3dr iris with lpe and 1.6.3 firmware version.

- LPE_FUSION: 12 and ATT_EXT_HDG_M : 2

- sys_companion: I am using a ubuntu pc (Kinetic and ubuntu 16.04) with vrpn_client_ros and mavros packets Installed. I am using a long cable connected to an USB-to-TTL adapter (xbee usb adapter with 921600 baudrate) using TELEM2 port.

- the offboard code: The code is almost the same.

I am usin vrpn_client_ros packet but a lot of people use mocap_optitrack packet, Do I have to change packages?

Here I have a link from my flight test: https://www.youtube.com/watch?v=XBbBbEaNlrc&feature=youtu.be

did you find the solution ?

 https://logs.px4.io/plot_app?log=0c81e0f8-bf97-469e-9c1d-431e9621a036

EdXian, 2017-08-18:
Hi @m1577 

1. Update your px4 firmware version to the latest one (1.6.5).
2. Do you send your mocap data by  `mavlink`? 
3. Either vrpn_client_ros or mocap_optitrack  is fine . what you have to do is check your reference frames.
If  you send your mocap data by mavlink , then you should stream the data  in NED.
You can join this channel where we are talking about `vision` in px4 team in slack: https://px4.slack.com/messages/C69QF7R1A/
Good luck!!

m1577, 2017-08-19:
Hello! @EdXian 
thanks for replying!

I am sending mocap data by mavros, below I will show procedure that I am making:

$ roslaunch vrpn_client_ros sample.launch server: = 192.168.1.192

I connect with a usb my ubuntu pc to fcu (pixhawk).

$ rosrun mavros mavros_node fcuurl: = / dev / ttyUSB0: 921600

To have the pose data sent to mavros, I use a relay node from the topic_tools:

$ rosrun topic_tools relay /vrpn_client_node/Quadcopter /pose/ mavros/mocap/pose

Finally, I send the setpoints to mavros/setpoint_position/local

$ rosrun quadrirotor offboard_example

I am using 1.6.3 firmware version because in this link (https://github.com/PX4/Firmware/releases) the 1.6.5 version do not have "firmware.zip", How can i get the px4fmu-v2_lpe.px4 file from the 1.6.5 firmware?

what does "reference frames" mean?

thanks and regards!

EdXian, 2017-08-19:
@m1577 
1. You can use `git clone https://github.com/PX4/Firmware.git` to get the latest px4 Firmware.
2. Make sure you have set up development environment on Ubuntu for px4 : https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html
3. After uploading px4fmu-v2_default , you can upload px4fmu-v2_lpe.
`make px4fmu-v2_default upload` 
 `make px4fmu-v2_lpe upload `
https://dev.px4.io/en/setup/building_px4.html
4. Download and install QGC.http://qgroundcontrol.com/
5. Use `rosrun mavros mavros_node _fcu_url:="/dev/ttyACM0:921600" _gcs_url:=udp://@QGC IP` to connect your drone to  QGC . QGC is good for debugging.
6. Use QGC Analyze to check LOCAL_POSITION_NED which shoule be in NED coordinates.
BTW,  mocap data you sent should be in ENU coordinates. 
https://dev.px4.io/en/ros/external_position_estimation.html

mindThomas, 2017-09-06:
@TSC21 
> pose.pose.orientation.w = 1; also I advise to send the whole quaternion instead.
Why is that so? The full quaternion doesn't seem to be used within the PX4 code anyways. It only seems to extract the Yaw/Heading and then fuse that with the rest.
Position inputs however seems to be used also to correct the Roll and Pitch bias based on the dynamic model.

TSC21, 2017-09-06:
> Why is that so? The full quaternion doesn't seem to be used within the PX4 code anyways. It only seems to extract the Yaw/Heading and then fuse that with the rest.

It has nothing to do with the PX4 side but rather with the `setpoint_position`. But I verified now that all ROS messages fields (except booleans, which are defaulted to false) are defaulted, in the constructor, to zero. So this does not represent an issue at all. Though, again, it has nothing to do with the PX4 side.

Do1122, 2018-05-26:
Hello, @EdXian 
I am now facing some problems about the lpe,optitrack,mocap,mavros,vrpn...
In fact,I really do not know what's wrong with them,so i just say maybe everything is wrong.
There is some parameter about my experiment,and i wish you can provide some ideas for me.

**Autopilot type and version**
PX4 : px4fmu-v3(with lpe)
Firmware Version: 1.7.4

**px4 parameter setting**
*LPE_FUSION : 0
*ATT_EXT_HDG_M : 2 (for mocap)
*LPE_VIC_P : 0.001 
*LPE_PN_V : 0.1 

**Companion computer**
Jeston TX2
ROS: Kinetic
Ubuntu: 16.04
ttyACM0 : 115200

I am trying to use POSCTL and OFFBOARD mode indoor,but both are  out of control.
I can read the mocap message in the PX4 ulog ,and the local position is the same to it.

I have some questions following:

1.What the LPE_FUSION should be? I don't find the description about in PX4 wiki,but almost every set it 12.
2.There are also someone talking about FAKE GPS,but i set it as default,do you have any idea about it?
3.The coordinate of optitrack is X to right,Y to upwards and Z to backwards,so i change it to ENU in ROS and send it to /mavros/mocap/pose,but  should i transform the q.x,q.y,q.z,q.w?If it should be done,i should use the same transform matrix or any else?
4.Someone said tools in /mavros_extern can change coordinate from ENU to NED,do you use it?
5.The bode of the ttyACM0 is 115200 DEFAULT,how do you change it to 921600?

Our team is small and new,and we have much more questions,anything you mentioned may be useful for us,wish for your regard.

Thank you.

EdXian, 2018-05-26:
Hi @Do1122,
I'm glad that I can help.
The system setting and building of our drone is as follows:
(It had been tested on our drone.)
## Autopilot type and version
PX4 : px4fmu-v2 lpe
Firmware Version: 1.7.4 

px4 parameter setting
* LPE_FUSION : 29
* ATT_EXT_HDG_M : 2 
* LPE_VIC_P : default
* LPE_PN_V : default

Companion computer
* UP board
* ROS: Kinetic
* Ubuntu: 16.04
* ttyACM0 : 921600

1. We set `LPE_FUSION` 29 . (It should also work if you set it 12. You can try it)
2.We dont use Fake GPS. We just refer to `external position estimatopn`. See [here](https://dev.px4.io/en/ros/external_position_estimation.html) 
3.We use package `vrpn_client_ros` to get the position data (ENU) from mocap and send the data directly to the rostopic '/mavros/mocap/pose'.
4.No idea , I never use that.
5.You can try `roslaunch mavros px4.launch fcu_url:="/dev/ttyACM0:921600"` or modify the default value of baudrate in launch file. You can use command 'rospack list' to find the mavros folder.

Good Luck!

Do1122, 2018-05-26:
@EdXian 
I am so glad you reply me ASAP.

Thanks for your advice and I will try to change the LPE_FUSION to 12,and also set the other parameter as you said .
Before i make an attempt,there are still some new questions:

1.You mentioned that you are now setting the LPE_FUSION to 29,so do you use the GPS?Are you making an outdoor experiment whit both GPS and mocap?
2.I am using the optitrack,and the coordinate of it is not ENU,the Y axis is up,and the version of my optitrack tools is low so maybe i must try to transform by hand,how do you get the position message in ENU?Or do you know whether i must transform both the pos and att?
3.I also use the package vrpn_client_ros and send the data of pos and att to '/mavros/mocap/pose' by a new node which created by me.Someone said they use 'remap' to deal it in launch,i have less knowledge about ROS and i do not know whether there are difference between them.So how do you mean when you said 'directly'?May you tell me any detail about it?

I will make some attempts next day,still thank you very much.
Best wishes.