mavlink/mavros #768: Feature: add Fast-RTPS support

TSC21, 2017-08-09:
PX4 already supports fast-RTPS, having a auto message generator for both client (running on the FCU) and the CDR agent, which will run on the other end of the bridge where the fast-RTPS Pubs/Subs are present. MAVROS should support message parsing from the CDR agent directly or integrate the fast-RTPS Pubs/Subs from where it parses the msgs to common ROS msg API, instead of relaying on the autogenerated msgs - it is MAVROS policy and way of working since it was created - use and support the common ROS msgs and only create new msgs (added to `mavros_msgs` package) only when needed.

If the first option is to be taken, should be analyzed the possibility of having the CDR agent, after its integration on MAVROS, only listening to CDR msgs on the serial stream, while `libmavconn` listens and parses Mavlink msgs on the same link. The second option will be more straighforward, but may require updating the code base to ROS2 in order to avoid having to depend on the ROS2-to-ROS1 bridge.

avinash-palleti, 2017-12-06:
Please check the WIP repo https://github.com/avinash-palleti/mavros/tree/rtps to integrate fast-RTPS support into MAVROS and provide feedback.

ArkadiuszNiemiec, 2019-03-30:
@avinash-palleti @TSC21 what's the status of your work? I checked your branch @avinash-palleti, are you still working on it?

TSC21, 2019-03-31:
@ArkadiuszNiemiec this work was abandoned some time ago. For PX4 integration with Fast-RTPS and ROS2, the current ongoing work is being done in https://github.com/PX4/px4_ros_com.

closed 2019-03-31 by TSC21

ArkadiuszNiemiec, 2019-03-31:
@TSC21 Ok, thank you. What I am looking for is the Fast-RTPS to QGroundControl bridge and was hoping that you managed to write it in mavros. The alternative is to have MAVLink and Fast-RTPS on one port but I cannot find a way to do it.

TSC21, 2019-03-31:
> @TSC21 Ok, thank you. What I am looking for is the Fast-RTPS to QGroundControl bridge and was hoping that you managed to write it in mavros. The alternative is to have MAVLink and Fast-RTPS on one port but I cannot find a way to do it.

We are not aiming into supporting that currently. We are splitting the Fast-RTPS approach from the Mavlink interfaces.

ArkadiuszNiemiec, 2019-03-31:
Got it. Good work, thank you!