mavlink/mavros #770: ROS connection failure fallback

b-rohit, 2017-08-09:
### Question
I am using quadcopter for indoor navigation with help of mocap system using mavros. I have seen in a situation of ROS connection failure, drone fallback to` Land ` flight mode from `Offboard` and fly to `home` location. I have some questions regarding it.
1. How can I set `home`  location.
2. how do I change the default settings like how much altitude it gains in `Land` flight mode
3. Is it possible to `hold position ` instead of flying back to `home`

TSC21, 2017-08-10:
1. `rosrun mavros mavcmd sethome <lat> <lon> <alt>`.
2. A look at the Firmware code may help - you should probably move this question to the Firmware side.
3. Set `COM_OBL_ACT` parameter to 1 or `COM_OBL_RC_ACT` to 0 - again this is Firmware related.

closed 2017-08-10 by TSC21

TSC21, 2017-08-10:
Also on MAVROS we currently only support setting the home position LLA. You should add support for XYZ if you are just using the MOCAP system. The FCU will automatically set the home position the first time it gets a position lock.