mavlink/mavros #771: Feature: support ROS 2

TSC21, 2017-08-09:
@vooon maybe we should start thinking on creating a dev branch for this and sync up with Beta 3 release of ROS 2 in September 13th, 2017.
The migration guide seems explainable, so as soon as we get bandwidth we should put some effort on it - https://github.com/ros2/ros2/wiki/Migration-Guide.
This can also be synced with the integration of Fast-RTPS Publishers/Subscribers on MAVROS in order to support PX4 uORB msg translation from and to ROS.

vooon, 2017-08-10:
That was interesting, but we need to check what API already exists.
Because last version i checked is beta1, where too many things were not ready.

I start project for that task.

fnoop, 2017-12-01:
ROS2 is due to come out of beta next week, Dec 8th.

ArkadiuszNiemiec, 2018-07-09:
Any update on this subject? ROS2 have just released a second non-beta version.

TSC21, 2018-07-09:
Currently too much on my back to move forward with this. Maybe @vooon has some updates

CMcColl1658, 2019-04-27:
Has there been any progress on MAVROS for ROS 2?

LorenzMeier, 2019-04-27:
Since PX4 is like ROS a pub/sub architecture, we’ve built a native interface r for ROS2: https://dev.px4.io/en/middleware/micrortps.html

clancky1658, 2019-04-27:
Is this compatible with arducopter?

TSC21, 2019-04-27:
@clancky1658 what @LorenzMeier referenced, no it isn't. It's quite tide to the PX4 messaging architecture. Eventually there will be a port of MAVROS to ROS2, but not in my plans atm. Maybe @vooon can clarify the plan for this.

CMcColl1658, 2020-05-06:
I read somewhere about the reasons why mavros was not being migrated to ros2 but can't find it at the moment. does anyone have a link to such an article?

ArkadiuszNiemiec, 2020-05-06:
@CMcColl1658 probably because of:
> Since PX4 is like ROS a pub/sub architecture, we’ve built a native interface r for ROS2: https://dev.px4.io/en/middleware/micrortps.html

vooon, 2020-05-07:
No, that is different project. Mavros still be a mavlink bridge. Problem that i just do not have enough time, so process is slow.

ramilmsh, 2020-08-06:
> Since PX4 is like ROS a pub/sub architecture, we’ve built a native interface r for ROS2: https://dev.px4.io/en/middleware/micrortps.html

@LorenzMeier 
if I sound confused, it's because I am new at this. There is no support for Fast-RTPS on ardupilot, right?

TSC21, 2020-08-06:
@ramilmsh you have your answer to that in https://github.com/mavlink/mavros/issues/771#issuecomment-487277305. And as far as I known is still applicable.

fnoop, 2020-08-06:
PX4/RTPS interface has nothing to do with MAVROS and like MAVSDK doesn't support ArduPilot.  MAVROS is a great project, hope it will be ported to ROS2 eventually.

TSC21, 2020-08-06:
> PX4/RTPS interface has nothing to do with MAVROS and like MAVSDK doesn't support ArduPilot. MAVROS is a great project, hope it will be ported to ROS2 eventually.

@fnoop no one said it was.

ramilmsh, 2020-08-11:
In that case, @TSC21 , I have a different question. Does mavros need any modifications/improvements, or does it just need porting? Because, in that case, it seems like a straightforward task

TSC21, 2020-08-11:
> In that case, @TSC21 , I have a different question. Does mavros need any modifications/improvements, or does it just need porting? Because, in that case, it seems like a straightforward task

If it is a straightforward task for you, then you are always welcomed to add support for it. Any one can contribute and we welcome who wants to do that.

vooon, 2020-08-12:
@ramilmsh  ROS2 itself has a lot of differences so i wanted to redesign core parts. Unfortunately my free time is very limited now (oh, more that half of the year passed from announcement...).
But if you want, you can migrate current code. That already would be big step.

ramilmsh, 2020-08-18:
@vooon i want ros2 mostly for realtime and encryption, but I am as of yet, unconvinced that it is actually necessary, if I decide to make a port I will let you know

HomeLH, 2020-09-28:
Is there any plan to port to ROS2? ROS Foxy version is the LTS version.

ramilmsh, 2020-09-29:
@HomeLH i have not gotten to that part yet. But I will try to make a 1 to 1 port, with no optimizations over the next month or two. My free time is also limited

vooon, 2021-05-28:
Initial port done. It's still alpha and needs a lot of work, but i think it's time to say that people can start testing it, so i'm doing release.

closed 2021-05-28 by vooon