mavlink/mavros #774: roslaunch error with px4 state (mavlink version)

siddhu45, 2017-08-17:
Hello community,

_I am getting the error_
 `Client [/rostopic_5927_1502794795953] wants topic /mavros/state to have datatype/md5sum [mavros_msgs/State/9e3d873fae342c8f48a8bd64c53d991e], but our version has [mavros_msgs/State/63d9a29b6abd9e0beeba06109556d062]. Dropping connection.` 
_in roslaunch when I try to change the pixhawk mode or when I subscribe to `/mavros/state` from a different machine. I can see the topic on the same pc (on which roslaunch is run) but I cannot see any data and get this error when I try to do it on another pc of the ros network. I can still see other topics like `/mavros/local_position/pose` or `/mavros/imu/data` etc. but just when I change or want to see something related to state I get this error. One of my systems is running on Ubuntu 16.04 with ROS Kinetic and the other on Ubunut 14.04 with ros indigo._

> Note: This never happened before. I used to run simulations on exactly this configuration for a long time and this was never a problem.

After searching for this error and trying to debug I found out that this is the mavlink/mavros version error. The mavlink/mavros on my ubuntu 16.04 has somehow upgraded to the latest 0.19 version and that is causing the problem. I know this because I checked the plugins which are loaded when we do `roslauch mavros px4.launch` and they are different then previous. So, I want to make sure that is this the version problem?? and how can I solve it?

If you need any further info do let me know.

Any ideas would help. Thanks!

TSC21, 2017-08-17:
That's indeed a version problem. Follow src build instructions presented in https://github.com/mavlink/mavros/tree/master/mavros#installation on your 14.04 Indigo machine.

siddhu45, 2017-08-17:
So that will get my mavlink updated on the ubuntu 14.04 machine? Ok great. But why is this happening now? When did it change. I'll inform as soon as I try and make it work. Thanks a lot!                                                                                                                                                                                                                                                                                                                                        Sent from my BlackBerry 10 smartphone.                                                                                                                                                                                                                From: Nuno MarquesSent: Friday, 18 August, 2017 12:58 AMTo: mavlink/mavrosReply To: mavlink/mavrosCc: Siddharth Patel; AuthorSubject: Re: [mavlink/mavros] roslaunch error with px4 state (mavlink version) (#774)That's indeed a version problem. Follow src build instructions presented in https://github.com/mavlink/mavros/tree/master/mavros#installation on your 14.04 Indigo machine.

TSC21, 2017-08-17:
> But why is this happening now? When did it change.

Probably you issued a `apt dist-upgrade` or `apt-get upgrade` on your 16.04 machine that lead to upgrading it to MAVROS 0.19.

siddhu45, 2017-08-17:
Not sure because the 16.04 is a common machine and someone might have upgraded, no idea. Till 2 weeks back it was working properly but for the last 2 days I'm seeing this error. Anyways, let me do as you suggested and will let you know the results. Thanks again                                                                                                                                                                                                                                                                                                                                         Sent from my BlackBerry 10 smartphone.                                                                                                                                                                                                                From: Nuno MarquesSent: Friday, 18 August, 2017 1:07 AMTo: mavlink/mavrosReply To: mavlink/mavrosCc: Siddharth Patel; AuthorSubject: Re: [mavlink/mavros] roslaunch error with px4 state (mavlink version) (#774)
But why is this happening now? When did it change.

Probably you issued a apt dist-upgrade or apt-get upgrade on your 16.04 machine that lead to upgrading it to MAVROS 0.19.

siddhu45, 2017-08-18:
Also, how do I uninstall previous mavros version? Or I just install again and it will overwrite.                                                                                                                                                                                                                                                                                                                                         Sent from my BlackBerry 10 smartphone.                                                                                                                                                                                                                From: Nuno MarquesSent: Friday, 18 August, 2017 1:07 AMTo: mavlink/mavrosReply To: mavlink/mavrosCc: Siddharth Patel; AuthorSubject: Re: [mavlink/mavros] roslaunch error with px4 state (mavlink version) (#774)
But why is this happening now? When did it change.

Probably you issued a apt dist-upgrade or apt-get upgrade on your 16.04 machine that lead to upgrading it to MAVROS 0.19.

TSC21, 2017-08-18:
`apt-get remove ros-kinetic-mavros*`. It does not overwrite since this is not an deb package install. See how catkin ws work.

siddhu45, 2017-08-18:
Yes, I can do that. But what I meant was, on one of the device I did source installation of mavros previously. So how can I work around that? How do I update that. Thanks                                                                                                                                                                                                                                                                                                                                         Sent from my BlackBerry 10 smartphone.                                                                                                                                                                                                                From: Nuno MarquesSent: Friday, 18 August, 2017 8:41 AMTo: mavlink/mavrosReply To: mavlink/mavrosCc: Siddharth Patel; AuthorSubject: Re: [mavlink/mavros] roslaunch error with px4 state (mavlink version) (#774)apt-get remove ros-kinetic-mavros*. It does not overwrite since this is not an deb package install. See how catkin ws work.

TSC21, 2017-08-18:
Just follow the same procedures.

closed 2017-08-18 by TSC21

siddhu45, 2017-08-18:
@TSC21 Hey! thanks a lot. I just did mavros source installations and it worked. Great! Thanks a lot again.

siddhu45, 2017-08-20:
Hi, Actually when I try to do the source installation on my ubuntu 14.04  machine with ros indigo, the mavlink it takes latest but the mavros is still 0.17.5 and not 0.19 which I need. I also, tried to specify the rosdistro in the mavros command just like the mavlink, but it gives error when I build it. Previously, when I said I did it, it was on Raspberry Pi actually with ros-kinetic. Can you please help me out on this source installation. 

Thanks

siddhu45, 2017-08-20:
Actually, to install mavros 0.19 on Ubuntu 14.04 machine with ros indigo I had to modify the command line which selects mavros in the source installation
`rosinstall_generator --upstream --rosdistro kinetic mavros | tee -a /tmp/mavros.rosinstall`
and then it works. Thanks

fabrizioschiano, 2020-04-16:
Hi, I had the same problem recently and thanks to this issue I understood that it could have been a mavros version issue

Therefore I decided to do
`sudo apt-get remove ros-kinetic-mavros*`

Then I looked well in my system if I had other occurrences of mavros that were in the wrong place with
```
sudo updatedb
locate mavros
```
I found some and deleted them 

I then followed [these](https://dev.px4.io/v1.9.0/en/ros/mavros_installation.html) instructions carefully and did a `catkin clean` (the equivalent of `catkin_make clean` if you built your environment with `catkin_make` instead of `catkin build`)

I then rebuilt my `catkin_ws` with `catkin build` and tried again to launch the ROS node that was causing the error and it worked. 

I hope it will help someone who will have the same problem.

siddhu45, 2020-04-17:
Hi @fabrizioschiano 
Thanks for the tip. Literally, just yesterday I was working with a new system and had this issue. So, upgraded `mavros` on both systems as you suggested, cleaned and rebuilt the `catkin_ws`. 
It works fine now!
Cheers

fabrizioschiano, 2020-04-17:
@siddhu45 , I am glad my procedure helped you