mavlink/mavros #783: Sending custom message from Mavros to PX4

JoonmoAhn, 2017-08-21:
Hi, I'm trying to send my custom message from Mavros to PX4. Basically, I want to test it by sending a 'char' from ROS and subscribe from PX4 module. I followed "Source Installation" from this [mavros github site](https://github.com/mavlink/mavros/blob/master/mavros). And, with ROS Indigo,  after that, followings are what I did.

**Mavros Level :**
1. I added "keyboard_command.cpp" as mavros plugin (workspace/src/mavros/mavros/src/plugins).
keyboard_command.cpp looks as below.

```
#include <mavros/mavros_plugin.h>
#include <pluginlib/class_list_macros.h>
#include <iostream>
#include <std_msgs/Char.h>

namespace mavplugin {

class KeyboardCommandPlugin : public MavRosPlugin {
public:
    KeyboardCommandPlugin() :
        nh("~keyboard_command"),
        uas(nullptr)
    { };

    void initialize(UAS &uas_)
    {
        uas = &uas_;
        keyboard_sub = nh.subscribe("keyboard_sub", 10, &KeyboardCommandPlugin::keyboard_cb, this);
    };

    const message_map get_rx_handlers() {
        return {/* RX disabled */ };
    }

private:
    ros::NodeHandle nh;
    UAS *uas;
    ros::Subscriber keyboard_sub;

    void send_to_pixhawk(char cmd)
    {
        mavlink_message_t msg;
        mavlink_msg_key_command_pack_chan(UAS_PACK_CHAN(uas), &msg, cmd);
        UAS_FCU(uas)->send_message(&msg);
    }

    void keyboard_cb(const std_msgs::Char::ConstPtr &req)
    {
        std::cout << "Got Char : " << req->data <<  std::endl;
        send_to_pixhawk(req->data);
    }
};
};

PLUGINLIB_EXPORT_CLASS(mavplugin::KeyboardCommandPlugin, mavplugin::MavRosPlugin)
```

2. Edited "mavros_plugins.xml" as below
```
    <class name="keyboard_command" type="mavplugin::KeyboardCommandPlugin" base_class_type="mavplugin::MavRosPlugin">
        <description>Accepts Keyboard Command </description>
    </class>
```

3. Edited "CMakeLists.txt" to include library
```
add_library( 
  src/plugins/keyboard_command.cpp
)
```

4. At ROS Mavlink pacakge (workspace/src/mavlink/message_definitions/v1.0), I created "key_command.xml" as following.
```
<?xml version="1.0"?>
<mavlink>
    <include>common.xml</include>
    <enums>
    </enums>
    <messages>
        <message id="150" name="key_command">
            <description> mavlink message creating test </description>
            <field type="char" name="command"> </field>
        </message>
    </messages>
</mavlink>
```

5.  Edited mavlink_dialect.h at (mavros/libmavconn/include/mavconn) by inserting these two lines.
```
#define _DIALECT_key_command 13
...
#elif _DIALECT(MAVLINK_DIALECT) == _DIALECT_key_command
#  include <mavlink/v1.0/key_command/mavlink.h>
```

6. catkin build mavros --cmake-args -DMAVLINK_DIALECT="key_command"

-> After these 6 procedures, I was able to build, and able to "rostopic pub /mavros/keyboard_command/keyboard_sub std_msgs/Char 97"

**PX4 Level:**

1. Added "key_command.xml" at message_definitions (Firmware/mavlink/include/mavlink/v1.0/message_definitions) as below. It is same as procedure 4 in Mavros Level, except that there is no "<include>common.xml</include>" The reason why I omitted this line is because with this line, I couldn't generate header file with "mavgenerate.py", which will be following in procedure 2.

```
<?xml version="1.0"?>
<mavlink>
    <enums>
    </enums>
    <messages>
        <message id="150" name="key_command">
            <description> mavlink message creating test </description>
            <field type="char" name="command"> </field>
        </message>
    </messages>
</mavlink>
```

2. With "python mavgenerate.py" as instructed in this [link](http://qgroundcontrol.org/mavlink/generator), I created my header file at (/Firmware/mavlink/include/mavlink/v1.0/custom_messages/key_command) directory. And, in this directory, I edited "mavlink_msg_key_command.h" a little, so that it has same form as other mavlink messages. The change I made is as following.

Before : 
```
typedef struct MAVLINK_PACKED __mavlink_key_command_t
{
  char command; 
 } mavlink_key_command_t;
```

After :
```
MAVPACKED(
typedef struct __mavlink_key_command_t
{
  char command; 
}) mavlink_key_command_t;
```

3. I made a simple uORB message "key_command.msg" that it only has "char cmd" as its message. 

4. Edited mavlink_receiver.h in (Firmware/src/modules/mavlink) as follwing.
```
#include <uORB/topics/key_command.h>
#include <v1.0/custom_messages/key_command/mavlink_msg_key_command.h>
...
class MavlinkReceiver
{
...
private:
    void handle_message_key_command(mavlink_message_t *msg);
...
    orb_advert_t _key_command_pub;
}
```

5. Edited mavlink_receiver.cpp as following. So that I can receive Mavlink message and publish as ORB topic.

```
...
MavlinkReceiver::MavlinkReceiver(Mavlink *parent) :
...
    _key_command_pub(nullptr)
...
void MavlinkReceiver::handle_message(mavlink_message_t *msg)
{
...
    case MAVLINK_MSG_ID_key_command:
        handle_message_key_command(msg);
        break;
...
}
...
void
MavlinkReceiver::handle_message_key_command(mavlink_message_t *msg)
{
    mavlink_key_command_t man;
    mavlink_msg_key_command_decode(msg, &man);

    struct key_command_s key = {};

    key.timestamp = hrt_absolute_time();
    key.cmd = man.command;

    if (_key_command_pub == nullptr) {
        _key_command_pub = orb_advertise(ORB_ID(key_command), &key);

    } else {
        orb_publish(ORB_ID(key_command), _key_command_pub, &key);
    }
}
```

6. Made my uORB subscription module named "key_receiver". It just simply subscribes uORB topic "key_command", which is published from procedure 5. Code for "key_reciever.cpp" is as following.

```
#include <px4_config.h>
#include <px4_tasks.h>
#include <px4_posix.h>
#include <unistd.h>
#include <stdio.h>
#include <poll.h>
#include <string.h>
#include <math.h>

#include <uORB/uORB.h>
#include <uORB/topics/key_command.h>

extern "C" {__EXPORT int key_receiver_main(int argc, char **argv);}

int key_receiver_main(int argc, char **argv)
{
    int key_sub_fd = orb_subscribe(ORB_ID(key_command));
    orb_set_interval(key_sub_fd, 200); // limit the update rate to 200ms

    px4_pollfd_struct_t fds[1];
    fds[0].fd = key_sub_fd, fds[0].events = POLLIN;

    int error_counter = 0;

    while(1)
    {
        int poll_ret = px4_poll(fds, 1, 1000);

        if (poll_ret == 0)
        {
            PX4_ERR("Got no data within a second");
        }

        else if (poll_ret < 0)
        {
            if (error_counter < 10 || error_counter % 50 == 0)
            {
                PX4_ERR("ERROR return value from poll(): %d", poll_ret);
            }

            error_counter++;
        }

        else
        {
            if (fds[0].revents & POLLIN)
            {
                struct key_command_s input;
                orb_copy(ORB_ID(key_command), key_sub_fd, &input);
                PX4_INFO("Recieved Char : %c", input.cmd);
             }
        }
    }
    return 0;
}
```

-> After all these procedures, when I execute "key_receiver" module at nutshell, and "rostopic pub -r 10 /mavros/keyboard_command/keyboard_sub std_msgs/Char 97", the message from ROS doesn't get sent to uORB message. The output at nutshell is just "Got no data within a second".
It was a kind of long story, but I'm not sure what I have missed, or what I have done wrong.

TSC21, 2017-08-21:
@JoonmoAhn at the MAVROS level, you have to use `pymavlink` also to generate the message definitions. Just clone the ROS mavlink repository into your catkin ws. Then modify `common` xml file and add the message that you want  and replace the `common` message definitions file and build your ws. Don't do what you are doing at step 4 and 5. A dialect without the rest of the msgs is a unusable dialect.
PX4 level: you don't need to do step 2. if you generate a C++ header.

JoonmoAhn, 2017-08-22:
@TSC21 As you said, at MAVROS level, I deleted what I did on step 4 and 5, then added following four lines at (workspace/src/mavlink/message_definitions/v1.0/common.xml). 
```
        <message id="150" name="key_command">
            <description> mavlink message creating test </description>
            <field type="char" name="command"> </field>
        </message>
```

Then, when I "catkin build mavros", I have some errors like following.
```
make[2]: *** [include/v1.0/ardupilotmega/ardupilotmega.h] Error 1
make[1]: *** [CMakeFiles/ardupilotmega.xml-v1.0.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [include/v1.0/autoquad/autoquad.h] Error 1
make[1]: *** [CMakeFiles/autoquad.xml-v1.0.dir/all] Error 2
make[2]: *** [include/v2.0/ardupilotmega/ardupilotmega.h] Error 1
make[1]: *** [CMakeFiles/ardupilotmega.xml-v2.0.dir/all] Error 2
make: *** [all] Error 2
```

TSC21, 2017-08-22:
`cd catkin_ws`, `catkin clean --all -y` and then `catkin build`

JoonmoAhn, 2017-08-22:
@TSC21 Actually, I already did "catkin clean", and "catkin build" but still got that message...

TSC21, 2017-08-22:
Then please paste all the output. From where did you get the mavlink repo btw? From src install?

JoonmoAhn, 2017-08-22:
@TSC21 Yes, I got mavlink repo from "Source Installtaion" in this [mavros github site](https://github.com/mavlink/mavros/tree/master/mavros).
When, I "catkin build mavros", all my error outputs are as following.

```
joonmo@joonmo:~/odarros/workspace$ catkin build mavros
-------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/joonmo/catkin_ws/devel:/opt/ros/indigo
Workspace:                   /home/joonmo/odarros/workspace
-------------------------------------------------------------------------
Source Space:       [exists] /home/joonmo/odarros/workspace/src
Log Space:         [missing] /home/joonmo/odarros/workspace/logs
Build Space:        [exists] /home/joonmo/odarros/workspace/build
Devel Space:        [exists] /home/joonmo/odarros/workspace/devel
Install Space:      [unused] /home/joonmo/odarros/workspace/install
DESTDIR:            [unused] None
-------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------------------------
[build] Found '10' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 2.5 seconds ]             
Starting  >>> mavlink                                                          
Starting  >>> mavros_msgs                                                      
_______________________________________________________________________________
Errors     << mavlink:make /home/joonmo/odarros/workspace/logs/mavlink/build.make.000.log
make[2]: *** [include/v1.0/ardupilotmega/ardupilotmega.h] Error 1
make[1]: *** [CMakeFiles/ardupilotmega.xml-v1.0.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [include/v2.0/ardupilotmega/ardupilotmega.h] Error 1
make[1]: *** [CMakeFiles/ardupilotmega.xml-v2.0.dir/all] Error 2
make: *** [all] Error 2
cd /home/joonmo/odarros/workspace/build/mavlink; catkin build --get-env mavlink | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << mavlink:make                         [ Exited with code 2 ]      
Failed    <<< mavlink                              [ 9.2 seconds ]             
Abandoned <<< libmavconn                           [ Unrelated job failed ]    
Abandoned <<< mavros                               [ Unrelated job failed ]    
Finished  <<< mavros_msgs                          [ 9.7 seconds ]             
[build] Summary: 2 of 5 packages succeeded.                                    
[build]   Ignored:   6 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 2 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 12.3 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

TSC21, 2017-08-22:
What's the output of `catkin clean --all -y` when you execute it on `/home/joonmo/odarros/workspace/`? You should get something like:
```bash
[clean] Removing develspace:/home/joonmo/odarros/workspace/devel
[clean] Removing buildspace: /home/joonmo/odarros/workspace/build
[clean] Removing log space: /home/joonmo/odarros/workspace/logs
```

TSC21, 2017-08-22:
If that doesn't work out, try to git clone the mavros repo into your `workspace`, together with the latest ROS mavlink repo (https://github.com/mavlink/mavlink-gbp-release/tree/debian/lunar/mavlink). And then update the `common` msg definition as you did above and build again.

vooon, 2017-08-22:
@JoonmoAhn code you posted for outdated mavros, did you checkout for Indigo or Jade?
I recommend to use kinetic or lunar (master branch), just pass `--rosdistro=lunar` to both rosinstall_generator calls.

Second: better not to edit mavros sources, make your own package with plugin library (use extras as example).

Third: easier add messages into common.xml, because with custom dialect you may have trouble on FCU side (and you have).
And as i remember PX4 uses pregenerated git submodule ([c_library_v2](https://github.com/mavlink/c_library_v2)), so adding your own may be problematic. If you can, use one of existing messages.

And in catkin log i see that mavlink build fails, so check your xml changes.
You may:
```
cd ws/src/mavlink
catkin build --this --no-deps -vi
```

JoonmoAhn, 2017-08-22:
@vooon  I reinstalled MAVLINK and MAVROS by following procedures at my workspace. I'm using ROS INDIGO. But, I tried to do "--rosdistro=kinetic" to use later version of mavlink than Indigo.
```
1. rosinstall_generator --rosdistro kinetic mavlink | tee /tmp/mavros.rosinstall
2.  rosinstall_generator --upstream mavros | tee -a /tmp/mavros.rosinstall
3. wstool merge -t src /tmp/mavros.rosinstall
4. wstool update -t src -j4
5. rosdep install --from-paths src --ignore-src -y
```

Then, I did exactly same as my "MAVROS Level" procedures 1 ~ 3. Then, I included  below 4 lines in common.xml (workspace/src/mavlink/message_definitions/v1.0)

```
     <message id="150" name="key_command">
       <description> mavlink message creating test </description>
       <field type="char" name="command"> </field>
     </message>
```
After that, I "catkin build mavros", but still got error message as below.

```
joonmo@joonmo:~/odarros/workspace$ catkin build mavros
-------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/joonmo/catkin_ws/devel:/opt/ros/indigo
Workspace:                   /home/joonmo/odarros/workspace
-------------------------------------------------------------------------
Source Space:       [exists] /home/joonmo/odarros/workspace/src
Log Space:         [missing] /home/joonmo/odarros/workspace/logs
Build Space:        [exists] /home/joonmo/odarros/workspace/build
Devel Space:        [exists] /home/joonmo/odarros/workspace/devel
Install Space:      [unused] /home/joonmo/odarros/workspace/install
DESTDIR:            [unused] None
-------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------------------------
[build] Found '10' packages in 0.0 seconds.                                         
[build] Updating package table.                                                     
Starting  >>> catkin_tools_prebuild                                                 
Finished  <<< catkin_tools_prebuild                [ 1.1 seconds ]                  
Starting  >>> mavlink                                                               
Starting  >>> mavros_msgs                                                           
____________________________________________________________________________________
Errors     << mavlink:make /home/joonmo/odarros/workspace/logs/mavlink/build.make.000.log
WARNING: Unable to load XML validator libraries. XML validation will not be performed
WARNING: Unable to load XML validator libraries. XML validation will not be performed
WARNING: Unable to load XML validator libraries. XML validation will not be performed
make[2]: *** [include/v1.0/ardupilotmega/ardupilotmega.h] Error 1
make[1]: *** [CMakeFiles/ardupilotmega.xml-v1.0.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/joonmo/odarros/workspace/build/mavlink; catkin build --get-env mavlink | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
....................................................................................
Failed     << mavlink:make                         [ Exited with code 2 ]           
Failed    <<< mavlink                              [ 0.8 seconds ]                  
Abandoned <<< libmavconn                           [ Unrelated job failed ]         
Abandoned <<< mavros                               [ Unrelated job failed ]         
Finished  <<< mavros_msgs                          [ 5.4 seconds ]                  
[build] Summary: 2 of 5 packages succeeded.                                         
[build]   Ignored:   6 packages were skipped or are blacklisted.                    
[build]   Warnings:  None.                                                          
[build]   Abandoned: 2 packages were abandoned.                                     
[build]   Failed:    1 packages failed.                                             
[build] Runtime: 6.7 seconds total.                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

Is this because I'm using ROS INDIGO instead of KINETIC??

vooon, 2017-08-22:
No, problem in mavgen.py failure. Install `python-lxml` to run xml validator (warnings). And build --this.

JoonmoAhn, 2017-08-22:
@vooon I installed python-lxml by "sudo apt-get install python3-lxml". Then, as you said, I did 
```
cd ws/src/mavlink
catkin build --this --no-deps -vi
```

But, still can't get rid of errors..
```
[mavlink:make] [ 15%] Built target ASLUAV.xml-v2.0                                   
[mavlink:make] make[1]: Leaving directory `/home/joonmo/odarros/workspace/build/mavlink'
[mavlink:make] make: *** [all] Error 2                                               
Failed    <<< mavlink                              [ 0.7 seconds ]                  
[catkin_tools_prebuild:cmake] -- Using CATKIN_ENABLE_TESTING: ON                     
[catkin_tools_prebuild:cmake] -- Call enable_testing()                               
[catkin_tools_prebuild:cmake] -- Using CATKIN_TEST_RESULTS_DIR: /home/joonmo/odarros/workspace/build/catkin_tools_prebuild/test_results
[catkin_tools_prebuild:cmake] -- Looking for include file pthread.h                  
[catkin_tools_prebuild:cmake] -- Looking for include file pthread.h - found          
[catkin_tools_prebuild:cmake] -- Looking for pthread_create                          
[catkin_tools_prebuild:cmake] -- Looking for pthread_create - not found              
[catkin_tools_prebuild:cmake] -- Looking for pthread_create in pthreads              
[catkin_tools_prebuild:cmake] -- Looking for pthread_create in pthreads - not found  
[catkin_tools_prebuild:cmake] -- Looking for pthread_create in pthread               
[catkin_tools_prebuild:cmake] -- Looking for pthread_create in pthread - found       
[catkin_tools_prebuild:cmake] -- Found Threads: TRUE                                 
[catkin_tools_prebuild:cmake] -- Found gtest sources under '/usr/src/gtest': gtests will be built
[catkin_tools_prebuild:cmake] -- Using Python nosetests: /usr/bin/nosetests-2.7      
[catkin_tools_prebuild:cmake] -- catkin 0.6.19                                       
[catkin_tools_prebuild:cmake] -- Configuring done                                    
[catkin_tools_prebuild:cmake] -- Generating done                                     
[catkin_tools_prebuild:cmake] -- Build files have been written to: /home/joonmo/odarros/workspace/build/catkin_tools_prebuild
Starting   >> catkin_tools_prebuild:make                                            
[catkin_tools_prebuild:make] /usr/bin/cmake -H/home/joonmo/odarros/workspace/build/catkin_tools_prebuild -B/home/joonmo/odarros/workspace/build/catkin_tools_prebuild --check-build-system CMakeFiles/Makefile.cmake 0
Subprocess  > catkin_tools_prebuild:make `cd /home/joonmo/odarros/workspace/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -`
[catkin_tools_prebuild:make] /usr/bin/cmake -E cmake_progress_start /home/joonmo/odarros/workspace/build/catkin_tools_prebuild/CMakeFiles /home/joonmo/odarros/workspace/build/catkin_tools_prebuild/CMakeFiles/progress.marks
[catkin_tools_prebuild:make] /usr/bin/make -f CMakeFiles/Makefile2 all               
[catkin_tools_prebuild:make] make[1]: Entering directory `/home/joonmo/odarros/workspace/build/catkin_tools_prebuild'
[catkin_tools_prebuild:make] make[1]: Nothing to be done for `all'.                  
[catkin_tools_prebuild:make] make[1]: Leaving directory `/home/joonmo/odarros/workspace/build/catkin_tools_prebuild'
[catkin_tools_prebuild:make] /usr/bin/cmake -E cmake_progress_start /home/joonmo/odarros/workspace/build/catkin_tools_prebuild/CMakeFiles 0
Starting   >> catkin_tools_prebuild:symlink                                         
[catkin_tools_prebuild:symlink] Symlinking /home/joonmo/odarros/workspace/devel/./setup.zsh
[catkin_tools_prebuild:symlink] Symlinking /home/joonmo/odarros/workspace/devel/./setup.sh
[catkin_tools_prebuild:symlink] Symlinking /home/joonmo/odarros/workspace/devel/./_setup_util.py
[catkin_tools_prebuild:symlink] Symlinking /home/joonmo/odarros/workspace/devel/./setup.bash
[catkin_tools_prebuild:symlink] Symlinking /home/joonmo/odarros/workspace/devel/./env.sh
[catkin_tools_prebuild:symlink] Symlinking /home/joonmo/odarros/workspace/devel/share/catkin_tools_prebuild/cmake/catkin_tools_prebuildConfig.cmake
[catkin_tools_prebuild:symlink] Symlinking /home/joonmo/odarros/workspace/devel/share/catkin_tools_prebuild/cmake/catkin_tools_prebuildConfig-version.cmake
[catkin_tools_prebuild:symlink] Symlinking /home/joonmo/odarros/workspace/devel/lib/pkgconfig/catkin_tools_prebuild.pc
Finished  <<< catkin_tools_prebuild                [ 1.2 seconds ]                  
[build] Summary: 1 of 2 packages succeeded.                                         
[build]   Ignored:   9 packages were skipped or are blacklisted.                    
[build]   Warnings:  None.                                                          
[build]   Abandoned: None.                                                          
[build]   Failed:    1 packages failed.                                             
[build] Runtime: 1.2 seconds total.                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

vooon, 2017-08-22:
I only need `[mavlink:*]` lines. Go to mavlink sources and run `catkin build --this --no-deps -vi -j1`

JoonmoAhn, 2017-08-22:
@vooon I did "catkin clean --all -y" at workspace to clear all. Then, at src/mavlink, I ran "catkin build --this --no-deps -vi -j1" as you said. Then, the error I got is as below.

```
...
Starting  >>> mavlink                                                                                                                                                    
Starting   >> mavlink:loadenv                                                                                                                                            
[mavlink:loadenv] Loading environment from: /home/joonmo/odarros/workspace/devel/env.sh                                                                                   
Starting   >> mavlink:mkdir                                                                                                                                              
Starting   >> mavlink:mkdir                                                                                                                                              
Starting   >> mavlink:cache-manifest                                                                                                                                     
Starting   >> mavlink:cmake                                                                                                                                              
Subprocess  > mavlink:cmake `cd /home/joonmo/odarros/workspace/build/mavlink; catkin build --get-env mavlink | catkin env -si  /usr/bin/cmake /home/joonmo/odarros/workspace/src/mavlink --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX=/home/joonmo/odarros/workspace/devel; cd -`
[mavlink:cmake] Not searching for unused variables given on the command line.                                                                                             
[mavlink:cmake] Re-run cmake no build system arguments                                                                                                                    
[mavlink:cmake] -- Found PythonInterp: /usr/bin/python (found suitable version "2.7.6", minimum required is "2")                                                          
[mavlink:cmake] -- Package version: 2017.8.8                                                                                                                              
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/ASLUAV.xml                                                        
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/ardupilotmega.xml                                                 
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/autoquad.xml                                                      
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/common.xml                                                        
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/matrixpilot.xml                                                   
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/paparazzi.xml                                                     
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/slugs.xml                                                         
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/standard.xml                                                      
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/uAvionix.xml                                                      
[mavlink:cmake] -- processing v1.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/ualberta.xml                                                      
[mavlink:cmake] -- processing v2.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/ASLUAV.xml                                                        
[mavlink:cmake] -- processing v2.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/ardupilotmega.xml                                                 
[mavlink:cmake] -- processing v2.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/autoquad.xml                                                      
[mavlink:cmake] -- processing v2.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/common.xml                                                        
[mavlink:cmake] -- processing v2.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/matrixpilot.xml                                                   
[mavlink:cmake] -- processing v2.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/paparazzi.xml                                                     
[mavlink:cmake] -- processing v2.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/slugs.xml                                                         
[mavlink:cmake] -- processing v2.0: /home/joonmo/odarros/workspace/src/mavlink/message_definitions/v1.0/standard.xml                                                      
[... cut ...]
[build]   Abandoned: None.                                                                                                                                               
[build]   Failed:    1 packages failed.                                                                                                                                  
[build] Runtime: 2.6 seconds total.                                                                                                                                      
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

TSC21, 2017-08-22:
You are usins the same msg ID (150) `SENSOR_OFFSETS` (it's readable you can see it there). Use another msg ID that's different from any of those you can find on the xml

JoonmoAhn, 2017-08-22:
@TSC21 Oh, you were right. I changed ID to 229, then build worked! But, still, the message is not sent to PX4.. For reminder, this is so far what I have done.

**MAVROS LEVEL :** 
1. I added "keyboard_command.cpp" as mavros plugin (workspace/src/mavros/mavros/src/plugins).
keyboard_command.cpp looks as below.

```
#include <mavros/mavros_plugin.h>
#include <pluginlib/class_list_macros.h>
#include <iostream>
#include <std_msgs/Char.h>

namespace mavplugin {

class KeyboardCommandPlugin : public MavRosPlugin {
public:
    KeyboardCommandPlugin() :
        nh("~keyboard_command"),
        uas(nullptr)
    { };

    void initialize(UAS &uas_)
    {
        uas = &uas_;
        keyboard_sub = nh.subscribe("keyboard_sub", 10, &KeyboardCommandPlugin::keyboard_cb, this);
    };

    const message_map get_rx_handlers() {
        return {/* RX disabled */ };
    }

private:
    ros::NodeHandle nh;
    UAS *uas;
    ros::Subscriber keyboard_sub;

    void send_to_pixhawk(char cmd)
    {
        mavlink_message_t msg;
        mavlink_msg_key_command_pack_chan(UAS_PACK_CHAN(uas), &msg, cmd);
        UAS_FCU(uas)->send_message(&msg);
    }

    void keyboard_cb(const std_msgs::Char::ConstPtr &req)
    {
        std::cout << "Got Char : " << req->data <<  std::endl;
        send_to_pixhawk(req->data);
    }
};
};

PLUGINLIB_EXPORT_CLASS(mavplugin::KeyboardCommandPlugin, mavplugin::MavRosPlugin)
```

2. Edited "mavros_plugins.xml" , and added 4 following lines. (workspace/src/mavros/mavros)

   ```
 <class name="keyboard_command" type="mavplugin::KeyboardCommandPlugin" base_class_type="mavplugin::MavRosPlugin">
        <description>Accepts Keyboard Command </description>
    </class>
```

3. Edited "CMakeLists.txt" to include library (workspace/src/mavros/mavros)

```
add_library( 
  src/plugins/keyboard_command.cpp 
)
```

[... 149 lines cut ...]
--> After all these procedures in MAVROS, and PX4 LEVELs, I was able to build without error. Thanks to @TSC21 , and @vooon . But, I'm back to original problem. When I run mavros, and px4 as following, 
<MAVROS at ROS>
```
1. roslaunch mavros px4.launch
2. rostopic pub -r 10 /mavros/keyboard_command/keyboard_sub std_msgs/Char 97
```

<PX4>
```
1. cd Firmware/Tools
2. ./mavlink_shell.py (my IP address):14557 --baudrate 57600
3. nsh>key_receiver
```

I still have outputs in px4 nutshell like,
```
...
ERROR [key_receiver] Got no data within a second
ERROR [key_receiver] Got no data within a second
ERROR [key_receiver] Got no data within a second
...
```

I guess message is still not properly transferred from MAVROS to PX4...
Am I still missing something??

JoonmoAhn, 2017-08-24:
Solved the problem. If anyone who has similar issue, I hope this [manual](https://github.com/JoonmoAhn/Sending-Custom-Message-from-MAVROS-to-PX4/issues/1) I wrote be some help

TSC21, 2017-08-24:
> Solved the problem. If anyone who has similar issue, I hope this manual I wrote would be some help. Thank you, @TSC21 @vooon

Good that this is solved to you -> what I would ask you to do is that you propose that tutorial entry in the PX4 [Devguide](https://github.com/PX4/Devguide). Can you please do that? You can integrate it on https://dev.px4.io/en/tutorials/tutorial_hello_sky.html, adding extra lines about the MAVROS steps and the Mavlink PX4 module updates for receiving/transmitting the msg.

closed 2017-08-24 by TSC21

rk800506, 2020-10-09:
Hello @vooon @TSC21, please have a look at the thread [https://discuss.px4.io/t/error-while-data-type-conversion-from-unsigned-char-to-const-mavlink-aka-mavlink-message-type/18951?u=ranjeet](url) , any suggestion regarding the issue faced?