mavlink/mavros #785: [mavros_extras] Streaming video to QGC

makquel, 2017-08-22:
My camera driver is publishing a topic called /drone/image_raw, and the following is the code that I used on my launch file in order to send it over mavlink to the QGC

  <node pkg="mavros_extras" type="gcs_image_bridge" name="gcs_image_bridge" clear_params="true" output="screen">
        <remap from="/mavlink/gcs_image/compressed" to="/drone/image_raw/compressed" />
        <param name="image_transport" value="compressed"/>

        <param name="gcs_url" value="tcp://127.0.0.1:5760"/>
        <param name="gcs_url" value="tcp-l://127.0.0.1:5760"/>

  </node>

However, I can't see the video streaming on the GCS widget for video, instead, I only see the image on the Pix4Flow tab, under vehicle setup menu.

makquel, 2017-08-22:
Re-posting my launch file code: 
  <node pkg="mavros_extras" type="gcs_image_bridge" name="gcs_image_bridge" clear_params="true" output="screen">
        <remap from="/mavlink/gcs_image/compressed" to="/drone/image_raw/compressed" />
        <param name="image_transport" value="compressed"/>

        <param name="gcs_url" value="tcp://127.0.0.1:5760"/>
        <param name="gcs_url" value="tcp-l://127.0.0.1:5760"/>

  </node>

arwagoner, 2017-08-22:
You can follow this guide to make sure you have gstreamer installed. The gst_video_server is a useful tool when using MAVROS. https://github.com/ProjectArtemis/gst_video_server 
Clone the repo into a catkin work space and compile it. The parameters can be changed in the qgc_streamer.launch file to suit your application. Be sure that you are using the most up-to-date version of QGC. You should see a menu under general settings where you can specify which port QGC should listen to for the video stream. That menu looks like this: 
[image]

makquel, 2017-08-22:
Thanks @arwagoner , should I use a different UDP port for video streaming that the one I use to stablish communication for telemetry?

arwagoner, 2017-08-22:
Yes, it should be a different port.

TSC21, 2017-08-23:
Thanks @arwagoner. @makquel is this solved to you already?

makquel, 2017-08-23:
@TSC21 I couldn't solve it! Even using the latests version of the QGroundControl. Something I don't get is why I can see the image (video feed) at the Pix4Flow tab and can't on the video streaming window.

TSC21, 2017-08-23:
@makquel this is something I have not tested so I can't help you. But what @arwagoner shared is basically a nodelet that works and streams to QGC. You should also be careful with the type of encoding and image.

makquel, 2017-08-23:
@arwagoner Do you have a MWE of the gst_video_server?, because It didn't work for me!
@TSC21 As far as I now, the gst_video_server is the video encoder (.h264) itself with upd transmission capabilities.

TSC21, 2017-08-23:
> @TSC21 As far as I now, the gst_video_server is the video encoder (.h264) itself with upd transmission capabilities.

True. I was saying in the case you pretend to change the encoding.

TSC21, 2017-08-23:
@mhkabir maybe you can lend a hand here?

arwagoner, 2017-08-23:
@makquel Can you please be more specific about what issues you are encountering? Are you getting any errors from the gst_video_server? It would also help to know more about your set up. Thanks

vooon, 2017-08-23:
@makquel did you configured remap on your image topic? Did you change gst pipeline to setd stream on your computer?

vooon, 2017-08-23:
Also two `gcs_url` looks like configuration error.

makquel, 2017-08-23:
@arwagoner I just remapped the image topic to the actual image topic that my image processing nodelet (On the gst_video_server launch file). Then, on the QGC set the proper UDP port as the gst_video_server has. I didn’t get any errors from it, however, I couldn’t see any image related object on the MAVLink inspector neither video streaming image in the QGC.
My setup is the following:
[Sensors nodes (including sub camera)] <---> [sensor topics to mavlink ] <----> [GQC v3.1]

makquel, 2017-08-23:
@vooon I don't quite understand what "change gst pipeline to setd stream" means, could you explain me better?
I did remap the image topic...and the nodelet was launched without errors...let me show you terminal output:

[ INFO] [1503494582.992117219]: Starting gst_video_server instance: /gst_video_server
[ INFO] [1503494583.006509001]: Gstreamer version: GStreamer 1.2.4
[ INFO] [1503494583.006554435]: Pipeline: videorate ! video/x-raw,framerate=30/1 ! videoconvert ! x264enc speed-preset=ultrafast tune=zerolatency byte-stream=true threads=1 key-int-max=15 intra-refresh=true ! video/x-h264, profile=baseline ! rtph264pay ! udpsink host=127.0.0.1 port=5600
[ INFO] [1503494583.016465718]: GST: pipeline paused.
[ INFO] [1503494583.204307919]: GST: appsrc caps: video/x-raw, format=(string)GRAY8, width=(int)1600, height=(int)1200, framerate=(fraction)0/1
[ INFO] [1503494583.222807438]: GST: pipeline changing state...

makquel, 2017-08-23:
@vooon I'm setting communication with QGC via gcs_image_bridge for all telemetry sensors using the gcs_url parameter

makquel, 2017-08-23:
@vooon  "Did you change gst pipeline to setd stream on your computer?" I did, since for testing purposes I'm streaming to my FCU itself i.e. udpsink host=127.0.0.1 port=5600

vooon, 2017-08-23:
Use rqt_graph to check topic connections, wireshark to check RTP stream on port 5000.

makquel, 2017-08-23:
@vooon like you said! The topic connections seem to be right 
[image]
And I also capture traffic of the udp port (udp.port == 5600) and could see anything coming in/out....seems like the gst_video_server nodelet is not working at all

makquel, 2017-08-23:
@arwagoner what were the setting you used to stream video using the gst_video_server nodelet? Image size, fps, and so on?

vooon, 2017-08-23:
Check that topic actually published, use rqt again... Also enable debug log via rqt.

makquel, 2017-08-23:
I might be wrong but I think that the gst_video_server doesn't publish any topic. It only subscribes from a image topic

arwagoner, 2017-08-23:
The gst_image_server is working correctly if that is the output you are getting. Last I checked, the gcu_image_bridge was removed from mavros_extras. It seems likely that you may be using outdated software. The gst_video_server interacts with the newest version of Qgc. It may not be compatible with the version of mavros/qgc that you have.

TSC21, 2017-08-24:
> @vooon I'm setting communication with QGC via gcs_image_bridge for all telemetry sensors using the gcs_url parameter, like you pointed out there was a config error, since the launch file interpreter only gets the last value.

Confirmed @arwagoner! @makquel what version are you using? Can you please use latest kinetic/lunar version or pull the current upstream master?

makquel, 2017-08-24:
@TSC21 I'm using ROS indigo (Ubuntu 14.04). Right now could be kind of difficult to move/upgrade to a latest version.

TSC21, 2017-08-24:
Why's difficult? You have src install on the instructions. You juat have to so it on your catkin ws. Without the kinetic/upstream version you won't be able to do what you pretend about QGC

TSC21, 2017-08-24:
That's why we have an issue template for people to follow. If you don't follow the issue template it gets more difficult for us maintainers understand the setup and help om implementing or fixing the problem. The detail of you using the indigo version makes all the difference and should have been clarified on the issue description.

makquel, 2017-08-24:
@TSC21 I've got a proprietary driver for the Xsens MTi-G-710, and the last time a tried to compile it using Kinetic (Ubuntu 16.04) I had many issues regarding Xsens's API. I'll try upgrade my ros version and I'll get back to you.

TSC21, 2017-08-24:
You don't have to upgrade your ros version. You kist have to follow the instructions so to install/compile from src

TSC21, 2017-08-29:
@makquel any luck?

TSC21, 2017-09-01:
Closing this now. @makquel let us know if updating the ROS version helped (though not required), and if not we will reopen the issue.

closed 2017-09-01 by TSC21

makquel, 2017-09-01:
@TSC21 I had to move on and solve other issues in my current project, I'll get back later on, hopefully updating ROS version will solve it.

TSC21, 2017-09-01:
@makquel instead of updating ROS, why don't you simply follow https://github.com/mavlink/mavros/tree/master/mavros#source-installation?

makquel, 2017-09-04:
@TSC21 thanks, I'll let you know if it works.