mavlink/mavros #786: "/mavros/vision_pose/pose" not adjusting x,y value in  "/mavros/local_position/pose"

JoonmoAhn, 2017-08-23:
Hi, I'm trying to publish position topic through "/mavros/vision_pose/pose" topic. And, see how "/mavros/local_position/pose" converges to pose by subscribing the topic.
I'm using "PX4 Firmware version 1.6.2dev".

Beforehand, I changed two EKF2 paramters through QgroundControl v3.1.3 as following:
```
EKF2_AID_MASK : 8 (vision position fusion)
EKF2_HGT_MODE : VIsion
```

When I publish my custom pose through "/mavros/vision_pose/pose", 'z' value converges very well. But, 'x and y' value doesn't converge and stays at 0. Following code is ROS node that publishes my custom position (x=1, y=2, z=300) through "/mavros/vision_pose/pose" topic, and subscribes x,y,z value of "/mavros/local_position/pose" topic.   

```
#include <ros/ros.h>
#include <iostream>

#include <geometry_msgs/PoseStamped.h>

int loopRate = 500;
geometry_msgs::PoseStamped msg_vicon_pose;

void Callback_fcu_pose(const geometry_msgs::PoseStamped msg_fcu_pose_dummy)
{
    ROS_INFO("Got data : %f, %f, %f", msg_fcu_pose_dummy.pose.position.x, msg_fcu_pose_dummy.pose.position.y, msg_fcu_pose_dummy.pose.position.z);
}

int main(int argc, char **argv)
{
    ros::init(argc, argv, "pub_sub");
    ros::NodeHandle n;

    ros::Publisher vicon_pub = n.advertise<geometry_msgs::PoseStamped>("/mavros/vision_pose/pose",loopRate);
    ros::Subscriber vicon_sub = n.subscribe("/mavros/local_position/pose",loopRate, Callback_fcu_pose);

    ros::Rate loop_rate(loopRate);

    while (ros::ok())
    {
        loop_rate.sleep();
        msg_vicon_pose.header.stamp = ros::Time::now();
        msg_vicon_pose.header.frame_id = "fcu"; //optional. Works fine without frame_id
        msg_vicon_pose.pose.position.x = 1;
        msg_vicon_pose.pose.position.y = 2;
        msg_vicon_pose.pose.position.z = 300;

        vicon_pub.publish(msg_vicon_pose);

        ros::spinOnce();
    }
    return 0;
}
```

The output in few seconds is as following.
```
[ INFO] [1503462134.449825006]: Got data : -0.000000, 0.000000, 300.099335
[ INFO] [1503462134.485823060]: Got data : -0.000000, 0.000000, 300.098816
[ INFO] [1503462134.519832517]: Got data : -0.000000, 0.000000, 300.097778
```
'z value' converges to '300' as I intended, but 'x value' and 'y value' don't converge to 1, and 2. But stays at 0. What should be done wrong?

TSC21, 2017-08-23:
Can you please try LPE instead and check the results?

JoonmoAhn, 2017-08-23:
@TSC21 I built with "make nuttx_px4fmu-v2_lpe" , but still didn't work.

TSC21, 2017-08-23:
What's your autopilot?

JoonmoAhn, 2017-08-23:
I use Pixhawk, and firmware PX4.

TSC21, 2017-08-23:
use `make px4fmu-v2_lpe` instead and load it to your FCU.

JoonmoAhn, 2017-08-24:
Still, the same.. when I rosrun my vision pose publisher node, I get message like 
```
[ INFO] [1503537011.779395447]: FCU: [lpe] vision position init:  2.00  1.00 -300.00 m
[ INFO] [1503537011.830472447]: FCU: [lpe] vision position fault, beta 354300.56

But, still , when I "rostopic echo /mavros/local_position/pose", not converging to what I published through "/mavros/vision_pose/pose"
```

TSC21, 2017-08-24:
> Still, the same.. when I rosrun my vision pose publisher node, I get message like 
```
[ INFO] [1503537011.779395447]: FCU: [lpe] vision position init:  2.00  1.00 -300.00 m
[ INFO] [1503537011.830472447]: FCU: [lpe] vision position fault, beta 354300.56
```
But, still , when I "rostopic echo /mavros/local_position/pose", not converging to what I published through "/mavros/vision_pose/pose"

It is launching the fault detection. It means that the vision estimation you are feeding to the estimator has a lot of interference and jumps and probably it is not consistent.

TSC21, 2017-08-24:
You may be facing this problem, as you are always sending the same value: https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/vision_pose_estimate.cpp#L120-L123. Or you remove that piece (which doesn't make sense, case that piece of code is there to evaluate if the vision estimator has a fault or freezed and is just sending the same estimate over and over - which is what your code is currently demonstrating) or you simulate a vision estimate generator with some random noise in a range - what I don't understand is why you are no using a real vision estimate fee.
Also you won't be able to publish the data to the Mavlink stream at that rate. Reduce the rate to 30hz for example, wich is more than enough.

TSC21, 2017-08-29:
@JoonmoAhn updates?

JoonmoAhn, 2017-08-30:
@TSC21 Acutally, tried deleteing the part you mentioned, but still the problem is not solved... So I was just working on other things, and leaving this problem for later time.

TSC21, 2017-08-30:
> @TSC21 Acutally, tried deleteing the part you mentioned, but still the problem is not solved... So I was just working on other things, and leaving this problem for later time.

Try then generating a variable output instead.

TobiMiller, 2017-08-31:
similar problem here with px4 1.6.5 stable. when i use master it works.

JoonmoAhn, 2017-09-01:
@TobiMiller Oh. Thank you, it worked as you said!!

TSC21, 2017-09-01:
You were using a version different from master this all time? Oh well...

closed 2017-09-01 by TSC21