mavlink/mavros #787: Local_position Topics Not Publishing

hartmac4, 2017-08-23:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

So, forgive me in advance. I am pretty new to the world of autopilots, so the problem may be pretty simple.

I am using a Pixhawk with PX4 firmware and am trying to read information from the local_position topics via ROS. px4_chase.launch is the same as px4.launch but with the FCU_URL modified based on what I need at the time.

I have been having this problem where I am unable to get any information from /mavros/local_position/pose or /mavros/local_position/velocity in certain situations. I can see messages come through these topics if I have my computer connected to the Pixhawk via its USB ports, but when I try using the telemetry ports, the frequency that I get information varies. More specifically, I am getting local_position information at a frequency of 30 hertz through the USB, 0.5 hertz through telemetry 1, and virtually nothing through telemetry 2. This problem persists whether I use a Pixhawk 1 or 2, no matter what cord I use, and no matter which computer I use. In addition, when using either of the telemetry ports, I get the "Unexpected Command 520" warning and the "Request param timeout" warning, neither of which I get when plugged in through USB.

Anyways, I know I can get this information through Gazebo as well, and I have updated my firmware to the latest versions. I know I can get other information (like imu data) through telemetry as well. I'm a bit lost right now, so if anyone can help me out on this, that would be great.

As one last aside, I don't get any local_position information whether I have a GPS lock or not.

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[*] PX4

Version: 1.6.5

### Node logs

```
risc@risc-pc:~/chase_ws$ roslaunch offboard px4_chase.launch 
... logging to /home/risc/.ros/log/1abfe0bc-8844-11e7-81d6-74da38aefede/roslaunch-risc-pc-19189.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://risc-pc:42667/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sens_Please describe the problem, or desired feature_or_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
[... 446 lines cut ...]
        key: Mean dt (ms)
        value: -0.086534
      - 
        key: Last system time (s)
        value: 419.502178000
      - 
        key: Time offset (s)
        value: 1503521161.133041620
---

```

### Check ID

```
^Crisc@risc-pc:~/chase_ws$ rosrun mavros checkid
WARNING: /mavros/target_system_id not set. Used default value: 1
WARNING: /mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages
```

TSC21, 2017-08-23:
You probably have issues on your telem link. Have you tried another set of radios?

hartmac4, 2017-08-24:
I actually have a cord that connects the telemetry port to one of the Odroid's/computer's USB ports. It's one of the telemetry cords that comes with the Pixhawk, with one end replaced with soldered connections to a FTDI Basic, which also connects to the computer. I have tried replacing what I could here to make sure it wasn't one of these components (minus the FTDI), but I am getting the same warnings regardless.

TSC21, 2017-08-24:
Then, set `SYS_COMPANION` PX4 parameter to 921600. You can use QGC or a nsh shell by typing `param set SYS_COMPANION 921600`.

hartmac4, 2017-08-24:
When I try that, I don't get the errors, but none of the data comes through (checking the state tells me that it isn't connected anymore).

TSC21, 2017-08-24:
I recommend you to get something like this: https://www.adafruit.com/product/954. Just to be sure it is not a problem on your hack or on the FTDI chip.

TSC21, 2017-08-24:
Before that, did you also change the launch configuration? Did you set /dev/ttyUSB0 @ 921600? Can you show me the output of `rostopic echo -n1 /diagnostics`?

hartmac4, 2017-08-25:
Hey, thank you so much! It was working when I came in this morning. I think I just needed to restart my computer after changing the baud rate. Sorry about any problems!

closed 2017-08-25 by TSC21