mavlink/mavros #788: Difference between setpoint_velocity and setpoint_raw

L4ncelot, 2017-08-24:
Hi,

I'm trying control my copter with px4 firmware in it using velocity control through MAVROS. But I also want to change headings during flight. I think the `/setpoint_raw/local` topic is the right one for this, however it's not behaving like the `/setpoint_velocity/cmd_vel`. With `/setpoint_velocity/cmd_vel`, the copter is flying like it should, but not with `/setpoint_raw/local`.

I think it's because both topics use different frame. I tried all 4 frames available in `mavros_msgs/PositionTarget` but none of them behaves like with `/setpoint_velocity/cmd_vel` topic.

I set my frame and type_mask like this:

```
position_target_.header.frame_id = mavros_msgs::PositionTarget::FRAME_BODY_NED;
position_target_.type_mask =
                    mavros_msgs::PositionTarget::IGNORE_AFX |
                    mavros_msgs::PositionTarget::IGNORE_AFY |
                    mavros_msgs::PositionTarget::IGNORE_AFZ |
                    mavros_msgs::PositionTarget::IGNORE_PX |
                    mavros_msgs::PositionTarget::IGNORE_PY |
                    mavros_msgs::PositionTarget::IGNORE_PZ |
                    mavros_msgs::PositionTarget::IGNORE_YAW |
                    mavros_msgs::PositionTarget::IGNORE_YAW_RATE;
```

Is this the right behavior or am I missing something? Is there a way to control velocities with heading using `/setpoint_raw/local` topic so it behaves like with `/setpoint_velocity/cmd_vel` topic?

AlexisTM, 2017-08-25:
`setpoint_velocity/cmd_vel` is a `TwistStamped` with linear and angular velocities.

The `type_mask` is `uint16_t ignore_all_except_v_xyz_yr = (1 << 10) | (7 << 6) | (7 << 0);` and The `frame` is `MAV_FRAME::LOCAL_NED`

TSC21, 2017-08-25:
@L4ncelot if you ignore yaw and yaw_rate, how do you expect to control the heading?

L4ncelot, 2017-08-28:
@AlexisTM well I thought that the cmd_vel velocity is the same as velocity in position target. I guess it's not. 

@TSC21 I know the yaw won't work if I ignore the yaw. I was just trying out if both topics work the same when the yaw is zero.

TSC21, 2017-08-28:
For `/setpoint_velocity/cmd_vel`, you control the `yaw_rate` only. It's by default [setting it to `LOCAL_NED`](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_velocity.cpp#L81). We should make it more general in the mean time through a parameter. `BODY_NED` is also [supported on the PX4 Firmware].(https://github.com/PX4/Firmware/commit/e01eaf172a50727ba241513303df0212b83e3558).
If you set:
```
                    mavros_msgs::PositionTarget::IGNORE_YAW |
                    mavros_msgs::PositionTarget::IGNORE_YAW_RATE;
```
You are ignoring both `yaw` and `yaw_rate`, which means you won't be able to control yaw using `setpoint_raw`.

L4ncelot, 2017-08-28:
Oh, now when I set the frame to `LOCAL_NED` and type mask to `(1 << 10) | (7 << 6) | (7 << 0)` it's working the same as `cmd_vel` topic. Thank you for clarification.

closed 2017-08-28 by L4ncelot

AlexisTM, 2017-08-28:
You're welcome

Jaroan, 2019-12-11:
How to set body frame and type_mask in Python for px4 firmware using velocity control through MAVROS. I'm using setpoint_velocity/cmd_vel topic .