mavlink/mavros #789: Consistency problem in the TF frames (in px4_config.yaml)

AlexisTM, 2017-08-25:
### Issue details

The default TF frames are not consistent in the configuration file.

Within the file, either `local_origin` and `map` are used for the same frame, except for local_position and global_position.

Also the `global_position/frame_id` and `local_position/frame_id` shouldn't point to fcu.

https://github.com/mavlink/mavros/blob/d283fb0a3f094caa4282922351fb965f69f9c898/mavros/launch/px4_config.yaml#L45-L55

* `global_position/frame_id` should be `earth` (**bug**)
* `global_position/tf/frame_id` should be `map` (**consistency**)

https://github.com/mavlink/mavros/blob/d283fb0a3f094caa4282922351fb965f69f9c898/mavros/launch/px4_config.yaml#L65-L72

* `local_position/frame_id` should be `map` (**bug**)
* `local_position/tf/frame_id` should be `map` (**consistency**)

TSC21, 2017-08-25:
@AlexisTM thanks for reviewing this out. This was something to be done some time ago but wasn't done yet.

> Within the file, either `local_origin` and `map` are used for the same frame, except for local_position and global_position.

Should all be replaced by `map` as `local_origin` is not a frame of reference of MAVROS.

> * `global_position/frame_id` should be `earth` (**bug**)

Should be `map` also. `earth` is for the global frame, though the pose published by the plugin is relative to the `map` frame.

TSC21, 2017-08-25:
Would you mind applying a PR with the above corrections? Thanks

TSC21, 2017-08-31:
Fixed with #791

closed 2017-08-31 by TSC21