mavlink/mavros #792: Is there a topic like /mavros/setpoint_velocity/cmd_vel but use body frame?

cnpcshangbo, 2017-08-25:
Hi,
I am doing vision-based control and the camera is on the drone facing down.
So the position estimation is in body frame. 
I found the topic /mavros/setpoint_velocity/cmd_vel can be used to change the drone's speed but it seems it is in NED frame. 

Is there a topic like /mavros/setpoint_velocity/cmd_vel but use body frame?

Or I need to convert my speed setpoint from body frame to NED frame to implement it?

I tried to find something that I can use and I found this code. https://github.com/weiweikong/px4_velocity_control_keyboard/blob/master/src/px4_velocity_control_node.cpp
```cpp
vs_body_axis.twist.linear.x = vs.twist.linear.x * cos(uavYawENU) + vs.twist.linear.y * sin(uavYawENU);
vs_body_axis.twist.linear.y = vs.twist.linear.x * sin(uavYawENU) - vs.twist.linear.y * cos(uavYawENU);
```
It seems this is the best solution I could find.
I am not sure if the Yaw if from IMU only or from fusion of IMU and magnetic. 
Do you have some comments or ideas about this?

### MAVROS version and platform

Mavros: 0.17.3
ROS: Indigo
Ubuntu: 14.04

### Autopilot type and version

APM:Copter V3.4-dev (fe724032)

### Node logs

bshang2@mechatronics-990:~/simulation/ardupilot/ArduCopter$ roslaunch ardupilot_sitl_gazebo_plugin erlecopter_mark.launch
... logging to /home/bshang2/.ros/log/ca9213bc-89bf-11e7-b657-1803733b65bb/roslaunch-mechatronics-990-28079.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mechatronics-990:45006/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/fcu_url: tcp://127.0.0.1:5760
 * /mavros/gcs_url: udp://127.0.0.1@1...
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: base_link
 * /mavros/local_position/tf/frame_id: world
 * /mavros/local_position/tf/send: True
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
[... 540 lines cut ...]
  - 
    level: 1
    name: mavros: Battery
    message: Low voltage
    hardware_id: tcp://127.0.0.1:5760
    values: 
      - 
        key: Voltage
        value: 0.00
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 100.0
---
### Check ID

bshang2@mechatronics-990:~/simulation/ardupilot/ArduCopter$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 1231 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 136, 152, 150, 24, 27, 29, 30, 32, 33, 162, 163, 164, 165, 42, 173, 178, 62, 193, 74, 35, 36, 241, 116

AlexisTM, 2017-08-28:
Hey there, 

You can use the setpoint_raw.

http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.setpoint_raw

On the target setpoint, you can change the coordinate frame.

http://docs.ros.org/api/mavros_msgs/html/msg/PositionTarget.html

TSC21, 2017-08-28:
For `setpoint_velocity`, I will add a fix which will allows choosing between the `LOCAL_NED` and the `BODY_NED`, since it currently is defaulted to `LOCAL_NED` (https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_velocity.cpp#L81). In the mean time, you can use `setpoint_raw` and set the `coordinate_frame` to `FRAME_BODY_NED` or `MAV_FRAME::BODY_NED` (both should work).

closed 2017-08-28 by TSC21

cnpcshangbo, 2017-08-31:
Thanks @AlexisTM and @TSC21 Your reply is helpful. I will try that later.

cnpcshangbo, 2017-09-08:
@TSC21 @AlexisTM Thanks for your help! I can control the velocity in body frame now. Video is recorded: https://www.youtube.com/watch?v=dyUQfFYX3pk&feature=youtu.be
```shell
rostopic pub /mavros/setpoint_raw/local mavros_msgs/PositionTarget '{header: {stamp: now, frame_id: "world"}, coordinate_frame: 8, type_mask: 3527, velocity: {x: 0.1, y: 0, z: 0}}' -r 10
```

TSC21, 2017-09-08:
@cnpcshangbo good to know

hmchung, 2018-08-17:
hi @cnpcshangbo, thank you for the video.
I post this to help others clearing a confusion that I had.
Even though the frame mask is  "FRAME_BODY_NED = 8", the **ROS command should still be in ENU**  . The ENU -> NED conversion will be done by mavros. 
In fact, from the video, the behavior is: 
- x = 0.1 ---> To the right
- y = 0.1 ---> Forward
Obviously it is still in ENU. However, this confusion cost me a week of pulling hair...

hmchung, 2018-08-20:
So am I correct that we actually have 3 conventions here:
1) NED: PX4 onboard estimation and control
2) ENU: Mavlink commands to PX4
3) NWD: Typical coordinate system in ROS

TSC21, 2018-08-20:
No. There's no NWD in ROS.
http://www.ros.org/reps/rep-0103.html

TSC21, 2018-08-20:
@AlexisTM north or east should no be used to specify body conventions, but rather orientations like front, left.

hmchung, 2018-08-20:
@TSC21, so what is the shortest way to call the Forward-Left-Up convention?

hmchung, 2018-08-20:
Okay. But if I am not wrong, I think we still have 3 conventions?
1) NED: Onboard control
2) Right-Forward-Up: How mavros understands raw (velocity) setpoints
3) Forward-Left-Up: ROS convention

TSC21, 2018-08-20:
Forward-Left-Up (FLU) is the ROS convention for the body frame. East-North-Up (ENU) is the ROS convention for the local/word frame.

Ali76farahmand, 2023-12-19:
> For `setpoint_velocity`, I will add a fix which will allows choosing between the `LOCAL_NED` and the `BODY_NED`, since it currently is defaulted to `LOCAL_NED` [...]

both of LOCAL_NED and BODY_NED have the same attitude (north, east, down). is there any way to command the drone in FRD (front, right, down) coordinate frame?