mavlink/mavros #793: setpoint_attitude subscriber missing

andresR8, 2017-08-29:
### Issue details

setpoint_attitude is not working in the latest release. 

### MAVROS version and platform

Mavros: 0.20.1-0xenial-20170828-082737-0800
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ X] PX4

Version: 1.6.5dev

### Node logs
Apparently mavros is not subscribing to the topic
```
rostopic info /mavros/setpoint_attitude/attitude
Type: geometry_msgs/PoseStamped

Publishers: 
 * /rover_teleop (http://kiwi:35311/)

Subscribers: None

```

### Diagnostics

```
header: 
  seq: 411
  stamp: 
    secs: 1503984484
    nsecs: 805333322
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Received packets:
        value: 32207
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 182
      - 
        key: Tx sequence number:
        value: 207
      - 
        key: Rx total bytes:
        value: 6863362
      - 
        key: Tx total bytes:
        value: 693265
      - 
        key: Rx speed:
        value: 19258.000000
      - 
        key: Tx speed:
        value: 1927.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Heartbeats since startup
        value: 533
      - 
        key: Frequency (Hz)
        value: 1.039960
      - 
        key: Vehicle type
        value: Ground rover
      - 
        key: Autopilot type
        value: PX4 Autopilot
      - 
        key: Mode
        value: OFFBOARD
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 40.3
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Voltage
        value: 65.54
      - 
        key: Current
        value: -0.0
      - 
        key: Remaining
        value: -1.0
  - 
    level: 0
    name: mavros: Time Sync
    message: Normal
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Timesyncs since startup
        value: 5324
      - 
        key: Frequency (Hz)
        value: 9.999631
      - 
        key: Last dt (ms)
        value: 0.162228
      - 
        key: Mean dt (ms)
        value: 0.010355
      - 
        key: Last system time (s)
        value: 5899.246498000
      - 
        key: Time offset (s)
        value: 1503978585.490676165
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 4570 messages, from 1 addresses
sys:comp   list of messages
  1:1     32, 0, 147, 36, 230, 231, 105, 74, 141, 77, 111, 241, 30, 245, 87, 1, 62, 31

```

TSC21, 2017-08-29:
How are you setting up `setpoint_attitude`? If you take a closer look at the code, you must feed both attitude and thrust msgs to it, while `messsage_filters` will compute an approximate sync of them. If the data is not received on one of the topics, it won't publish attitude setpoints at all.

So you have to publish `mavros_msgs/Thrust` msgs to `/mavros/setpoint_attitude/thrust` and `geometry_msgs/PoseStamped` msgs to `/mavros/setpoint_attitude/target_attitude` (or `geometry_msgs/TwistStamped` to `/mavros/setpoint_attitude/cmd_vel` if you pretend to control rates instead of quaternions). So you need both thrust **and** attitude topics in order that the node syncs them before sending the setpoint.

andresR8, 2017-08-29:
I'm basically doing this, but It doesn't work.  

```
  int main(int argc, char** argv){
  signal(SIGINT, signalHandler);
  ros::init(argc, argv, "rover_teleop");
  ros::NodeHandle n;    
  int rate = 20;
  ros::Rate r(rate);     
  ros::Publisher attitude_pub = n.advertise<geometry_msgs::PoseStamped>("/mavros/setpoint_attitude/target_attitude",100);
  ros::Publisher att_thrust_pub = n.advertise<mavros_msgs::Thrust>("/mavros/setpoint_attitude/thrust", 100); 

  while (n.ok()){ 
                    if(current_state.mode != "OFFBOARD") {
				ROS_INFO("No Offboard mode");
				if( set_mode_client.call(set_mode) &&  set_mode.response.mode_sent){
					ROS_INFO("Offboard enabled");
				}                
			}	

                           set_attitude_control(roll_sp,pitch_sp,yaw_sp); // here I`m filling the msgs.
			   attitude_pub.publish(attitude_msg);
			   att_thrust_pub.publish(att_thrust_msg);	
   } 
}
```

I've also set  ```use_quaternion: true``` and   ```tf:listen: false ``` in px4_config.yaml

Maybe an update in the wiki should be helpful.
http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.setpoint_attitude

TSC21, 2017-08-29:
Are you setting the `header` timestamp also for the `attitude_msg` and `att_thrust_msg`?

> Maybe an update in the wiki should be helpful.
http://wiki.ros.org/mavros#mavros.2BAC8-Plugins.setpoint_attitude

I know. That's on the todo list. Many stuff still pendent and not much time/people to do it.

andresR8, 2017-08-29:
I`m filling the messages in this way.
```
void set_attitude_control(double r,double p, double y){ //theta in degrees
  tf::Quaternion qt;
  qt.setRPY (r,p,y); 
  
  attitude_msg.header.stamp = ros::Time::now();  
  attitude_msg.header.frame_id = 1.0;
  attitude_msg.pose.position.x = 0.0;
  attitude_msg.pose.position.y = 0.0;
  attitude_msg.pose.position.z = 0.0;
 
  attitude_msg.pose.orientation.x =(double)qt.x();
  attitude_msg.pose.orientation.y = (double)qt.y();
  attitude_msg.pose.orientation.z =(double) qt.z();
  attitude_msg.pose.orientation.w =(double) qt.w();
  
  att_thrust_msg.header.stamp = ros::Time::now();  
  att_thrust_msg.header.frame_id = 1.0;
  att_thrust_msg.thrust=getLinearVelocity();
  
}
```

Maybe att_thrust_msg.header.stamp &  attitude_msg.header.stamp  should be equals?

I`ve also saw this when I check for the topics.

[image]

> I know. That's on the todo list. Many stuff still pendent and not much time/people to do it.
Maybe I can help

TSC21, 2017-08-29:
> Maybe att_thrust_msg.header.stamp & attitude_msg.header.stamp should be equals?

They don't. `setpoint_attitude` handles an approximation for the stamp through an algorithm of `message_filters`.

> I`ve also saw this when I check for the topics.

`attitude_msg.header.frame_id = 1.0;` You know that `frame_id` is supposed to be a string right? Not a number. I suppose it should be `/map` as you want to setup the vehicle attitude on the local NED frame.

andresR8, 2017-08-29:
> You know that frame_id is supposed to be a string right? 

I was confused because the previous definition. I`ve tested with 
```
attitude_msg.header.frame_id = "/map";
attitude_msg.header.frame_id = "map";
att_thrust_msg.header.frame_id = "/map";
att_thrust_msg.header.frame_id = "map";
```
Nothing works

TSC21, 2017-08-29:
I t has nothing to do with the `frame_id`. It should be something else you are missing or is incorrect on your node.

TSC21, 2017-08-29:
Also, use `rqt_graph` to check if the topics are properly sub/publishing.

andresR8, 2017-08-29:
I've launched mavros alone, and it doens't show the subscription to the mavros/setpoint_attitude/target_attitude or mavros/setpoint_attitude/thrust topics. Maybe the plugin isn't loading?

TSC21, 2017-08-29:
> I've launched mavros alone

What about your node?

andresR8, 2017-08-30:
It's publishing correctly to the topics, but there are not any subscriber.

andresR8, 2017-08-30:
After running roswtf ( and mavros running alone ) this is the output.
```
Found 3 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /mavros:
   * /mavros/actuator_control
   * /mavros/hil/optical_flow
   * /mavros/global_position/set_gp_origin
   * /mavros/setpoint_raw/attitude
   * /mavros/hil/state
   * /mavros/global_position/home
   * /mavros/setpoint_velocity/cmd_vel
   * /mavros/hil/gps
   * /mavros/setpoint_position/local
   * /mavros/odometry/odom
   * /mavros/mocap/pose
   * /mavros/hil/imu_ned
   * /mavros/hil/rc_inputs
   * /tf_static
   * /mavros/setpoint_raw/global
   * /mavros/vision_pose/pose_cov
   * /mavros/fake_gps/mocap/tf
   * /mavlink/to
   * /mavros/rc/override
   * /mavros/setpoint_raw/local
   * /mavros/setpoint_velocity/cmd_vel_unstamped
   * /mavros/vision_pose/pose
   * /mavros/setpoint_accel/accel
   * /mavros/vision_speed/speed_vector
   * /mavros/adsb/send
   * /mavros/home_position/set
```
The topics mavros/setpoint_attitude/target_attitude  mavros/setpoint_attitude/thrust  should be listed here, right? The mavros console output shows that the plugin is loaded and initialized.
```
[ INFO] [1504054160.287075915]: Plugin setpoint_attitude loaded
[ INFO] [1504054160.310443113]: Plugin setpoint_attitude initialized
```

TSC21, 2017-08-30:
> The topics mavros/setpoint_attitude/target_attitude mavros/setpoint_attitude/thrust should be listed here, right? The mavros console output shows that the plugin is loaded and initialized.

Check the output. It states: The following node subscriptions are **unconnected**. So if they are not listed, means they are **connected**.
Check and post, with mavros **and** your node running, the output of:
1. `rostopic echo /mavros/setpoint_attitude/target_attitude`
2. `rostopic echo /mavros/setpoint_attitude/thrust`
3. `rosrun rqt_graph rqt_graph`

Also, post the output of `rostopic echo -n1 /diagnostics`.

andresR8, 2017-08-31:
> Check the output. It states: The following node subscriptions are unconnected. So if they are not listed, means they are connected.

You're right but I was launching mavros alone, so the topics: /mavros/setpoint_attitude/target_attitude /mavros/setpoint_attitude/thrust should be listed there as **unconnected**. So, mavros apparently is not subscribing to this topics ( but it's subscribed to i.e. `/mavros/setpoint_accel/accel ` ). 

Here my results running my node + mavros:

1. 
```
---
header: 
  seq: 286
  stamp: 
    secs: 1504141397
    nsecs: 778209374
  frame_id: map
pose: 
  position: 
    x: 0.0
    y: 0.0
    z: 0.0
  orientation: 
    x: 0.0140046526344
    y: -0.0186352234946
    z: -0.826393816902
    w: 0.562609862636
---
```

2.
```

---
header: 
  seq: 936
  stamp: 
    secs: 1504141479
    nsecs: 777001823
  frame_id: map
thrust: 0.319999992847
---

```

3. 
[image]

4.
```

header: 
  seq: 126
  stamp: 
    secs: 1504141744
    nsecs: 836912177
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Received packets:
        value: 49331
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 169
      - 
        key: Tx sequence number:
        value: 120
      - 
        key: Rx total bytes:
        value: 2050889
      - 
        key: Tx total bytes:
        value: 109264
      - 
        key: Rx speed:
        value: 12720.000000
      - 
        key: Tx speed:
        value: 685.000000
  - 
    level: 2
    name: mavros: GPS
    message: No satellites
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Satellites visible
        value: 0
      - 
        key: Fix type
        value: 0
      - 
        key: EPH (m)
        value: Unknown
      - 
        key: EPV (m)
        value: Unknown
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Heartbeats since startup
        value: 159
      - 
        key: Frequency (Hz)
        value: 0.999971
      - 
        key: Vehicle type
        value: Ground rover
      - 
        key: Autopilot type
        value: PX4 Autopilot
      - 
        key: Mode
        value: OFFBOARD
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 40.5
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Voltage
        value: 65.54
      - 
        key: Current
        value: -0.0
      - 
        key: Remaining
        value: -1.0
  - 
    level: 0
    name: mavros: Time Sync
    message: Normal
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Timesyncs since startup
        value: 1589
      - 
        key: Frequency (Hz)
        value: 9.999690
      - 
        key: Last dt (ms)
        value: 0.067916
      - 
        key: Mean dt (ms)
        value: 0.009175
      - 
        key: Last system time (s)
        value: 1252.232729000
      - 
        key: Time offset (s)
        value: 1504140492.527852535
---

```

TSC21, 2017-08-31:
One thing I can think of is the plugin must be restarting the subscribers every time.

TSC21, 2017-08-31:
@andresR8 can you please push latest master and retest?

andresR8, 2017-08-31:
> @andresR8 can you please push latest master and retest?

Did you mean pull right?

TSC21, 2017-08-31:
Yes pull. Sorry

andresR8, 2017-09-01:
Tested with the latest commit with the same results

TSC21, 2017-09-01:
Well right now I don't have ideas on what could be wrong. If you in the mean time find the solution let me know.

TSC21, 2017-09-08:
@andresR8 any updates?

AlexisTM, 2017-09-08:
@andresR8  Can you check the blacklisted plugins?

I do have the topic you are missing.

Mavros: 0.19.0, from deb packages.

```bash
alexis@alexis-master:~$ rostopic info /odroid100/setpoint_raw/attitude
Type: mavros_msgs/AttitudeTarget

Publishers: None

Subscribers: 
 * /odroid100 (http://192.168.1.100:35389/)
```

andresR8, 2017-09-09:
@AlexisTM  but the topic that I'm looking for is /mavros/setpoint_attitude/target_attitude and ( or /mavros/setpoint_attitude/thrust ) @TSC21 Has the name changed?

AlexisTM, 2017-09-09:
I have 3 setpoint_attitude subscribers: 

```
/mavros/setpoint_attitude/att_throttle
/mavros/setpoint_attitude/attitude
/mavros/setpoint_attitude/cmd_vel
```

TSC21, 2017-09-09:
@AlexisTM no you are not using latest MAVROS release. The latest uses the topics stated above using `message_filters` to syncronize them. I'm not usre though if they appear at all if they are not connected to any publisher (since `message_filters` subscribers is a different implementation of a subscriber). I'm short on time to debug this though.

TSC21, 2017-09-09:
If you want to take some time on understanding and debugging this here is how it works http://wiki.ros.org/message_filters

TSC21, 2017-09-09:
You can also try to change from an `AproximateTime` policy to an `ExactTime` policy and see where it gets you. Also I'm not sure if having the two topics being published from the same node may interefere or not. Also you can increase the queue size on the message_filters subscribers from 1 to 10 and check if it allows topics sync. I believe the problem is having the topics sync between each other.

andresR8, 2017-09-19:
I'm triying to debug it. I think that is a problem with the NodeHandle, because I've added a regular ros subscriber
 `ros::Subscriber test_sub = sp_nh.subscribe("test_sub",10,&SetpointAttitudePlugin::test_cb,this);` 
in the initialize method. The result is the same, the plugin is not subscribing anyways.

TSC21, 2017-09-19:
> The result is the same, the plugin is not subscribing anyways.

Did you create the callback to that method?

andresR8, 2017-09-19:
Yes, but the topic doesn't appear as a subscribed topic

```
void test_cb(const geometry_msgs::PoseStamped::ConstPtr &pose_msg){	
  ROS_INFO("TEST CB");	
}
```

vooon, 2017-09-20:
@TSC21 i really do not see any subscribers of setpoint_attitude. Please check init code.
`field: init` notation in initialize() looks suspicious, i though that this only may be in initializer list (in constructors).

I made this changes to master:
```diff
diff --git a/mavros/launch/px4_config.yaml b/mavros/launch/px4_config.yaml
index 96137c3..2ac0f47 100644
--- a/mavros/launch/px4_config.yaml
+++ b/mavros/launch/px4_config.yaml
@@ -89,7 +89,7 @@ setpoint_accel:
 # setpoint_attitude
 setpoint_attitude:
   reverse_thrust: false     # allow reversed thrust
-  use_quaternion: false     # enable PoseStamped topic subscriber
+  use_quaternion: true     # enable PoseStamped topic subscriber
   tf:
     listen: false           # enable tf listener (disable topic subscribers)
     frame_id: "map"
diff --git a/mavros/src/plugins/setpoint_attitude.cpp b/mavros/src/plugins/setpoint_attitude.cpp
index d227dc7..b51e2a8 100644
--- a/mavros/src/plugins/setpoint_attitude.cpp
+++ b/mavros/src/plugins/setpoint_attitude.cpp
@@ -85,6 +85,9 @@ public:
 			 */
 			message_filters::Synchronizer<SyncPoseThrust> sync(SyncPoseThrust(10), pose_sub, th_sub);
 			sync.registerCallback(boost::bind(&SetpointAttitudePlugin::attitude_pose_cb, this, _1, _2));
+
+			ROS_INFO("Addind test sub");
+			test_sub = sp_nh.subscribe("test_sub", 10, &SetpointAttitudePlugin::test_cb, this);
 		}
 		else {
 			twist_sub: sp_nh, "cmd_vel", 1;
@@ -107,6 +110,8 @@ private:
 	message_filters::Subscriber<geometry_msgs::PoseStamped> pose_sub;
 	message_filters::Subscriber<geometry_msgs::TwistStamped> twist_sub;
 
+	ros::Subscriber test_sub;
+
 	std::string tf_frame_id;
 	std::string tf_child_frame_id;
 
@@ -210,6 +215,10 @@ private:
 			send_attitude_ang_velocity(req->header.stamp, ang_vel, thrust_msg->thrust);
 	}
 
+	void test_cb(const mavros_msgs::Thrust::ConstPtr &msg) {
+		ROS_INFO("got att thrust msg");
+	}
+
 };
 }	// namespace std_plugins
 }	// namespace mavros
```

And then `rosnode info /mavros`:
```
--------------------------------------------------------------------------------
Node [/mavros]
Publications: 
 * /mavros/manual_control/control [mavros_msgs/ManualControl]
 * /mavros/adsb/vehicle [mavros_msgs/ADSBVehicle]
 * /tf [tf2_msgs/TFMessage]
 * /mavros/radio_status [mavros_msgs/RadioStatus]
 * /mavros/setpoint_raw/target_local [mavros_msgs/PositionTarget]
 * /mavros/imu/atm_pressure [sensor_msgs/FluidPressure]
 * /mavros/altitude [mavros_msgs/Altitude]
 * /mavros/global_position/compass_hdg [std_msgs/Float64]
 * /mavros/hil/actuator_controls [mavros_msgs/HilActuatorControls]
 * /mavros/global_position/global [sensor_msgs/NavSatFix]
 * /mavros/target_actuator_control [mavros_msgs/ActuatorControl]
[... 97 lines cut ...]

contacting node http://06dc880427d0:37285/ ...
Pid: 13039
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
 * topic: /tf
    * to: /mavros
    * direction: outbound
    * transport: INTRAPROCESS
 * topic: /tf
    * to: /mavros (http://06dc880427d0:37285/)
    * direction: inbound
    * transport: INTRAPROCESS

root@06dc880427d0:/# rostopic info /mavros/setpoint_attitude/test_sub
Type: mavros_msgs/Thrust

Publishers: None

Subscribers: 
 * /mavros (http://06dc880427d0:37285/)
```

vooon, 2017-09-20:
Note in rosnode `[unknown type]` is ok.
```
root@06dc880427d0:/# rostopic info /mavros/setpoint_attitude/test_sub
Type: mavros_msgs/Thrust

Publishers: None

Subscribers: 
 * /mavros (http://06dc880427d0:37285/)
```

AlexisTM, 2017-09-20:
Hi there, issue is also in the 0.21.0 version. 

The #817 PR should correct it.

TSC21, 2017-09-20:
@andresR8, @AlexisTM fixed this with #817. You should now be able to subscribe.

closed 2017-09-20 by TSC21

andresR8, 2017-09-21:
@TSC21  solved with this [commit](https://github.com/mavlink/mavros/commit/66db2fd7bc0d855c8510d243315e351570eb6c5d)

btw It's possible to release a patch, fixing this bug or I have to wait for the next major release?

TSC21, 2017-09-21:
We are waiting for another PR to get it before doing a release.

vooon, 2017-09-21:
@andresR8 use source install, because even if i release it now you have to wait weeks before repo will sync.

AlexisTM, 2017-09-21:
@andresR8 You can follow the following guide to do the source install: https://github.com/mavlink/mavros/blob/master/mavros/README.md#user-content-source-installation

andresR8, 2017-09-21:
@AlexisTM @vooon Thanks for the suggestion, However I'm deploying the code in multiple robots and I don't want to make the source installation in each one. How many weeks do you think that it's going to take ?

AlexisTM, 2017-09-22:
@andresR8 You can do a catkin install on one of the robots and then use
rsync or scp to transfer the compiled version across the different robots
with the same board.

andresR8, 2017-09-22:
@TSC21 @AlexisTM Do you know what's the correct way to publish to the Sync topics?  I'm publishing to both topics one after the other. However  the function &SetpointAttitudePlugin::attitude_pose_cb is not triggered.

TSC21, 2017-09-22:
It's published the same way. If you have doubts on how it works, check the docs I shared above.

andresR8, 2017-09-22:
I'm publishing in the correct way, this is a section of my config yaml file.
```
# setpoint_attitude
setpoint_attitude:
  reverse_thrust: true     # allow reversed thrust
  use_quaternion: true     # enable PoseStamped topic subscriber
  tf:
    listen: false           # enable tf listener (disable topic subscribers)
    frame_id: "map"
    child_frame_id: "target_attitude"
    rate_limit: 50.0
```

TSC21, 2017-09-22:
Why do you have reverse thrust set to true?

andresR8, 2017-09-22:
Because my rover can move backwards

TSC21, 2017-09-22:
Can you post here the output of echoing both topics (for thrust and posestamped)? Are you setting the timestampo for the thrust topic also?

andresR8, 2017-09-22:
target_attitude topic published
```
---
header: 
  seq: 630
  stamp: 
    secs: 1506115052
    nsecs: 297204736
  frame_id: map
pose: 
  position: 
    x: -1.06005066563e-15
    y: 2.13386149441e-31
    z: 8.65597057343
  orientation: 
    x: -0.908122613751
    y: -0.418624268257
    z: -0.0055378250679
    w: 0.00604243117515
---

```

Thrust topic published
```

---
header: 
  seq: 623
  stamp: 
    secs: 1506115050
    nsecs: 663761808
  frame_id: map
thrust: 0.319999992847
---

```

TSC21, 2017-09-22:
And you say they it does not trigger the callback? Try to change from an approximate to an exact syn policy and check if it does any difference.

andresR8, 2017-09-22:
I've made this change and nothing happened.   Btw this is the way on how I'm filling the messages, and when I used the exact sync policy I set the same ros::Time in both messages's timestamps.

```
 attitude_msg.header.stamp = ros::Time::now();  
  attitude_msg.header.frame_id = "map";
  attitude_msg.pose=attitude_pose;

  att_thrust_msg.header.stamp = ros::Time::now();  //  =attitude_msg.header.stamp    For exact sync pol
  att_thrust_msg.header.frame_id = "map"; 
  att_thrust_msg.thrust = getLinearVelocity();
```

TSC21, 2017-09-22:
ros::Time::now() gives you the time the line is processed in runtime. It does not mean it is exactly the same if set to both headers. In the matter of fact, it code is run sequentially, so it will never be the same. Though, I don't think that is the problem because it should already have sync using the approximate sync.
So to eliminate the hypothesis, try to save ros::Time::now() in a variable and then set that value for both headers (thrust and attitude).

andresR8, 2017-09-22:
tested with Exact and Approximate policies
```
 ros::Time now_time=ros::Time::now(); 
  attitude_msg.header.stamp =  now_time;
  attitude_msg.header.frame_id = "map";
  attitude_msg.pose=attitude_pose;

  att_thrust_msg.header.stamp = now_time; 
  att_thrust_msg.header.frame_id = "map"; 
  att_thrust_msg.thrust = getLinearVelocity();
```

AlexisTM, 2017-09-23:
I never used those so I really cannot give you any about it except what is written is the doc :

It should send the synced message automatically.

andresR8, 2017-09-23:
I've found the solution let me add a PR

TSC21, 2017-09-26:
@andresR8 can you please try to replaces https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_attitude.cpp#L112-L113 with `std::shared_ptr`s instead?

andresR8, 2017-09-30:
@TSC21 It's working I had a my px4 config.yaml misconfigured.

siddhu45, 2017-10-10:
Hi guys! Thanks for your effort @andresR8 @TSC21 @TSC21 @vooon 
So, the latest source installed mavros should work right with the changes mentioned here and in #817 & #821 ? Because just now I tried using `/mavros/setpoint_attitude/thrust` and `/mavros/setpoint_attitude/target_attitude` topics to send quaternion and thrust command. When I run `rqt_graph` I see this
[image]

It is just sending the thrust command and not the attitude commands. It is publishing because when I run `rostopic echo /mavros/setpoint_attitude/target_attitude` I see the command. I tried all the things you guys mentioned related to this. Changed the use_quaternion in the px4_config.yaml, tried 

> can you please try to replace https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_attitude.cpp#L112-L113 with std::shared_ptrs and recompile?

didnt work. And the other minor changes. Can you please let me know if you were able to solve this issue, how did you do it. That would be great. Thanks a lot.

PS: Previously I was using the bedian built mavros and even the throttle was not publishing but just then I came across this issue and did source installation and atleast was able to send something but not all. Anyways, it would be great if you help me out here. 
Thanks. Cheers!

vooon, 2017-10-10:
@siddhu45 please fire new issue and follow template. On what commit rev?

siddhu45, 2017-10-11:
@vooon Ok I will do that. And the latest source build, I just did yesterday, I guess it is version 0.22?

siddhu45, 2017-10-11:
@vooon Ok I will do that. And the latest source build, I just did yesterday, I guess it is version 0.21?

Jaeyoung-Lim, 2018-06-08:
This issue is still not fixed

I am currently trying to use setpoint_atttude/cmd_vel with a px4 SITL simulation but has the same problem as described above.

mavros is not subscribing to setpoint_attitude/cmd_vel but subscribes to setpoint_attitude/thrust

[7 more replies]