mavlink/mavros #794: Using the PositionTarget message

Seanmatthews, 2017-08-29:
Is there a configuration with using setpoint_raw and the PositionTarget message where I'm able to command the vehicle to a position at a velocity? After reading a slew of velocity control issues, I can't figure it. Alternatively, can I set the maximum velocity used by position control over the XY (horizontal) plane?

TSC21, 2017-08-29:
Check what's done on `test_mavros` with the `control_toolbox`.

closed 2017-09-08 by TSC21

Seanmatthews, 2017-09-14:
@TSC21 I took a look at the packages and it doesn't look like they're just sending setpoints. Is velocity mode required to have any sort of direct control over the travel velocity between setpoints?

TSC21, 2017-09-14:
> Is velocity mode required to have any sort of direct control over the travel velocity between setpoints?

Yes. You can directly control the velocity or use the PID controller.