mavlink/mavros #800: Setting PX4 parameters using MAVROS

shening, 2017-08-31:
Hi 

I am trying to change some parameters on a PX4 flight stack using MAVROS. 

I am able to read the parameter with mavros using
rosservice call /mavros/param/get MPC_LAND_SPEED

however I  am not able to set it using mavros. I do not  know  what he correct syntax is 

I tried
rosservice call /mavros/param/set MPC_LAND_SPEED '0.4' 
however it gives me an error. If anyone can help I would be very grateful!

Thank you

Sebastian

vooon, 2017-08-31:
Does you tried `mavparam` script? `rosrun mavros mavparam set MPC_LAND_SPEED 0.4`?

AlexisTM, 2017-09-01:
It is because you are trying to call the set method via a service.

Via a service, the ROS way is:

TSC21, 2017-09-01:
Seem you have two working options @shening. Closing

closed 2017-09-01 by TSC21

bigbellmercy, 2020-05-11:
> It is because you are trying to call the set method via a service.
> 
> Via a service, the ROS way is:
> rosservice call /param/set "param_id: 'MPC_LAND_SPEED'
> value:
> integer: 0
> real: 0.4"

The above command makes an error in my trial. But it worked after a blank space was added before 'integer' and 'real' as like:

`rosservice call /param/set "param_id: 'SENS_EN_THERMAL'
value:
 integer: 55
 real: 0.0"`

Jaeyoung-Lim, 2020-05-11:
@bigbellmercy Is this issue apparent only in command line arguments? or is this also true when you use rosservice calls over roscpp?

AlexisTM, 2020-05-11:
@bigbellmercy Indeed, that was 3 years ago and the space was not obvious, I just added ``` around it and now it is obvious. The typical way I use rosservice it is to make abusive use of <TAB> to prevent typos :laughing:

bigbellmercy, 2020-05-18:
> @bigbellmercy Is this issue apparent only in command line arguments? or is this also true when you use rosservice calls over roscpp?

I failed to 'rosservice calls over roscpp' using the command lines.

But the below code works for PX4 parameter change in roscpp:

`system("rosrun mavros mavparam set SENS_EN_THERMAL 55");`

Jaeyoung-Lim, 2020-05-18:
@Jaeyoung-Lim I meant roscpp by using the [ros service calls](http://wiki.ros.org/roscpp/Overview/Services), not using command line arguments

bigbellmercy, 2020-05-18:
@Jaeyoung-Lim I couldn't find the solution for it in my trials.

tropappar, 2021-01-25:
Using [roscpp service calls](https://wiki.ros.org/roscpp/Overview/Services), I was also unable to directly get or set a PX4 parameter. I always got the following error:
```
[/mavros:std_plugins::ParamPlugin::get_cb]: PR: Unknown parameter to get: MPC_XY_VEL_MAX
[/mavros:std_plugins::ParamPlugin::set_cb]: PR: Unknown parameter to set: MPC_XY_VEL_MAX
```
Is there some prefix to add to the parameter ID?

It worked without problems using the [mavparam](https://wiki.ros.org/mavros#mavparam) script:
`rosrun mavros mavparam -n /mavros get MPC_XY_VEL_MAX`

However, I want to do it programmatically. I found a workaround by pulling the parameters to ROS, making the desired changes, and then pushing them all back:
```
ServiceClient pull_client = nh.serviceClient<mavros_msgs::ParamPull>("mavros/param/pull");
pull_client.waitForExistence();
mavros_msgs::ParamPull pull;
if (pull_client.call(pull)) {
    if (pull.response.success) {
        ROS_DEBUG("Pulled %d parameters from FCU", pull.response.param_received);
        nh.setParam("mavros/param/MPC_XY_VEL_MAX", max_velocity);
        ServiceClient push_client = nh.serviceClient<mavros_msgs::ParamPush>("mavros/param/push");
        push_client.waitForExistence();
        mavros_msgs::ParamPush push;
        if (push_client.call(push)) {
            if (push.response.success)
                ROS_DEBUG("Pushed %d parameters to FCU", push.response.param_transfered);
            else
                ROS_ERROR("Failed push parameters to FCU, cannot set maximum horizontal velocity!");
        }
        else {
            ROS_ERROR("Failed to push parameters to FCU, cannot set maximum horizontal velocity!");
        }
    }
    else {
        ROS_ERROR("Failed to pull parameters from FCU, cannot set maximum horizontal velocity!");
    }
}
else {
    ROS_ERROR("Failed to pull parameters from FCU, cannot set maximum horizontal velocity!");
}
```