mavlink/mavros #806: too much home position info

Doolly, 2017-09-10:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I updated mavros to 0.19.0 and 
[ INFO] [1505049051.460521094, 1049.234000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002

this msg publishes every second how can I turn off?

### MAVROS version and platform

Mavros: 0.19.0
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ 0 ] PX4

Version: 1.5.5

### Node logs

```
[ INFO] [1505049056.459067869, 1054.214000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049057.465803059, 1055.218000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049057.759337680, 1055.512000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049058.124152742, 1055.876000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049058.458771789, 1056.212000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049059.465740591, 1057.214000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049059.758815375, 1057.504000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049060.124985056, 1057.872000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049060.459737732, 1058.204000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049061.465739008, 1059.208000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049061.759324474, 1059.504000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049062.125523372, 1059.870000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049062.458906340, 1060.200000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049063.465512566, 1061.208000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049063.760870254, 1061.502000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049064.126555001, 1061.868000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049064.463412773, 1062.200000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049065.467666920, 1063.206000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002
[ INFO] [1505049065.774239499, 1063.510000000]: HP: Home lat 47.397743, long 8.545593, alt 488.190002

```

### Diagnostics

```
place here result of:
rostopic echo -n1 /diagnostics

header: 
  seq: 218
  stamp: 
    secs: 977
    nsecs: 764000000
  frame_id: ''
status: 
  - 
    level: 0
    name: camila0/mavros: FCU connection
    message: connected
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Received packets:
        value: 43150
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 164
      - 
        key: Tx sequence number:
        value: 228
      - 
        key: Rx total bytes:
        value: 4414377
      - 
        key: Tx total bytes:
        value: 369365
      - 
        key: Rx speed:
        value: inf
      - 
        key: Tx speed:
        value: inf
  - 
    level: 0
    name: camila0/mavros: GPS
    message: 3D fix
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Satellites visible
        value: 10
      - 
        key: Fix type
        value: 3
      - 
        key: EPH (m)
        value: 0.00
      - 
        key: EPV (m)
        value: 0.00
  - 
    level: 0
    name: camila0/mavros: Heartbeat
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Heartbeats since startup
        value: 213
      - 
        key: Frequency (Hz)
        value: 1.000000
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: PX4 Autopilot
      - 
        key: Mode
        value: AUTO.LOITER
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: camila0/mavros: System
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: camila0/mavros: Battery
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Voltage
        value: 12.15
      - 
        key: Current
        value: -1.0
      - 
        key: Remaining
        value: 100.0
  - 
    level: 0
    name: camila0/mavros: Time Sync
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Timesyncs since startup
        value: 10
      - 
        key: Frequency (Hz)
        value: 9.208103
      - 
        key: Last dt (ms)
        value: -0.161228
      - 
        key: Mean dt (ms)
        value: -0.537418
      - 
        key: Last system time (s)
        value: 977.500337000
      - 
        key: Time offset (s)
        value: 0.177598508

```

### Check ID

```
rosrun mavros checkid

WARNING: /mavros/target_system_id not set. Used default value: 1
WARNING: /mavros/target_component_id not set. Used default value: 1
NOTE: Target parameters may be unset, but that may be result of incorrect --mavros-ns option.Double check results!
ERROR. I got 0 addresses, but not your target 1:1

---
Received 0 messages, from 0 addresses
sys:comp   list of messages

```

TSC21, 2017-09-10:
You are not using latest release. That was already changed. Closing

closed 2017-09-10 by TSC21

TSC21, 2017-09-10:
Since the 0.20.1 is not yet on the deb repositories, you can do a source install of it on your catkin ws.

Doolly, 2017-09-10:
thank you ! I will
and updates in deb package would be nice too

TSC21, 2017-09-10:
It was done already but it will only be released when a general ROS packages releases is done. It does not depend on us now. We can only wait

filiperosa, 2018-01-25:
Hi, I'm having the same issue. The “[ INFO] [1516822408.641390818]: HP: requesting home position” being printed every second and preventing me from arming the vehicle. I installed the lates MAVROS from source as you advised @TSC21 .

TSC21, 2018-01-25:
You don't the latest mavros. Those msgs do not appear on latest master

ykhedar, 2018-01-31:
Hi, I am also having the same error `HP: requesting home position`. This persists even though I use the latest master branch from git. Or the master is not the latest branch and I am missing something. :)

Thanks.

TSC21, 2018-01-31:
Yeah seems like this is still there (what changed was actually the home position itself being just presented when debug is on). If you all feel this spams the command line, it can be changed. Otherwise, this is just a reminder you don't have a position lock.

ykhedar, 2018-01-31:
I don't have any issue with the INFO to the console, I am trying to figure out the problem. I am using the PX4 SITL and just found out that this occurs when the position estimator is set to 'ekf2' or 'inav'. No INFO is recorded to the console when using 'lpe' as the estimator. But I think this is then an issue for the PX4 devel. 
Thanks.

TSC21, 2018-01-31:
This happens if your vehicle does not set its home position given it does not have a position source or the estimator is failing on estimating it.