mavlink/mavros #807: Unable to switch to AUTO.MISSION. But QGC does it.

shakthi-prashanth-m, 2017-09-11:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I am unable to enter to MISSION mode.

```
 // Set to Mission mode
  ROS_INFO("Now, Setting to Mission mode...");
  mavros_msgs::SetMode mission_set_mode;
  mission_set_mode.request.custom_mode = "AUTO.MISSION";
  ros::ServiceClient set_mode_client =
      nh.serviceClient<mavros_msgs::SetMode>("mavros/set_mode");

  if (set_mode_client.call(mission_set_mode) &&
      mission_set_mode.response.success) {
    ROS_INFO("In Mission mode now");  /* **************** => I got this printed **********/
    // ARM
    mavros_msgs::CommandBool arm_cmd;
    arm_cmd.request.value = true;
    ros::ServiceClient arming_client =
        nh.serviceClient<mavros_msgs::CommandBool>("mavros/cmd/arming");

    // Lets not overload FCU
    sleep(5.0);

    if (!current_state.armed) {
      if (arming_client.call(arm_cmd) && arm_cmd.response.success) {
        ROS_INFO("Vehicle armed");
      }
    }
  } else {
    ROS_ERROR("Failed to set Mission mode!");
    ros::shutdown();
}
```
I am sending Waypoint missions to PX4 & setting to Mission mode. From the state callback (`/mavros/state`) I came to know that mode switch didn't happen. But after uploading waypoints, I am able to do this in QGC. What QGC does differently ? Am I missing something?

Kindly help to resolve this.
Thanks in advance.

--Shakthi

MAVROS version and platform

Mavros: 0.19.0
ROS: Kinetic
Ubuntu: 16.04
Autopilot type and version

PX4
Version: 4.4.0

TSC21, 2017-09-12:
> I am sending Waypoint missions to PX4 & setting to Mission mode. Though `mission_set_mode.response.succes` is true, from the state callback (`/mavros/state`) I came to know that mode switch didn't happen. How could this happen ? If setting to Mission mode fails, it should be reported in `mission_set_mode.response.succes` as `false` isn't it ?

This was already addressed here: https://github.com/mavlink/mavros/commit/599c5887ad2aadc124d7e1354f3a1597402118da. Or you wait that the deb package of 0.20.1 is released or you can always clone upstream master and build from src on you catkin_ws.

shakthi-prashanth-m, 2017-09-12:
Thanks @TSC21 . I changed `mission_set_mode.response.succes` to `mission_set_mode.response.mode_sent` & cloned `0.20.1` version of mavros. But the actually issue is not that. Inspite of this change, mode switch is not happening. **It happens only if I launch QGC. Its strange.**

TSC21, 2017-09-12:
I suppose something in the implementation needs to be reviewed. Though I'm short on time to do it myself. If you are able to find a solution let me know and we can push a fix.

shakthi-prashanth-m, 2017-09-12:
OK. Sure. Thanks.

TSC21, 2017-09-26:
This was already clarified. Closing.

closed 2017-09-26 by TSC21