mavlink/mavros #808: Mavlink timesync errors with px4 SITL

msadowski, 2017-09-12:
### Issue details

I'm following the tutorial on interfacing [Px4 sitl to ROS](https://dev.px4.io/en/simulation/ros_interface.html) however I'm running into an issue where I get constant timesync errors of the form:

```
ERROR [mavlink] [timesync] Hard setting offset.
[ WARN] [1505138592.839174029, 427.000000000]: TM: Clock skew detected (0.012838448 s). Hard syncing clocks.
ERROR [mavlink] [timesync] Hard setting offset.
[ INFO] [1505138593.274281014, 427.282000000]: HP: Home lat 47.397742, long 8.545594, alt 488.101013
ERROR [mavlink] [timesync] Hard setting offset.
```

I'm running the [mavros_posix_sitl.launch](https://github.com/PX4/Firmware/blob/master/launch/mavros_posix_sitl.launch), as per Px4 SITL docs site.

I noticed that the frequency of errors increases when the FPS in gazebo decreases (at 30-35 fps the error rate is around 3-5 errors a second, at around 40 FPS it's around 1 error every 2-3 seconds). 

Not sure if this is helpful but my gzserver runs at ~80% CPU and gzclient 50%.

Is there anything I can do to fix those synchronization problems? 

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[x] PX4

Version: 1.6.5dev (694de3)

### Node logs

```
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
[ INFO] [1505138590.269306043, 425.524000000]: HP: Home lat 47.397742, long 8.545594, alt 488.101013
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
ERROR [mavlink] [timesync] Hard setting offset.
[ WARN] [1505138592.839174029, 427.000000000]: TM: Clock skew detected (0.012838448 s). Hard syncing clocks.
ERROR [mavlink] [timesync] Hard setting offset.
[ INFO] [1505138593.274281014, 427.282000000]: HP: Home lat 47.397742, long 8.545594, alt 488.101013
ERROR [mavlink] [timesync] Hard setting offset.
```

### Diagnostics

```
header: 
  seq: 63
  stamp: 
    secs: 59
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Received packets:
        value: 22809
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 49
      - 
        key: Tx sequence number:
        value: 245
      - 
        key: Rx total bytes:
        value: 946245
      - 
        key: Tx total bytes:
        value: 29775
      - 
        key: Rx speed:
        value: 15890.000000
      - 
        key: Tx speed:
        value: 493.000000
  - 
    level: 0
    name: mavros: GPS
    message: 3D fix
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Satellites visible
        value: 10
      - 
        key: Fix type
        value: 3
      - 
        key: EPH (m)
        value: 0.00
      - 
        key: EPV (m)
        value: 0.00
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Heartbeats since startup
        value: 55
      - 
        key: Frequency (Hz)
        value: 0.900000
      - 
        key: Vehicle type
        value: Quadrotor
      - 
        key: Autopilot type
        value: PX4 Autopilot
      - 
        key: Mode
        value: AUTO.LOITER
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Voltage
        value: 12.15
      - 
        key: Current
        value: -1.0
      - 
        key: Remaining
        value: 100.0
  - 
    level: 0
    name: mavros: Time Sync
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Timesyncs since startup
        value: 16
      - 
        key: Frequency (Hz)
        value: 9.411765
      - 
        key: Last dt (ms)
        value: -4.512335
      - 
        key: Mean dt (ms)
        value: -2.896674
      - 
        key: Last system time (s)
        value: 55.895641000
      - 
        key: Time offset (s)
        value: 3.004554066
```

### Check ID

```
OK. I got messages from 1:1.

Received 5678 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 24, 30, 31, 32, 33, 74, 76, 83, 85, 87, 102, 230, 105, 111, 241, 242, 245
```

TSC21, 2017-09-12:
This is mainly related to low processing power on your system. Try to disable shadows on gazebo or apply what was done here https://github.com/PX4/sitl_gazebo/commit/0686806fd94bd429aa66d2500e5e2aad16d2835b on the Iris model. If it still doesn't work, you should run this on a more powerful system.

closed 2017-09-12 by TSC21

YCH188, 2018-02-04:
@TSC21 @msadowski I also have the problem.And I'm not sure what this error means.

chekakta, 2018-02-05:
@TSC21 I have the same problem, and I spent a lot of time with this error but nothing working
could someone help us with this issue?

msadowski, 2018-02-05:
@YCH188 @chekakta - in my case everything was indicating the problem was with my PC not being powerful enough (as @TSC21 mentioned). If I remember correctly, when run on a more powerful machine, the problem went away.

YCH188, 2018-02-05:
Thanks a lot. @msadowski