mavlink/mavros #809: Problem with apm2.launch RPI 3B

jonatanjosm, 2017-09-13:
Hi, i'm trying use an apm2.8 with Raspberrypi 3B but i have problems when i use the command 

` roslaunch mavros apm2.launch `

### MAVROS version and platform

Mavros:  installed from binary pacakge. "sudo apt-get install ros-kinetic-mavros ros-kinetic-mavros-extras"
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

ArduPilot

Version: 2.8

### Node logs

command  ` roslaunch mavros apm2.launch `
```
... logging to /home/murcia/.ros/log/62e57b36-9811-11e7-b269-b827eb7b8a84/roslaunch-murcia-desktop-2256.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://murcia-desktop:37497/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 0
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fcu_url: /dev/ttyS0:57600
 * /mavros/gcs_url:
 * /mavros/global_position/frame_id: fcu
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: fcu_utm
 * /mavros/global_position/tf/frame_id: local_origin
 * /mavros/global_position/tf/send: False
 * /mavros/image/frame_id: px4flow
 * /mavros/imu/angular_velocity_stdev: 0.000349065850399
 * /mavros/imu/frame_id: fcu
 * /mavros/imu/linear_acceleration_stdev: 0.0003
 * /mavros/imu/magnetic_stdev: 0.0
 * /mavros/imu/orientation_stdev: 1.0
 * /mavros/local_position/frame_id: fcu
 * /mavros/local_position/tf/child_frame_id: fcu
 * /mavros/local_position/tf/frame_id: local_origin
 * /mavros/local_position/tf/send: False
 * /mavros/mission/pull_after_gcs: True
 * /mavros/mocap/use_pose: True
 * /mavros/mocap/use_tf: False
 * /mavros/plugin_blacklist: ['actuator_contro...
[... 107 lines cut ...]
process has died [pid 2298, exit code -6, cmd /opt/ros/kinetic/lib/mavros/mavros_node __name:=mavros __log:=/home/murcia/.ros/log/62e57b36-9811-11e7-b269-b827eb7b8a84/mavros-2.log].
log file: /home/murcia/.ros/log/62e57b36-9811-11e7-b269-b827eb7b8a84/mavros-2*.log
Initiating shutdown!
================================================================================
[mavros-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

i follow the tutorial  http://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html
 
command `mavproxy.py --master=/dev/ttyS0 --baudrate 57600 --aircraft MyCopter`
```
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect /dev/ttyS0 source_system=255
no script MyCopter/mavinit.scr
Log Directory: MyCopter/logs/2017-09-12/flight1
Telemetry log: MyCopter/logs/2017-09-12/flight1/flight.tlog
MAV> Waiting for heartbeat from /dev/ttyS0
online system 1
STABILIZE> Mode STABILIZE
```

TSC21, 2017-09-13:
> WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk

So have you done this? Also `mavproxy` is one thing, mavros is another.

jonatanjosm, 2017-09-13:
Yes you are rigth but the point is, the conection test works so why apm2.launch doesn't?

TSC21, 2017-09-13:
The fact that it is sending EOF may be related to: 1. bad wiring; 2. busy port; 3. a bad connection shutdown that doesn't allow further connections on the port. Related: https://github.com/mavlink/mavros/issues/566

closed 2017-09-13 by TSC21

jonatanjosm, 2017-09-13:
Ummm but, why doesn't works with apm2.launch and with the conection test of mavproxy do?

TSC21, 2017-09-13:
Reboot and retry without launching maxproxy before

jonatanjosm, 2017-09-13:
Oh, i feel like a complete idiot, thank you so much and i'm sorry for bothering, emm i have another question, when i run apm2.launch it show me this
```
 WARN] [1505332668.058478344]: init: message from slugs, MSG-ID 193 ignored! Table has different entry.
[ WARN] [1505332668.058618710]: init: message from slugs, MSG-ID 194 ignored! Table has different entry.
[ INFO] [1505332668.061131394]: serial0: device: /dev/ttyS0 @ 57600 bps
[ INFO] [1505332668.061915934]: GCS bridge disabled
[ INFO] [1505332668.294284934]: Plugin 3dr_radio loaded
[ INFO] [1505332668.304073481]: Plugin 3dr_radio initialized
[ INFO] [1505332668.304365672]: Plugin actuator_control blacklisted
[ INFO] [1505332668.495366501]: Plugin adsb loaded
[ INFO] [1505332668.523590723]: Plugin adsb initialized
[ INFO] [1505332668.524020779]: Plugin altitude blacklisted
[ INFO] [1505332668.524757090]: Plugin cam_imu_sync loaded
[ INFO] [1505332668.531509769]: Plugin cam_imu_sync initialized
[ INFO] [1505332668.532323112]: Plugin command loaded
[ INFO] [1505332668.569883687]: Plugin command initialized
[ INFO] [1505332668.570565413]: Plugin distance_sensor loaded
[ INFO] [1505332668.637069020]: Plugin distance_sensor initialized
[ INFO] [1505332668.637361107]: Plugin ftp blacklisted
[ INFO] [1505332668.638022260]: Plugin global_position loaded
[ INFO] [1505332668.714848482]: Plugin global_position initialized
[ INFO] [1505332668.715646460]: Plugin hil_actuator_controls loaded
[ INFO] [1505332668.721094541]: Plugin hil_actuator_controls initialized
[ INFO] [1505332668.722252002]: Plugin hil_controls loaded
[ INFO] [1505332668.726118556]: Plugin hil_controls initialized
[ INFO] [1505332668.727094088]: Plugin home_position loaded
[ INFO] [1505332668.766942657]: Plugin home_position initialized
[ INFO] [1505332668.767802198]: Plugin imu_pub loaded
[ INFO] [1505332668.832247292]: Plugin imu_pub initialized
[ INFO] [1505332668.833468973]: Plugin local_position loaded
[ INFO] [1505332668.878356297]: Plugin local_position initialized
[ INFO] [1505332668.879099847]: Plugin manual_control loaded
[ INFO] [1505332668.883638908]: Plugin manual_control initialized
[ INFO] [1505332668.883829899]: Plugin mocap_fake_gps blacklisted
[ INFO] [1505332668.883995839]: Plugin mocap_pose_estimate blacklisted
[ INFO] [1505332668.884761316]: Plugin param loaded
[ INFO] [1505332668.903033343]: Plugin param initialized
[ INFO] [1505332668.903206834]: Plugin px4flow blacklisted
[ INFO] [1505332668.903824706]: Plugin rc_io loaded
[ INFO] [1505332668.931518296]: Plugin rc_io initialized
[ INFO] [1505332668.931687673]: Plugin safety_area blacklisted
[ INFO] [1505332668.932332524]: Plugin setpoint_accel loaded
[ INFO] [1505332668.962610206]: Plugin setpoint_accel initialized
[ INFO] [1505332668.963389850]: Plugin setpoint_attitude loaded
[ INFO] [1505332669.063316004]: Plugin setpoint_attitude initialized
[ INFO] [1505332669.064219243]: Plugin setpoint_position loaded
[ INFO] [1505332669.118570774]: Plugin setpoint_position initialized
[ INFO] [1505332669.119780215]: Plugin setpoint_raw loaded
[ INFO] [1505332669.199389835]: Plugin setpoint_raw initialized
[ INFO] [1505332669.200238542]: Plugin setpoint_velocity loaded
[ INFO] [1505332669.241351700]: Plugin setpoint_velocity initialized
[ INFO] [1505332669.242615203]: Plugin sys_status loaded
[ INFO] [1505332669.284344097]: Plugin sys_status initialized
[ INFO] [1505332669.285317232]: Plugin sys_time loaded
[ INFO] [1505332669.317304509]: TM: Timesync mode: NONE
[ INFO] [1505332669.322916497]: Plugin sys_time initialized
[ INFO] [1505332669.323817341]: Plugin vfr_hud loaded
[ INFO] [1505332669.333137338]: Plugin vfr_hud initialized
[ INFO] [1505332669.333652292]: Plugin vibration blacklisted
[ INFO] [1505332669.333900003]: Plugin vision_pose_estimate blacklisted
[ INFO] [1505332669.334056359]: Plugin vision_speed_estimate blacklisted
[ INFO] [1505332669.334985223]: Plugin waypoint loaded
[ INFO] [1505332669.364199917]: Plugin waypoint initialized
[ INFO] [1505332669.364604453]: Built-in SIMD instructions: None
[ INFO] [1505332669.364822581]: Built-in MAVLink package version: 2017.8.8
[ INFO] [1505332669.365026697]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1505332669.365207585]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1505332670.274061989]: CON: Got HEARTBEAT, connected. FCU: ArduPilotMega / ArduCopter
[ WARN] [1505332672.279761236]: VER: broadcast request timeout, retries left 4
[ WARN] [1505332673.279488049]: VER: broadcast request timeout, retries left 3
[ WARN] [1505332674.292242639]: VER: unicast request timeout, retries left 2
[ WARN] [1505332675.310815141]: VER: unicast request timeout, retries left 1
[ WARN] [1505332676.292554944]: VER: unicast request timeout, retries left 0
[ WARN] [1505332677.291731480]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ INFO] [1505332680.275359145]: HP: requesting home position
[ERROR] [1505332680.299278205]: FCU: ArduCopter V3.2.1 (36b405fb)
[ERROR] [1505332680.309139853]: FCU: Frame: QUAD
[ INFO] [1505332685.289341426]: WP: mission received
[ INFO] [1505332690.275406478]: HP: requesting home position
[ INFO] [1505332692.451551579]: PR: parameters list received
[ INFO] [1505332700.275267547]: HP: requesting home position

```
I was looking for and i found that the latest firmware version for APM2.8 is 3.2.1, is it true?
```
[ERROR] [1505332680.299278205]: FCU: ArduCopter V3.2.1 (36b405fb)
```
could work with mavros and that version?

pavloblindnology, 2017-11-17:
Found solution without restarting a companion computer [#865 (comment)](https://github.com/mavlink/mavros/issues/856#issuecomment-345217529).