mavlink/mavros #812: "Waypoint Message" is_current should be uint8_t ,not bool

Ahrovan, 2017-09-17:
[MISSION_ITEM ](http://mavlink.org/messages/common#MISSION_ITEM) define current as uint8_t type, and in ardupilot for send MISSION_ITEM like qgc:

`missionItem.current =           altChangeOnly ? 3 : 2;
`
need to be uint8_t ,not bool. Reference to issues [number #772 ](https://github.com/mavlink/mavros/issues/772)

### Issue details

The purpose is sending a MISSION_ITEM command by the mavros, which copter goes to particular  lat,lon in GUIDED flight mode

### MAVROS version and platform

Mavros: 0.18.4
ROS: Kinetic
Ubuntu: 16.04
[ OK] ArduPilot (LAST)

khancyr, 2017-09-17:
This is deviation from the mavlink specification for ardupilot that shouldn't be use anymore, so the current implentation on mavros is correct. 
For goto message, you need to use the proper setpoint.

vooon, 2017-09-17:
I cannot imagine usage of that special values of `current` in mission list.
Only where it might be used — GOTO like commands (when you send `MISSION_ITEM` not from your wp list).
But that is soo quirky and from pre setpoint era. I assume that better to check and implement [`SET_POSITION_TARGTET_GLOBAL_INT`][1]. Because it should be simpliest way to upgrade handling of special `MISSION_ITEM`s.

[1]: http://mavlink.org/messages/common#SET_POSITION_TARGET_GLOBAL_INT

Ahrovan, 2017-09-17:
Thank you very much @khancyr @vooon  
Ok. problem solved with setpoint_raw (#772)

closed 2017-09-17 by vooon