mavlink/mavros #815: Can't connect to mavproxy broadcast

kdkalvik, 2017-09-19:
Hi,  I have a PixHawk connected to a pi running mavproxy and I am trying to connect to the pixhawk using mavros over wifi. But I get the following error 

[ INFO] [1505794148.244756780]: Plugin waypoint initialized
[ INFO] [1505794148.244806806]: Built-in SIMD instructions: SSE, SSE2
[ INFO] [1505794148.244832336]: Built-in MAVLink package version: 2017.9.9
[ INFO] [1505794148.244860125]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1505794148.244884133]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ INFO] [1505794148.572600421]: udp0: Remote address: 192.168.1.17:14550
[mavros_set_stream-3] process has finished cleanly
log file: /home/kalvik/.ros/log/482f99aa-9cf0-11e7-ba5a-4c3488afce59/mavros_set_stream-3*.log
[ INFO] [1505794149.237649372]: CON: Got HEARTBEAT, connected. FCU: No valid autopilot
[ WARN] [1505794151.240314760]: VER: broadcast request timeout, retries left 4
[ WARN] [1505794152.241231238]: VER: broadcast request timeout, retries left 3
[ WARN] [1505794153.239805177]: VER: unicast request timeout, retries left 2
[ WARN] [1505794154.239789623]: VER: unicast request timeout, retries left 1
[ WARN] [1505794155.239556822]: VER: unicast request timeout, retries left 0
[ WARN] [1505794156.241119686]: VER: your FCU don't support AUTOPILOT_VERSION, switched to default capabilities
[ INFO] [1505794159.239002097]: HP: requesting home position
[ WARN] [1505794160.239625774]: PR: request list timeout, retries left 2
[ WARN] [1505794161.240753795]: PR: request list timeout, retries left 1
[ WARN] [1505794162.241407198]: PR: request list timeout, retries left 0
[ WARN] [1505794165.239345261]: WP: timeout, retries left 2
[ WARN] [1505794166.240063826]: WP: timeout, retries left 1
[ WARN] [1505794167.240709058]: WP: timeout, retries left 0
[ERROR] [1505794168.241576605]: WP: timed out.
[ INFO] [1505794169.237920916]: HP: requesting home position
[ INFO] [1505794179.237805981]: HP: requesting home position

I have ArduSub flashed on the pixhawk. But I don't think that is the issue I flashed the pixhawk with ArduCopter and got the same issue and I am able to connect to the pixhawk with QGroundControl. So mavproxy is working.

TSC21, 2017-09-19:
Please follow the issue template. Also why do you have mavproxy working simultaneously with mavros?

kdkalvik, 2017-09-19:
I have mavproxy on my companion(Pi) and mavros on my surface machine. I asked around in gitter and was told it is possible. And I don't want to run ROS on the Pi.

TSC21, 2017-09-19:
But again, if you don't follow what is asked on the issue template, we are not able to help you.

kdkalvik, 2017-09-19:
Fixed it. Sorry about that.

kdkalvik, 2017-09-19:
So, is it not possible to connect mavros to a MAVProxy?

TSC21, 2017-09-19:
> ERROR. I got 1 addresses, but not your target 1:1

There is your problem.

kdkalvik, 2017-09-19:
So fcu_url is wrong?

TSC21, 2017-09-19:
your `tgt_sys` is wrong.

kdkalvik, 2017-09-19:
I just checked my pixhawk's parameters and it's SYSID_THISMAV parameter's value is 1. which is what I am using.

TSC21, 2017-09-19:
what about your `tgt_comp`?

kdkalvik, 2017-09-19:
I don't know I couldn't find the parameter in the list from ardusub. do you know the parameter name? @TSC21

TSC21, 2017-09-19:
I'm not an APM user but I suppose a quick check on the mavlink publisher on the Firmware side would help you.

kdkalvik, 2017-09-19:
I guess I will look into it. Thanks. Looks like this is not an issue with mavros but you mind if I leave it open until I make sure of it?

kdkalvik, 2017-09-19:
Ok I check my pixhawk parameters. My system gcs id is 255 and mavlink id is 1. On mavros I am using 255 as the system id and 1 as the target. and I get this error for checkid

```
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/timer.py", line 223, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/opt/ros/kinetic/lib/mavros/checkid", line 74, in timer_cb
    tgt_ids[0], tgt_ids[1]))
TypeError: %d format: a number is required, not str

```

kdkalvik, 2017-09-20:
Got it mavros was talking to itself.

closed 2017-09-20 by kdkalvik

sudosurf, 2018-04-20:
@kdkalvik I'm having the same issue, can you explain how to implement your solution? My px4 parameters on pixhawk are set so sys id, comp = 1,1, and GCS are QGC defaults 255,(I don't know default comp value).

I posted my errors and diagnostics here: https://github.com/mavlink/mavros/issues/1010

huhu1030, 2019-04-18:
Same here if someone can help, I've been looking for answers for 2 days now.
System id is set to 1 (check with QGroundControl), target set to 1 (in launch file) and GCS is default to 255. But I have the same weird issue "ERROR. I got 1 addresses, but not your target 1:1".