mavlink/mavros #818: Inconsistent global altitude values between mavros topics in STIL

lamping7, 2017-09-20:
### Issue details

Look at the node log section below and observe the differences in the reported altitude. I'm not sure where the issue lies, but I did see some changes relatively recently to the global_position plugin. 

### MAVROS version and platform

Mavros: 0.21 deb pkg
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[  ] ArduPilot
[ X ] PX4

Version: 1.6.5

### Node logs

```
$ rostopic echo -c /mavros/altitude
header: 
  seq: 8318
  stamp: 
    secs: 835
    nsecs: 361787484
  frame_id: map
monotonic: 0.132097646594
amsl: 115.11151123
local: 0.0135146770626
relative: 0.0135116577148
terrain: 115.103996277
bottom_clearance: 0.00751495361328
---

$ rostopic echo -c /mavros/global_position/global
header: 
  seq: 38551
  stamp: 
    secs: 774
    nsecs: 662675872
  frame_id: map
status: 
  status: 0
  service: 1
latitude: 39.133459
longitude: -84.36502
altitude: 80.2207765691
position_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
position_covariance_type: 1
---

$ rostopic echo -c /mavros/vfr_hud
header: 
  seq: 678
  stamp: 
    secs: 70
    nsecs: 290000000
  frame_id: ''
airspeed: 0.0
groundspeed: 0.00551259331405
heading: 72
throttle: 0.0
altitude: 115.109169006
climb: 0.0658637434244
---
```

### Diagnostics

```
$ rostopic ech-n1 /diagnostics
header: 
  seq: 599
  stamp: 
    secs: 619
    nsecs:         0
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: connected
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Received packets:
        value: 43870
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 123
      - 
        key: Tx sequence number:
        value: 35
      - 
        key: Rx total bytes:
        value: 12396158
      - 
        key: Tx total bytes:
        value: 319655
      - 
        key: Rx speed:
        value: 19940.000000
      - 
        key: Tx speed:
        value: 517.000000
  - 
    level: 0
    name: mavros: GPS
    message: 3D fix
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Satellites visible
        value: 10
      - 
        key: Fix type
        value: 3
      - 
        key: EPH (m)
        value: 0.00
      - 
        key: EPV (m)
        value: 0.00
  - 
    level: 0
    name: mavros: Heartbeat
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Heartbeats since startup
        value: 617
      - 
        key: Frequency (Hz)
        value: 1.047619
      - 
        key: Vehicle type
        value: Octorotor
      - 
        key: Autopilot type
        value: PX4 Autopilot
      - 
        key: Mode
        value: AUTO.LOITER
      - 
        key: System status
        value: Standby
  - 
    level: 0
    name: mavros: System
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Sensor present
        value: 0x00000000
      - 
        key: Sensor enabled
        value: 0x00000000
      - 
        key: Sensor helth
        value: 0x00000000
      - 
        key: CPU Load (%)
        value: 0.0
      - 
        key: Drop rate (%)
        value: 0.0
      - 
        key: Errors comm
        value: 0
      - 
        key: Errors count #1
        value: 0
      - 
        key: Errors count #2
        value: 0
      - 
        key: Errors count #3
        value: 0
      - 
        key: Errors count #4
        value: 0
  - 
    level: 0
    name: mavros: Battery
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Voltage
        value: 24.30
      - 
        key: Current
        value: -1.0
      - 
        key: Remaining
        value: 100.0
  - 
    level: 0
    name: mavros: Time Sync
    message: Normal
    hardware_id: udp://:14540@localhost:14557
    values: 
      - 
        key: Timesyncs since startup
        value: 6163
      - 
        key: Frequency (Hz)
        value: 10.000000
      - 
        key: Last dt (ms)
        value: 0.238959
      - 
        key: Mean dt (ms)
        value: 0.219913
      - 
        key: Last system time (s)
        value: 618.363718000
      - 
        key: Time offset (s)
        value: 0.536377583
---
```

### Check ID

```
$ rosrun mavros checkid
OK. I got messages from 1:1.

---
Received 7427 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 24, 30, 31, 32, 33, 74, 76, 83, 85, 87, 230, 105, 111, 241, 242, 245
```

TSC21, 2017-09-20:
That's because global_position/global has its altitude WRT WGS-84 ellipsoid and the altitude on altitude publisher is the height over the geoid (amsl). So it is no issue, it follows the proper convetion (you can see NavSatFix docs).

lamping7, 2017-09-20:
Thanks... either I missed it before or that change came with other geolib stuff.

closed 2017-09-20 by lamping7