mavlink/mavros #819: Switch to offboard

wwwefwehfrgjnrek, 2017-09-21:
Hi everyone, I used mavros to switch to offboard mode to encounter some problems.And got some messages like that
 [ INFO] [1505965077.413415981]: FCU: no offboard
[ERROR] [1505965077.464574476]: FCU: failsafe mode on
[ERROR] [1505965078.201263885]: FCU: failsafe mode off
[ WARN] [1505965078.390183987]: PR: request param #25 timeout, retries left 2, and 189 params still missing
[ INFO] [1505965078.720566982]: FCU: no offboard
[ERROR] [1505965078.772950419]: FCU: failsafe mode on
[ WARN] [1505965079.393036151]: PR: request param #26 timeout, retries left 2, and 188 params still missing
[ INFO] [1505965079.617096375]: Offboard Mode
[ INFO] [1505965080.030280157]: FCU: no offboard
[ WARN] [1505965080.401756279]: PR: request param #27 timeout, retries left 2, and 187 params still missing
[ERROR] [1505965080.808562478]: FCU: failsafe mode off
[ INFO] [1505965081.334248742]: FCU: no offboard
I have already advertise topic /mavros/setpoint_position/local ,but the average rate is too low,the specific information is as follows,
average rate: 0.787
	min: 1.271s max: 1.271s std dev: 0.00000s window: 2
average rate: 0.762
	min: 1.271s max: 1.354s std dev: 0.04128s window: 3
no new messages
average rate: 0.779
	min: 1.225s max: 1.354s std dev: 0.05320s window: 4
average rate: 0.779
	min: 1.225s max: 1.354s std dev: 0.04608s window: 5
average rate: 0.781
	min: 1.225s max: 1.354s std dev: 0.04147s window: 6
average rate: 0.777
	min: 1.225s max: 1.354s std dev: 0.04075s window: 7
I donot know why like that,perhaps the reason is connection.However, when I commented piece of code.Everything is so good.I've pasted some code.I think it might be a problem with the frequency of the advertise.
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
  std::string cam_topic(vk::getParam<std::string>("svo/cam_topic", "/usb_cam/image_raw")); //接收图像话题信息
 
  image_transport::ImageTransport it(nh);
  image_transport::Subscriber it_sub = it.subscribe(cam_topic, 5, &VoNode::imgCb, &vo_node);
  std::string cam_topic(vk::getParam<std::string>("svo/cam_topic", "/usb_cam/image_raw")); 
 
  image_transport::ImageTransport it(nh);
  image_transport::Subscriber it_sub = it.subscribe(cam_topic, 5, &VoNode::imgCb, &vo_node);
  vo_node.sub_remote_key_ = nh.subscribe("uav/remote_key", 5, &VoNode::remoteKeyCb, &vo_node);
//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
The above part of the code which is commented out.
  setPositionPublisher = nh.advertise<geometry_msgs::PoseStamped>("/mavros/setpoint_position/local",10);
  currentViconPositionPublisher = nh.advertise<geometry_msgs::PoseStamped> ("/mavros/mocap/pose",10);
  ros::Subscriber sub = nh.subscribe("vicon/liu/liu",1000,svo::Receiving::calc);      
  ros::Subscriber sub1 = nh.subscribe("vicon/liu/liu",1000,&viconPositionReceived);   
  ros::Subscriber stateSubscriber = nh.subscribe("mavros/state", 10, stateReceived);
  ros::Subscriber localPositionSubsciber = nh.subscribe("/mavros/local_position/pose", 10, localPositionReceived);
  ros::Subscriber commandSubscriber = nh.subscribe("/keyboard/keydown",1,sendCommand);
  arming_client = nh.serviceClient<mavros_msgs::CommandBool>("mavros/cmd/arming");   
  arm_cmd.request.value = false;
  isLockLocalPos = false; 
  hasSet = false; 
  
  pos_value = 0.2f;
  tmpRoll = 0.0;
  tmpPitch = 0.0;
  tmpYaw= 0.0;

  ros::Rate loopRate(50.0);
  while(ros::ok())
  {    
     if(hasSet)
     {       
         ps.header.seq++;
         ps.header.stamp = ros::Time::now();
         setPositionPublisher.publish(ps);
     }
     uavCurrentViconPose.header.stamp = ros::Time::now();
     uavCurrentViconPose.header.frame_id = "/world";
     currentViconPositionPublisher.publish(uavCurrentViconPose);put
  vo_node.sub_remote_key_ = nh.subscribe("uav/remote_key", 5, &VoNode::remoteKeyCb, &vo_node);

  // subscribe to uav position
  setPositionPublisher = nh.advertise<geometry_msgs::PoseStamped>("/mavros/setpoint_position/local",10);
  currentViconPositionPublisher = nh.advertise<geometry_msgs::PoseStamped> ("/mavros/mocap/pose",10);
  ros::Subscriber sub = nh.subscribe("vicon/liu/liu",1000,svo::Receiving::calc);      
  ros::Subscriber sub1 = nh.subscribe("vicon/liu/liu",1000,&viconPositionReceived);   
  ros::Subscriber stateSubscriber = nh.subscribe("mavros/state", 10, stateReceived);
  ros::Subscriber localPositionSubsciber = nh.subscribe("/mavros/local_position/pose", 10, localPositionReceived);
  ros::Subscriber commandSubscriber = nh.subscribe("/keyboard/keydown",1,sendCommand);

  arming_client = nh.serviceClient<mavros_msgs::CommandBool>("mavros/cmd/arming");   

  arm_cmd.request.value = false;
  isLockLocalPos = false; 
  hasSet = false; 
  
  pos_value = 0.2f;
  tmpRoll = 0.0;
  tmpPitch = 0.0;
  tmpYaw= 0.0;
  ros::Rate loopRate(50.0);
  while(ros::ok())
  {    
     if(hasSet)
     {       
         ps.header.seq++;
         ps.header.stamp = ros::Time::now();
         setPositionPublisher.publish(ps);
     }
     uavCurrentViconPose.header.stamp = ros::Time::now();
     uavCurrentViconPose.header.frame_id = "/world";
     currentViconPositionPublisher.publish(uavCurrentViconPose);
}
I hope someone can answer this question!!

AlexisTM, 2017-09-21:
Can you check that you are in Offboard mode? See: https://dev.px4.io/en/ros/mavros_offboard.html

The topic `/mavros/setpoint_position/local` is coming from your computer. If you are sending it at 50Hz, you will see it at 50Hz. Therefore, I would say the `cvWaitKey(33)` command is blocking.

vooon, 2017-09-21:
Param rerequests - seems that you have lossy link to PX. Check that you have enough bandwidth, then check wiring.
How you setup pixhawk?

wwwefwehfrgjnrek, 2017-09-21:
Yeah，I am sure in the Offboard mode. Because  when I note a part of the code,and then view the topic of
/mavros/state,the message is header: 
  seq: 272
  stamp: 
    secs: 1505996297
    nsecs: 313893168
  frame_id: ''
connected: True
armed: True
guided: True
mode: OFFBOARD.
And I refer to the URL you give, add the for loop to the topic /mavros/setpoint_position/local. Then got the message like that
average rate: 23.109
	min: 0.000s max: 18.983s std dev: 0.48019s window: 2905
average rate: 22.885
	min: 0.000s max: 18.983s std dev: 0.48065s window: 2906
no new messages
average rate: 23.200
	min: 0.000s max: 18.983s std dev: 0.47492s window: 2994
average rate: 23.165
	min: 0.000s max: 18.983s std dev: 0.47352s window: 3014
average rate: 23.336
	min: 0.000s max: 18.983s std dev: 0.47038s window: 3056
average rate: 23.417
	min: 0.000s max: 18.983s std dev: 0.46782s window: 3094
average rate: 23.508
	min: 0.000s max: 18.983s std dev: 0.46635s window: 3114
average rate: 23.664
	min: 0.000s max: 18.983s std dev: 0.46350s window: 3156
average rate: 23.554
	min: 0.000s max: 18.983s std dev: 0.46291s window: 3171
no new messages
average rate: 23.901
	min: 0.000s max: 18.983s std dev: 0.45753s window: 3263
The average rate is ok.But it still have no new messages .I don't know why. If i note a part of the code which is subscribe to the package of usb_cam's topic information and the mouse key's topic information.This phenomenon does not happen. I think it may be due to no enough bandwidth.

TSC21, 2017-10-03:
@wwwefwehfrgjnrek is this solved to you?

wwwefwehfrgjnrek, 2017-10-03:
@TSC21 NO

TSC21, 2017-11-05:
@wwwefwehfrgjnrek since you are not giving any more feedback, I'm closing this as stale. We can reopen then if you can give us feedback on what you tried in the mean time.

closed 2017-11-05 by TSC21