mavlink/mavros #820: LIbmavconn compile error on Mavlink binary distribution

Corusco, 2017-09-21:
### Issue details

I use a custom blend of Mavros based on 0.18.0. It has been working among several members of our team since March of this year. Last week I set up a new installation on a new machine using the recent 2017.9.9 binary distribution of Mavlink, which is our typical setup. I now get a compile error when building the mavlink_helpers.cpp.o object file, on line 781 of the mavlink_helpers.h file: `'MAVLINK_START_SIGN_STREAM' was not declared in this scope`. The same error is on `MAVLINK_END_SIGN_STREAM`. When compiling the current 0.18.9 Mavros (no mods), everything works fine. I should stress that no changes have been made to our custom version since July, and we've done fresh installations as recently as mid-August without incident.

Barring any update for backward compatibility, I'm just looking for some understanding what has happened over the last several releases of both Mavros and Mavlink so I can integrate the changes without having to do a full update of our version at this time. Any input on where to get the declarations of these funcs back in our code?

### MAVROS version and platform

Mavros: 0.18.0 with custom code to publish and accept NED 
ROS: Kinetic
Ubuntu: 16.04

AlexisTM, 2017-09-22:
The Mavlink version you are using should be set to a specific version. Indeed, the releases are not simultaneous.

The mavros version 0.18.0: https://github.com/mavlink/mavros-release#mavros-kinetic---0180-0

probably match with either:

* Mavlink 2016.6.24-0 https://github.com/mavlink/mavlink-gbp-release#mavlink-kinetic---2016624-0
* Mavlink 2016.6.22-1 https://github.com/mavlink/mavlink-gbp-release#mavlink-kinetic---2016622-1

vooon, 2017-09-22:
I recommend to update. It is always better to strip custom plugins into custom package.
Or backport https://github.com/mavlink/mavros/commit/2c4604d78c17cede52713cceeef76188a028c05d

TSC21, 2017-09-28:
@Corusco is this solved to you?

Corusco, 2017-10-02:
I ended up rewriting to accommodate the update, as there was quite a bit of new code we wanted to try out. Much appreciation to Vooon and AlexisTM for your input.

closed 2017-10-02 by Corusco