mavlink/mavros #825: High latency with intel joule

gui2dev, 2017-09-26:
This is only bug and feature tracker, please use it
to report bugs or request features.

----

### Issue details

I'm using mavros on an intel joule. I am experiencing very high latency and timeouts with waypoints and param plugins. My first guess is that it's hardware related, because changing the computer that executes the launch file seems to solve the issue. In addition to that, uploading waypoints from UGS seems flawless. 

### MAVROS version and platform

Mavros: 0.21.2
ROS: Kinetic
Ubuntu: 16.04

### Autopilot type and version

[X] ArduPilot
[  ] PX4

Version: 3.5

### Node logs

```
roslaunch falcon_v2 start.launch 
... logging to /home/gdev/.ros/log/e337335a-a285-11e7-9847-a0c5893bcefa/roslaunch-jBoard-3312.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jBoard:45966/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /alexmos/baudrate: 115200
 * /alexmos/device: /dev/ttyUSB0
 * /falcon/altitude: 100.0
 * /falcon/distance: 18000.0
 * /falcon/heading: 270.0
 * /falcon/log_path: /media/gdev/7976-...
 * /falcon/mission_path: /media/gdev/7976-...
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 0.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 0.0
 * /mavros/distance_sensor/rangefinder_pub/field_of_view: 0.0
 * /mavros/distance_sensor/rangefinder_pub/frame_id: lidar
 * /mavros/distance_sensor/rangefinder_pub/id: 0
 * /mavros/distance_sensor/rangefinder_pub/send_tf: False
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/rangefinder_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/rangefinder_sub/id: 1
 * /mavros/distance_sensor/rangefinder_sub/orientation: PITCH_270
 * /mavros/distance_sensor/rangefinder_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
[... 575 lines cut ...]
    hardware_id: /dev/ttyACM0:57600
    values: 
      - 
        key: Voltage
        value: 24.95
      - 
        key: Current
        value: 0.0
      - 
        key: Remaining
        value: 99.0
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 84 messages, from 1 addresses
sys:comp   list of messages
  1:1     40, 0, 253, 22, 47
```

TSC21, 2017-09-26:
`/dev/ttyACM0:57600` -> try to increase to baudrate to 921600. Also set `SERIAL1_BAUD` to `921`.

gui2dev, 2017-09-26:
Thanks for the answer. I just tried it, without success. And I guess it's `SERIAL0_BAUD` that I should change as I'm connected to my pixhawk 2 USB port ?

gui2dev, 2017-09-26:
Putting waypoint plugin output level to DEBUG, I got something like this:
`
[DEBUG] [1506422968.584678109]: WP: send item #232  F:0 C:115 p: 135.000000 0.000000 -1.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[DEBUG] [1506422968.593246869]: WP: FCU requested waypoint 232
[DEBUG] [1506422968.593594694]: WP: send item #232  F:0 C:115 p: 135.000000 0.000000 -1.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[DEBUG] [1506422968.598412674]: WP: FCU requested waypoint 233
[DEBUG] [1506422968.598799423]: WP: send item #233  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: 45.051233 y: 5.800261 z: 100.000000
[DEBUG] [1506422968.604025764]: WP: Received INVALID_SEQUENCE ack
[DEBUG] [1506422969.610167177]: WP: FCU requested waypoint 234
[DEBUG] [1506422969.610542945]: WP: send item #234  F:0 C:115 p: 135.000000 0.000000 -1.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[DEBUG] [1506422969.616371093]: WP: FCU requested waypoint 234
[DEBUG] [1506422969.616760118]: WP: send item #234  F:0 C:115 p: 135.000000 0.000000 -1.000000 0.000000 x: 0.000000 y: 0.000000 z: 0.000000
[DEBUG] [1506422969.621158504]: WP: FCU requested waypoint 235
[DEBUG] [1506422969.621405778]: WP: send item #235  F:3 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: 45.051282 y: 5.800336 z: 100.000000
[DEBUG] [1506422969.626836776]: WP: Received INVALID_SEQUENCE ack
`

So my first guess is that's transfer is somehow ok, but just very slow.

vooon, 2017-09-26:
Lets calculate, 605 waypoints it is: `605 (MISSION_INTEM + MISSION_REQUEST) + MISSION_REQUEST_LIST + MISSION_COUT` bytes to exchange.

`605 * (38 + 10 + 5 + 10) + 3 + 10 + 5 + 10 = 38143` bytes, for 115200 you can pass about 10 kiB/sec.
So transmission takes at least 3 sec. Hm that's look okay, 10x less than default timeout (30 sec).

Lanch htop and look on CPU load.

vooon, 2017-09-26:
Double requesting of same wp item seems invalid, check wiring.

TSC21, 2017-09-26:
> 115200 you can pass about 10 kiB/sec.
Is serial connection is set to 57600.

TSC21, 2017-09-26:
> Thanks for the answer. I just tried it, without success. And I guess it's SERIAL0_BAUD that I should change as I'm connected to my pixhawk 2 USB port ?

Try both.

gui2dev, 2017-09-26:
Actually, it does not seems to be linked with connectivity at whatever speed I have the same issue while connected to the joule, but not when connected to another machine (my laptop).

It's really as if each first try of sending WP times out.
Setting `WP_TIMEOUT_MS` to `25` let me upload all the waypoints with only one timeout.

What I also don't understand is that the waypoint plugin seems to be not happy with my first WP (takeoff) as I got this INFO:

`[ INFO] [1506423994.490924413]: WP: item #0  F:0 C: 16 p: 0.000000 0.000000 0.000000 0.000000 x: 45.051647 y: 5.801180 z: 328.070007`

When my first waypoint should be a takeoff:

`
currentWps.clear();
[...]
  mavros_msgs::Waypoint wpTakeOff;
  wpTakeOff.frame = 3;
  wpTakeOff.command = 22;  // MAV_CMD_NAV_TAKEOFF;
  wpTakeOff.param4 = 0.0;
  wpTakeOff.x_lat = 0.0;
  wpTakeOff.y_long = 0.0;
  wpTakeOff.z_alt = global_alt;
  wpTakeOff.autocontinue = true;
  currentWps.push_back(wpTakeOff);
`

TSC21, 2017-09-26:
@JMare could any of your changes have influence here?

gui2dev, 2017-09-26:
The only think I added to the original package is that calls a service with the waypoint sequence number on 'ITEM_MISSION_REACHED' form the the appropriate handler

JMare, 2017-09-26:
So @gui2dev first off waypoint 0 is home position so your takeoff waypoint should be in position 1.

This is curious. The handling of `WP: Received INVALID_SEQUENCE ack` is something that we added in #798 , but the previous behaviour was to abort the transfer in this circumstance.

And the order of the debug messages seems strange here.

@gui2dev you mention you have modified the wp plugin, can you show us the branch you are working off?

Its worth noting that parameter and waypoint transfers are surprisingly intolerant of any significant link loss. From my testing even 10% link loss makes significant transfers basically impossible.

Are you able to show us the code that is calling the WaypointPush service?

```
[ WARN] [1506408178.311390820]: WP: timeout, retries left 2
[ WARN] [1506408180.320554139]: WP: timeout, retries left 2
[ WARN] [1506408182.331194313]: WP: timeout, retries left 2
[ WARN] [1506408184.343025752]: WP: timeout, retries left 2
[ WARN] [1506408186.354143266]: WP: timeout, retries left 2
[ WARN] [1506408188.365744648]: WP: timeout, retries left 2
[ WARN] [1506408190.375305900]: WP: timeout, retries left 2
[ WARN] [1506408192.383309896]: WP: timeout, retries left 2
[ WARN] [1506408194.392562067]: WP: timeout, retries left 2
```

I don't know why this is happening but it looks bad.

@JamesStewy what do you think of this

JamesStewy, 2017-09-27:
> the order of the debug messages seems strange here.

@gui2dev, do the DEBUG log messages consistently follow the pattern you posted? The pattern being:

1.  send item #x
2.  FCU requested waypoint x
3.  send item #x
4.  FCU requested waypoint x+1
5.  send item #x+1
6.  Received INVALID_SEQUENCE ack
7.  Repeat with x=x+2

> Its worth noting that parameter and waypoint transfers are surprisingly intolerant of any significant link loss. From my testing even 10% link loss makes significant transfers basically impossible.

I also have to echo this. We changed the waypoint module to ignore the `INVALID_SEQUENCE` ack to increase the fault tolerance, as, despite the retry mechanism implemented at both ends, dropping a single message _could_ cause the waypoint transfer to fail completely. Just looking at that small snippet of debug logs suggests there are messages that are failing to be sent. If this test is run on a mavros version without our changes I would suspect the transfer would fail long before you get to the 232nd waypoint.

gui2dev, 2017-09-27:
@jmare I issued a pull request -- #826 
@JamesStewy the pattern is somehow the same, sometimes one waypoint is accepted without receiving` INVALID_SEQUENCE` ack. I first worked on a version that would not include the `INVALID_SEQUENCE` ignore. At that time transfer just failed after first waypoint.
After updating my sources, transfer would just timeout on `LIST_TIMEOUT_MS`, so I started to play with the timeouts duration to allow a full transfer.
As it only happens with the intel joule but not when using SITL, I am starting to think that it's somehow kernel-related.

JMare, 2017-09-27:
@gui2dev It seems more likely to me that the issue only occurs on low powered computers, than it being specific to the joule.

If you share the code you are using to call the WaypointPush service from I could test it on a Raspberry Pi or Odroid and see if I can replicate.

gui2dev, 2017-09-27:
@JMare that's the other option. You can get the code here: https://github.com/gui2dev/mavros

lamping7, 2017-09-27:
Just to share my experience, our team had no issues when we used the Joule in the past with:
- PX4 - v1.5.5
- Mavros - 0.18.7
- ROS - Kinetic
- Ubuntu - 16.04
We were using a Pixhawk 2.1 connected via UART on the GPIO @ 921600 baud

gui2dev, 2017-09-27:
Did you used pullups?

lamping7, 2017-09-27:
> Did you used pullups?

I'm not sure what you're asking.

gui2dev, 2017-09-27:
@lamping7 GPIO levels are at 1.8V right? Or maybe you were not using the dev board.

lamping7, 2017-09-27:
@gui2dev From my understanding, the Joule itself is 1.8v, but the dev kit expansion board is 3.3v. We **didn't** use a level shifter, like one you'd use for the OROID XU4, for example.

gui2dev, 2017-09-27:
@lamping7, so we have a similar use of the joule + dev kit except you're using serial while I'm using usb. We are using gpio 55 (j13 15) for trigging purposes and high level is definitely 1.8V. Maybe 22/24 is 3.3V.

lamping7, 2017-09-27:
@gui2dev Checkout the Intel post above. We didn't use any of the other GPIO, but again, we had no communication issues the way we had it setup once we figured out **how** to do it. You need the Ubuntu flavor packed with the Joule specific kernel in order to use the UART on `/dev/ttyS1` for the pins I listed previously. 

Another, but unrelated, issue that we found was that using a USB type C hub connected to the Joule introduced odd EMI to the system, causing GPS signal loss and compass errors.

gui2dev, 2017-09-27:
@lamping7 Thanks for the link. I relied on an intel reference for  [GPIO Signal Termination and Conditioning](https://software.intel.com/en-us/node/731191)
We are using the type A with a usb hub and we did not observe EMI issues (pixhawk 2.1).

gui2dev, 2017-09-27:
@lamping7 I am also running a joule flavored kernel (latest I guess) along with latest bios.

lamping7, 2017-09-27:
@gui2dev No problem. I'd see if your problem exists with that hardware interface. Should be a quick test if you have the correct software, which you say you do. Good luck. 

Also, we switched to the type A and had no issues with interference.

gui2dev, 2017-09-27:
@lamping7 allright. By any chance did you had any connectivity issue when using usb connection to pixhawk ?

lamping7, 2017-09-27:
@gui2dev We used a USB FTDI serial converter connected from the Joule USB A port to the telem2 on the Pixhawk at first and had no issues @ 921600 baud.

TSC21, 2017-11-05:
@gui2dev is this solved to you already?

gui2dev, 2017-11-07:
sorry @TSC21 I've been too busy to make tests again. It really looks as if it's power related, I'll confirm asap, but feel free to close the issue if I'm taking too long.

TSC21, 2017-11-08:
We can wait some days. We just usually don't like to keep the bug tracking stalled for long (except with feature requests).

TSC21, 2017-12-23:
@gui2dev I'm closing this as stale. Let us know if this still persists.

closed 2017-12-23 by TSC21