mavlink/mavros #827: How to get the feedback of attiude angles in Body frame

mohit004, 2017-09-28:
Dear Developers,

Actually, I am in the process of developing the low-level controller for my tilt-rotor tricopter. Regarding that, I would like to get the feedback of the attitude angles in the Body frame. From my intuition, I thought that the feedback of orientation in /mavros/imu/data topic would be in the Body frame, but to my surprise, the orientation feedback is in the global ENU frame. Is there any way to get the feedback of the attitude angle in Body frame right away?

Thanks in advance for your help!
Mohit

TSC21, 2017-09-28:
https://github.com/mavlink/mavros/blob/master/mavros/src/lib/uas_data.cpp#L130

closed 2017-09-28 by TSC21

mohit004, 2017-09-29:
Hi @TSC21,

Thanks for your reply!
Actually, I am sorry that I couldn't get it completely. Are you trying to say that I need to modify that part of the 'uas_data.cpp' file? Or do you mean that the orientation data coming is already in the local body frame?
Just to make myself clear, by local Body frame I mean the local bank and pitch angle of the UAV.

Thanks once again for your help!
Mohit

TSC21, 2017-09-29:
You don't have to modify nothing. You just have to call `m_uas->get_attitude_imu_ned()`, where `m_uas` is a UAS object.