mavlink/mavros #833: setpoint_attitude still not publishing attitude in latest mavros

siddhu45, 2017-10-11:
Hi guys! @andresR8 @TSC21 @TSC21 @vooon
I followed your work and thanks for all your effort as you posted in #821 and #817. I tried to use the `/mavros/setpoint_attitude/` to publish attitude angles and throttle. I was not aware of the change in topic names and other features but after following this I found out that it has changed. Even with the latest `mavros` source install, I am not able to use this topic. I publish to `mavros/setpoint_attitude/thrust` and `/mavros/setpoint_attitude/target_attitude` topics to send quaternion and thrust command but `mavros` only subscribes to the thrust and not the attitude as can be seen from the `rqt_graph` below:

[image]

It is publishing the commands because when I run `rostopic echo /mavros/setpoint_attitude/target_attitude` I see the commands. I tried all the things you guys mentioned related to this. Changed the `use_quaternion` in the px4_config.yaml, tried

> can you please try to replace https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/setpoint_attitude.cpp#L112-L113 with std::shared_ptrs and recompile?

didnt work. And the other minor changes. Can you please let me know if you were able to solve this issue, how did you do it. That would be great. Thanks a lot.

PS: Previously I was using the debian built mavros and even the throttle was not publishing but just then I came across this issue and did source installation and atleast was able to send something but not all. Anyways, it would be great if you help me out here.

### MAVROS version and platform

Mavros: ?0.21.2?
ROS: ?Kinetic?
Ubuntu: ?16.04?

### Autopilot type and version
[ X ] PX4

Version: 1.6.5dev

### Node logs
```
rostopic info /mavros/setpoint_attitude/target_attitude
Type: geometry_msgs/PoseStamped

Publishers: 
 * /pub_setpoints (http://fnn-ST:44347/)

Subscribers: 
 * /mavros (http://fnn-ST:36699/)
```
```
rostopic info /mavros/setpoint_attitude/thrust 
Type: mavros_msgs/Thrust

Publishers: 
 * /pub_setpoints (http://fnn-ST:44347/)

Subscribers: 
 * /mavros (http://fnn-ST:36699/)

```

roslaunch: Also, what is this unexpectedd command at the end. I get it everytime even without sending any commands
```
... logging to /home/fnn/.ros/log/286dbe58-ae39-11e7-bab8-30f77280179f/roslaunch-fnn-ST-11545.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fnn-ST:40747/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
[... 411 lines cut ...]
        value: -0.115670
      - 
        key: Last system time (s)
        value: 1135.656011000
      - 
        key: Time offset (s)
        value: 1507693958.434074163
---

```

### Check ID

```
OK. I got messages from 1:1.

---
Received 5939 messages, from 1 addresses
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 24, 30, 31, 32, 33, 74, 76, 85, 87, 102, 230, 231, 105, 111, 241, 242, 245

```

Let me know if any more info is required.
Thanks. Cheers!

vooon, 2017-10-11:
What this command returns?
```
rosnode info /mavros
```

siddhu45, 2017-10-11:
```
rosnode info /mavros
--------------------------------------------------------------------------------
Node [/mavros]
Publications: 
 * /mavros/manual_control/control [mavros_msgs/ManualControl]
 * /mavros/adsb/vehicle [mavros_msgs/ADSBVehicle]
 * /tf [tf2_msgs/TFMessage]
 * /mavros/radio_status [mavros_msgs/RadioStatus]
 * /mavros/setpoint_raw/target_local [mavros_msgs/PositionTarget]
 * /mavros/imu/atm_pressure [sensor_msgs/FluidPressure]
 * /mavros/altitude [mavros_msgs/Altitude]
 * /mavros/global_position/compass_hdg [std_msgs/Float64]
 * /mavros/hil/actuator_controls [mavros_msgs/HilActuatorControls]
 * /mavros/global_position/global [sensor_msgs/NavSatFix]
 * /mavros/target_actuator_control [mavros_msgs/ActuatorControl]
 * /mavlink/from [mavros_msgs/Mavlink]
 * /mavros/local_position/pose [geometry_msgs/PoseStamped]
 * /mavros/imu/data [sensor_msgs/Imu]
 * /mavros/wind_estimation [geometry_msgs/TwistStamped]
 * /mavros/setpoint_raw/target_global [mavros_msgs/GlobalPositionTarget]
 * /tf_static [tf2_msgs/TFMessage]
 * /mavros/global_position/gp_origin [geographic_msgs/GeoPointStamped]
 * /mavros/imu/temperature [sensor_msgs/Temperature]
 * /mavros/px4flow/raw/optical_flow_rad [mavros_msgs/OpticalFlowRad]
 * /diagnostics [diagnostic_msgs/DiagnosticArray]
 * /mavros/global_position/raw/fix [sensor_msgs/NavSatFix]
 * /mavros/battery [sensor_msgs/BatteryState]
 * /mavros/setpoint_raw/target_attitude [mavros_msgs/AttitudeTarget]
 * /mavros/imu/data_raw [sensor_msgs/Imu]
 * /mavros/mission/waypoints [mavros_msgs/WaypointList]
 * /mavros/cam_imu_sync/cam_imu_stamp [mavros_msgs/CamIMUStamp]
 * /mavros/state [mavros_msgs/State]
 * /mavros/rc/in [mavros_msgs/RCIn]
 * /rosout [rosgraph_msgs/Log]
 * /mavros/global_position/local [nav_msgs/Odometry]
 * /mavros/vfr_hud [mavros_msgs/VFR_HUD]
 * /mavros/px4flow/ground_distance [sensor_msgs/Range]
 * /mavros/px4flow/temperature [sensor_msgs/Temperature]
 * /mavros/global_position/rel_alt [std_msgs/Float64]
 * /mavros/imu/mag [sensor_msgs/MagneticField]
 * /mavros/local_position/odom [nav_msgs/Odometry]
 * /mavros/extended_state [mavros_msgs/ExtendedState]
 * /mavros/hil/controls [mavros_msgs/HilControls]
 * /mavros/local_position/velocity [geometry_msgs/TwistStamped]
 * /mavros/global_position/gp_lp_offset [geometry_msgs/PoseStamped]
 * /mavros/global_position/raw/gps_vel [geometry_msgs/TwistStamped]
 * /mavros/rc/out [mavros_msgs/RCOut]
 * /mavros/home_position/home [mavros_msgs/HomePosition]
 * /mavros/time_reference [sensor_msgs/TimeReference]

Subscriptions: 
 * /mavros/actuator_control [unknown type]
 * /mavros/hil/optical_flow [unknown type]
 * /mavros/global_position/set_gp_origin [unknown type]
 * /tf [tf2_msgs/TFMessage]
 * /mavros/setpoint_raw/attitude [unknown type]
 * /mavros/hil/state [unknown type]
 * /mavros/global_position/home [unknown type]
 * /mavros/setpoint_velocity/cmd_vel [geometry_msgs/TwistStamped]
 * /mavros/hil/gps [unknown type]
 * /mavros/setpoint_accel/accel [unknown type]
 * /mavros/odometry/odom [unknown type]
 * /mavros/mocap/pose [geometry_msgs/PoseStamped]
 * /mavros/hil/imu_ned [unknown type]
 * /mavros/hil/rc_inputs [unknown type]
 * /tf_static [tf2_msgs/TFMessage]
 * /mavros/setpoint_raw/global [unknown type]
 * /mavros/vision_pose/pose_cov [unknown type]
 * /mavros/fake_gps/mocap/tf [unknown type]
[... 80 lines cut ...]
    * direction: inbound
    * transport: INTRAPROCESS
 * topic: /mavros/mocap/pose
    * to: /fake_mocap (http://fnn-ST:33429/)
    * direction: inbound
    * transport: TCPROS
 * topic: /mavros/setpoint_attitude/thrust
    * to: /safe_y6 (http://fnn-ST:37465/)
    * direction: inbound
    * transport: TCPROS
 * topic: /mavros/setpoint_attitude/target_attitude
    * to: /safe_y6 (http://fnn-ST:37465/)
    * direction: inbound
    * transport: TCPROS
 * topic: /mavros/setpoint_position/local
    * to: /safe_y6 (http://fnn-ST:37465/)
    * direction: inbound
    * transport: TCPROS
 * topic: /mavros/setpoint_velocity/cmd_vel
    * to: /safe_y6 (http://fnn-ST:37465/)
    * direction: inbound
    * transport: TCPROS
```

Actually, I formatted my system and did a fresh install of everything and now the `setpoint_atttitude` command seems to work. Still it does not show up in the `rqt_graph`, it only shows `/mavros/setpoint_attitude/thrust` sending commands to `/mavros`. But the UAV goes in offboard and I am able to send commands using it. Sorry, I apologize for the trouble. Thanks for the help and effort @vooon.

closed 2017-10-11 by siddhu45