mavlink/mavros #835: Distance sensor has a hardcoded "fcu" frame

AlexisTM, 2017-10-11:
Problem:
==========

The distance sensor frames are hardcoded. to `fcu` while we changed them all to `base_link` in PR #791

See: https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/distance_sensor.cpp#L255

Solutions:
==========

Adding a parameter `fcu_frame_id` to distance_sensor, yet, it interferes into the yaml parsing.

```yaml
distance_sensor:
  fcu_frame_id: "base_link"
  hrlv_ez4_pub:
    id: 0
    frame_id: "hrlv_ez4_sonar"
    orientation: ROLL_180 # RPY:{180.0, 0.0, 0.0}
    field_of_view:  !degrees 3
    send_tf: true
    sensor_position: {x:  0.0, y:  0.0, z:  -0.1}
```

Or we can add a `frame_id` and `child_frame_id` parameters for every sensor. But it interferes with the current config of users if they just update to the new version. 

```yaml
distance_sensor:
  fcu_frame_id: "base_link"
  hrlv_ez4_pub:
    id: 0
    orientation: ROLL_180 # RPY:{180.0, 0.0, 0.0}
    field_of_view: !degrees 3
    send_tf: true
    sensor_position: {x:  0.0, y:  0.0, z:  -0.1}
    frame_id: "base_link"                            # UAV frame_id
    child_frame_id: "hrlv_ez4_sonar" # Distance sensor frame
```

Adding the hardcoded `base_link` is less versatile for custom setups.

AlexisTM, 2017-10-11:
A last solution can be to split parameters for TF: 

```yaml
distance_sensor:
  fcu_frame_id: "base_link"
  hrlv_ez4_pub:
    id: 0
    orientation: ROLL_180 # RPY:{180.0, 0.0, 0.0}
    field_of_view: !degrees 3
    sensor_position: {x:  0.0, y:  0.0, z:  -0.1}
    frame_id: "hrlv_ez4_sonar" # Distance sensor frame
    tf:
       send: true                  # send TF?
       frame_id: "base_link"  # fcu frame_id
```

closed 2017-10-11 by vooon

vooon, 2017-10-11:
I added `base_frame_id` as a base for all distance sensor parent frame_id.
But imho better do not use send_tf: preconfigure TF in URDF and set orientation (to check consistence between FCU and URDF).
That way you will use static tf, that save a lot of bandwidth.

AlexisTM, 2017-10-11:
It is indeed a better solution.

commit 4b991770 (2017-10-11): extras: Configurable base frame id on distance_sensor

Fix #835