mavlink/mavros #836: Takeoff service is not working as expected

padmashree9110, 2017-10-11:
Hi, my application uses takeoff service with altitude set to 10m. But it fails do so, Instead of going up with given altitude drone move sideways.

I tried the same with command-line as below

`rosservice call /mavros/cmd/takeoff "{min_pitch: 0.0, yaw: 0.0, latitude: 0.0, longitude: 0.0, altitude: 10.0}" `
It is also not working as expected.

As suggested in issue#547
 I checked setpoint_position/local topic. But it is empty!

`$ rostopic echo /mavros/setpoint_position/local`

I tried setting this to (0,0,0), Still same issue.

I even issued takeoff using the [command long ](http://docs.ros.org/api/mavros_msgs/html/srv/CommandLong.html) interface

Mavros: 0.21.0
ROS: Kinetic
Ubuntu: 16.04

[  ] PX4
Version: 1.6.5 0

```
$ roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
[... 359 lines cut ...]
      - 
        key: Last dt (ms)
        value: -0.002925
      - 
        key: Mean dt (ms)
        value: -0.012327
      - 
        key: Last system time (s)
        value: 213.433899000
      - 
        key: Time offset (s)
        value: 1507716532.895795107

```

### Check ID
```
$rosrun mavros checkid
sys:comp   list of messages
  1:1     0, 1, 2, 140, 141, 147, 24, 30, 31, 32, 33, 36, 74, 76, 83, 85, 87, 102, 230, 231, 105, 111, 241, 242, 245
```

I appreciate your help
regards
padmashree

okalachev, 2017-10-12:
As far as I understood, in PX4 the altitude in command MAV_CMD_NAV_TAKEOFF works in global frame. I haven't found the way to change it.
So you should set the altitude much more than 10m (current altitude above the sea + 10m).

padmashree9110, 2017-10-12:
@okalachev :Thank You
I tried setting altitude above 10 with 30,40,100 etc. Still it's behaving the same.

okalachev, 2017-10-12:
@padmashree9110, also, I think it might be flying to actual lat=0, lon=0 location.

You should either put real latitude and longitude, or fill them with NaN's to ignore them (`.nan` in YAML notation).

I checked it in PX4, and it worked.

padmashree9110, 2017-10-13:
@okalachev : Thank you
I tried by setting real latitude and longitude, It is working now.

closed 2017-10-13 by padmashree9110